lardemua/atom #132: A change in the json file format - labels

miguelriemoliveira, 2020-03-04:
Hi @eupedrosa ,

I am working on the RWHE converter and looking into the chessboard detection.

I think you changed the format of the json file, in the collections->labels dictionary.

Before it was:

https://jsoneditoronline.org/#left=local.qeweju&right=local.wigozi

and now you have 

https://github.com/lardemua/AtlasCarCalibration/blob/71148546811bdb1a868400865819c47e23b0de1a/interactive_calibration/src/interactive_calibration/patterns.py#L29

I understand why you changed it, to accomodate the charuco ids for each keypoint, but if I understand correclty this brakes the previous datasets or not?

eupedrosa, 2020-03-04:
It only breaks very old hand-eye datasets. I no longer use those dataset. It does not break Atlas datasets.
The change adds an extra entry to the dictionary which can be safely ignore.

miguelriemoliveira, 2020-03-04:
OK, thanks for the quick reply. I will use the latest format then.

miguelriemoliveira, 2020-03-04:
Sorry again ... but I was again looking at the change and I think it breaks some things ... namely the calibration code ... look here

https://github.com/miguelriemoliveira/OptimizationUtils/blob/0205a64b2b06d356d863b1cf957214b23be8d7cb/test/sensor_pose_json_v2/objective_function.py#L157

Am I missing something? 

I mean, I agree with the change, I just want to be sure everything else keeps working. This part will have to be adjusted I think ...

eupedrosa, 2020-03-04:
> Am I missing something?

No, but I believe that you are reading it wrong. There are two keys, "idxs" and "ids".
The first is for labels and the seconds for corners.

miguelriemoliveira, 2020-03-04:
I am already a bit sleepy, but I feel lost. I have for example your dataset **eye_in_hand_10** and the labels are like this:

```
"labels": {
        "world_camera": {
          "detected": true, 
          "idxs": [
            {
              "id": 0, 
              "x": 371.713379, 
              "y": 187.831207
            }, 
            {
              "id": 1, 
              "x": 347.365448, 
              "y": 186.528259
            }, 
            {
              "id": 2, 
              "x": 322.853302, 
              "y": 185.459839
            }, 
           (...)
```

but when I look and the code here the dictionary that is produced in different (for example, it contains a new key 'keypoints')...

https://github.com/lardemua/AtlasCarCalibration/blob/71148546811bdb1a868400865819c47e23b0de1a/interactive_calibration/src/interactive_calibration/patterns.py#L29

Can you upload a recent dataset? Is is the same?

eupedrosa, 2020-03-04:
You really should go to sleep. The **eye_in_hand_10** is a recent dataset. 
You are looking at half the code. That is just what the "Pattern" detects. Then, the `interactive_data_labeler.py` is the one that creates the output json.

Look here:
https://github.com/lardemua/AtlasCarCalibration/blob/71148546811bdb1a868400865819c47e23b0de1a/interactive_calibration/src/interactive_calibration/interactive_data_labeler.py#L310

miguelriemoliveira, 2020-03-05:
Hi @eupedrosa ,

got it (finally :)). Thanks.

closed 2020-03-05 by miguelriemoliveira