lardemua/atom #133: Create a description of the Agrob robot system

miguelriemoliveira, 2020-03-06:
Hi @aaguiar96 ,

So, lets assume the name of your system will be  **agrob**, ok?

For calibrating, we need a description of the robotic system to be calibrated (if you read the paper I think you understand our non-intrusive approach to the calibration of robotic systems).

This description is normally in the form of an urdf or a xacro file(s), and is  placed 
in the **agrob_description** ros package. This will also contain the meshes (collada dae files) for drawing the robot in rviz.

here's an example:

https://github.com/lardemua/atlas-core

You could also have a **agrob_bringup** ros package where you put the launch files, as in the rustbot example

https://github.com/miguelriemoliveira/RustBot

I am not sure if you have this or not, but if you don't we should start working on this. 
What fo you say?

miguelriemoliveira, 2020-03-09:
https://github.com/tu-darmstadt-ros-pkg/hector_models/tree/kinetic-devel/hector_sensors_description/meshes/vlp16

miguelriemoliveira, 2020-03-09:
https://github.com/lardemua/atlas-core/blob/master/atlas2_description/urdf/atlas2_macro.urdf.xacro

aaguiar96, 2020-03-11:
Hi @miguelriemoliveira 

I've been studying this issue and a college of mine also introduced me this topic.

So, our robot's base is a Husky which provides the `xacro` [here](https://github.com/husky/husky/tree/melodic-devel/husky_description).
Additionally, I found the [urdf file](https://github.com/stereolabs/zed-ros2-wrapper/tree/master/zed_wrapper/urdf) for the ZED camera. Is that enough or do we need the `xacro` file?

I'll start designing the tower that contains the sensors upon the Husky base of our robot to then export it using [urdf exporter](http://wiki.ros.org/sw_urdf_exporter).

What do you think?

eupedrosa, 2020-03-11:
> Additionally, I found the urdf file for the ZED camera. Is that enough or do we need the xacro file?

A `xacro` is a macro language based on `xml`. It help maintaining complex urdf files and promotes the re-usability of description files. You should read more about it http://wiki.ros.org/xacro

aaguiar96, 2020-03-14:
Hi @miguelriemoliveira and @eupedrosa 

What do you think? :)

[image]

miguelriemoliveira, 2020-03-14:
Hi @aaguiar96 ,

looks really nice. Can you post picture for comparison?

Once this is complete we can start working on the calibration, not sure if we should start next week or do a hiatus because of the fct project ,,,

eupedrosa, 2020-03-14:
> Once this is complete we can start working on the calibration, not sure if we should start next week or do a hiatus because of the fct project

https://www.fct.pt/noticias/index.phtml.pt?id=508&/2020/3/Concurso_de_Bolsas_de_Doutoramento_2020_e_Concurso_de_Projetos_de_IC&DT

Maybe you should ignore this.....

aaguiar96, 2020-03-14:
> looks really nice. Can you post picture for comparison?

Here is the real tower @miguelriemoliveira 

[image]

miguelriemoliveira, 2020-03-16:
The picture and the model look very similar. I think it was a good work. Another thing I would ask you to post is the tf tree for this model (after that I think we can close the issue). 

I think we should work on the fct project for now (at least we should kick start it), and then go back to the calibration.

The flir camera we can also ignore for now. Once we are calibrating more stuff on the agrob, we can expand the xacro description accordingly.

aaguiar96, 2020-03-17:
> The picture and the model look very similar. I think it was a good work. Another thing I would ask you to post is the tf tree for this model (after that I think we can close the issue).

Here it is! :) 

[image]

> I think we should work on the fct project for now (at least we should kick start it), and then go back to the calibration.

I agree. I've already created the document. So, we can start writing!

aaguiar96, 2020-05-27:
Closing this issue. All work is done here.

closed 2020-05-27 by aaguiar96