lardemua/atom #135: Create dataset translator for tabb et al

eupedrosa, 2020-03-11:
code: https://github.com/amy-tabb/RWHEC-Tabb-AhmadYousef

dataset: https://data.nal.usda.gov/dataset/data-solving-robot-world-hand-eyes-calibration-problem-iterative-methods

miguelriemoliveira, 2020-03-14:
I think I got the first version working ... here's the dataset.json for hte DS7

https://jsoneditoronline.org/#left=local.yotigu&right=local.xivasu

I tested with DS1 and with DS7 and it seems to be working. 

This one takes more time since we calibrate the sensors intrinsics everytime the script is called.

How to call for DS1

```bash
rosrun interactive_calibration convert_from_tabb_dataset.py -ds /home/mike/datasets/tabb/DS1/ -out /home/mike/datasets/DS1_converted -json /home/mike/catkin_ws/src/AtlasCarCalibration/interactive_calibration/calibrations/tabb/ds1.json
```

and for DS7

```bash
rosrun interactive_calibration convert_from_tabb_dataset.py -ds /home/mike/datasets/tabb/DS7/ -out /home/mike/datasets/DS7_converted -json /home/mike/catkin_ws/src/AtlasCarCalibration/interactive_calibration/calibrations/tabb/ds7.json
```

miguelriemoliveira, 2020-03-14:
@eupedrosa , I think it is working, please try it out. 

I will do the opposite conversion later on ...

eupedrosa, 2020-03-14:
Nice.. I'm gonna test it.

eupedrosa, 2020-03-15:
It is not working, it does not converge to a solution. I'm still investigating why. Maybe a bad first guess? Or something in the data import?

miguelriemoliveira, 2020-03-15:
Hi Eurico.

Could be something  in the conversion.

There was no first guess so I just put 0 0 0 0 0 0 for x y z r p y.

Tell me how I can help.

eupedrosa, 2020-03-15:
I think I solved it.
The transformation had to be inverted and the translation were in milimeters.

miguelriemoliveira, 2020-03-15:
Great!

eupedrosa, 2020-03-17:
The Tabb dataset is, in my opinion, not great. It has no rotations and most of the views are perpendicular to the pattern plane (i.e. ccd and and patern plane are parallel). This is specially noticeable when optimizing the intrinsic parameters and I get a value of 30 to the skew and the orientation of the sensor is compensated for this. This means that there is a degree of freedom that is neglected by this dataset.

@miguelriemoliveira, what is opinion about this?

miguelriemoliveira, 2020-04-08:
Startting to work to the inverse process of translation ...

miguelriemoliveira, 2020-04-08:
> The Tabb dataset is, in my opinion, not great. It has no rotations and most of the views are perpendicular to the pattern plane (i.e. ccd and and patern plane are parallel). This is specially [...]

I agree with what you say.

miguelriemoliveira, 2020-04-08:
Hi @eupedrosa ,

I have a convert_to_tabb script functional. Can you test it?

Instruction on how to use here:

https://github.com/lardemua/AtlasCarCalibration#to-convert-to-an-tabb-dataset-run

Be careful about this note:

https://github.com/lardemua/AtlasCarCalibration/blob/16f3622d2698829e0419c28f30643e06c1f71389/interactive_calibration/scripts/convert_to_tabb_dataset.py#L113-L114

Tell me if it works.

miguelriemoliveira, 2020-04-29:
Hi @eupedrosa 

this is working right? Should we close?

eupedrosa, 2020-04-29:
Yes, we can close.

closed 2020-04-29 by eupedrosa