lardemua/atom #137: Chessboard is drawn in the wrong side of the board?

miguelriemoliveira, 2020-04-18:
Hi @eupedrosa ,

I think the chessboard is being draw on the wrong side of the board. 

If I run this 

`roslaunch interactive_calibration eye_calibration_view_optimization.launch`

and then this

`./test/general_hand_eye/general_hand_eye.py -si -rv -json ~/datasets/eye_to_base/data_collected.json -i  -csf "lambda x: int(x)%4==0"`

I get an image like this:

[image]

Can you help?

eupedrosa, 2020-04-18:
It is not a *bug* it is a *feature*.

The chessboard is being drawn with correct side. The chessboard detector uses a *left-hand* axis but we assume to be a *right-hand* axis. This results in a mirror effect when drawing the chessboard. The charuco detectors uses a *right-hand* axis, therefore, optimizations with that type chessboard results in a correct draw. Did you understand the *bug*?

miguelriemoliveira, 2020-04-18:
Hum, not sure I understood fully why this happens, but can't we do something to change it? I mean, the reviewer will just say, "why is the camera is on the wrong side of the charuco", won't he?

eupedrosa, 2020-04-18:
> Hum, not sure I understood fully why this happens, but can't we do something to change it? I mean, the reviewer will just say, "why is the camera is on the wrong side of the charuco", won't he?

We can always argue that we model has the pattern drawn in both sides.

miguelriemoliveira, 2020-04-18:
Yes but that way there are some perspectives which we cannot show, because
they show an empty board.

I can try to see if i can solve it with a flag which we use just to get
nice drawings...

miguelriemoliveira, 2020-04-18:
That's strange ... for the eye in hand the charuco seems to be drawn on the correct side

[image]

eupedrosa, 2020-04-19:
> That's strange ... for the eye in hand the charuco seems to be drawn on the correct side

I guess you did not read my reply in full.. ;)
> the charuco detectors uses a right-hand axis, therefore, optimizations with that type chessboard results in a correct draw. 

Yes, the charuco draws on the correct side. 
The chessboard has its origin in the top left side, while the charuco has its origin in the top bottom side. Check how the axis are drawn with TF in RVIZ.

miguelriemoliveira, 2020-04-19:
Hi @eupedrosa ,

Sorry, I am still confused. I get it that we are using a different convention form the aruco detector, and that that mirrors the pattern. 

But why the difference between both use cases? They both have charucos ...

Thanks,
Miguel

miguelriemoliveira, 2020-04-19:
Got it ... they are both charucos but in the eye-to-base case you are using a chessboard detector ... here's the config.json

```json
(...)
  "calibration_pattern" : {
        "link": "chessboard_link",
        "parent_link": "ee_link",
        "origin": [0.0, 0.0, 0.0, 1.5, 0.0, 0.0],
        "fixed": true,
        "pattern_type": "chessboard",
        "border_size": 0.5,
        "dimension": {"x": 9, "y": 7},
        "size": 0.019,
        "inner_size": 0.014
    }
```

while in the eye-on-hand you use a charuco detector

```json
  "calibration_pattern" : {
        "link": "chessboard_link",
        "parent_link": "base_link",
        "origin": [-0.37, 0.0, 0.0, 0, 0.0, 3.8],
        "fixed": true,
        "pattern_type": "charuco",
        "border_size": 0.5,
        "dimension": {"x": 13, "y": 8},
        "size": 0.02,
        "inner_size": 0.015
    }
```

but why on earth do you not use a charuco detector in the eye-to-base use case?

eupedrosa, 2020-04-19:
> but why on earth do you not use a charuco detector in the eye-to-base use case?

Check the dataset images and verify that it uses charuco. If is uses a charuco we should change the detector.

miguelriemoliveira, 2020-04-19:
Yes, it does, check new fig 3 of the paper. That was why I was getting confused ...
I will change it.
Thanks

miguelriemoliveira, 2020-04-19:
Hi @eupedrosa , can you please change it instead of me?

I am not sure if I should also change the other parameters ...

eupedrosa, 2020-04-19:
Changing the calibration pattern in `data_collecte.json` changes nothing.
I may have to send you the datasets that I am using for the paper. We have a lot of datasets and we may not be in sync.

miguelriemoliveira, 2020-04-19:
yes, please send me what you are using. We should be synced.

Thanks,
Miguel
*  Miguel Riem de Oliveira                                             *
email:    mriem@ua.pt    |    m.riem.oliveira@gmail.com
office: 22.2.18                 |    ext: 23887
Dep. de Engenharia Mecânica, Universidade de Aveiro
*Campus Universitário de Santiago, 3810-193  Portugal*

miguelriemoliveira, 2020-04-19:
Thanks, @eupedrosa . 

Now everything is clear. Once we create new datasets we will also need to change 

interactive_calibration/calibrations/ur10e/eye_in_hand/config.json

closed 2020-04-19 by miguelriemoliveira