lardemua/atom #138: eye-on-hand or eye-in-hand???

miguelriemoliveira, 2020-04-18:
the most important existential doubt that ever existed ... 

@eupedrosa , which one are we using after all?

eupedrosa, 2020-04-19:
*on* - preposition
1. physically in contact with and supported by (a surface). "on the table was a water jug"
2. forming a distinctive or marked part of the surface of. "a scratch on her arm"

*in* - preposition
1. expressing the situation of something that is or appears to be enclosed or surrounded by something else. "I'm living in London"
2. expressing a period of time during which an event happens or a situation remains the case. "they met in 1885"

Given the definition of the prepositions, we should go with ***on*** !!!!

miguelriemoliveira, 2020-04-19:
Great, it is setteled then. Thanks.

closed 2020-04-19 by miguelriemoliveira