lardemua/atom #139: Create labelling procedure for 3D LiDAR scans

aaguiar96, 2020-04-20:


miguelriemoliveira, 2020-04-20:
Start with a standalone python code that isolates the points from the pattern board using ransac on the point cloud (or a subset of it)

Ransac in python can be taken from 
https://github.com/miguelriemoliveira/RansacPlaneDetection

aaguiar96, 2020-04-20:
Hi @miguelriemoliveira  and @eupedrosa 

I created a standalone repo so that you can follow the code.
You can find it [here](https://github.com/aaguiar96/3DScanLabeler).

I've written the PCL callback today, and now I'll start the algorithm itself. :)

miguelriemoliveira, 2020-04-21:
Hi André,

when I said standalone I means a standalone file. It did not have to be an
entirely new repository,

In any case, the important thing is that you are moving forward!

Regards,
Miguel
*  Miguel Riem de Oliveira                                             *
email:    mriem@ua.pt    |    m.riem.oliveira@gmail.com
office: 22.2.18                 |    ext: 23887
Dep. de Engenharia Mecânica, Universidade de Aveiro
*Campus Universitário de Santiago, 3810-193  Portugal*

aaguiar96, 2020-04-23:
Hi @miguelriemoliveira and @eupedrosa 

I think RANSAC works... :) 

[image]
[image]

aaguiar96, 2020-04-23:
Now, my idea is to:

- Validate the results more extensivelly
- Use the `Publish Point` of Rviz to set the seed point (now I'm setting it on the code)
- Trank the chessboard points frame-by-frame

What do you think?

miguelriemoliveira, 2020-04-23:
Great! Looks good. But I think now is the time to integrate with the data_collector and labellor

Because the second functionality we have it already implemented ...

Not sure what's best, but I think this may be a good time to start integrating ... If you want we can meet and I guide you through the process.

aaguiar96, 2020-04-23:
Ok, sure. 
We can integrate and then implement the missing features.

Ok Miguel, we can meet. I'll send you an e-mail.

aaguiar96, 2020-04-23:
I think I found a solution much simpler than RANSAC.
Just selecting the middle point of the chessboard on the initialization, and then tracking it gives a great result.
In the gif, the white points are the points estimated to be on the chessboard, and the big white point is the seed point, resulting from the tracking.

[image]

On the contrary, RANSAC is failing to estimate the chessboard plane in many iterations and considers the person body in a lot of them to...
Of course, it could be possible to refine the RANSAC solution since it is more elegant.

eupedrosa, 2020-04-23:
There are a few points in white that belong to the user, they are _unwanted_.

> For example, we can feed the RANSAC only with these points to eliminate some outliers that might appear.

This may work.

miguelriemoliveira, 2020-04-23:
Hi @aaguiar96 ,

results look really nice. I think @eupedrosa  is right about doing ransac as a post processing to refine the results.

I did not really understand how you did the tracking part:

> Just selecting the middle point of the chessboard on the initialization, and then tracking it gives a great result.
> In the gif, the white points are the points estimated to be on the chessboard, and the big white point is the seed point, resulting from the tracking.

Can you give more details? I am interested.

aaguiar96, 2020-04-23:
> Can you give more details? I am interested.

Since RANSAC was not working as well as expected, I tried out a simple tracking approach.
So, the user initializes the system selecting a point on the chessboard (close to the middle one preferably).
Then, the chessboard is clustered using its approximate real size.
The tracking is done computing the seed point in each iteration by the average of all points in the cluster. :+1:

aaguiar96, 2020-04-24:
> For example, we can feed the RANSAC only with these points to eliminate some outliers that might appear.

Ok, I think this approach is working.
I was having problems with RANSAC because I forgot to reset the number of inliers to 0 before starting RANSAC, for each frame received. Also, I added the feature of checking if the three selected points are colinear.

This is the result with `distance_threshold = 0.02` and 100 iterations, feeding the RANSAC only with the tracked points.

[image]

Do you think is good enough or still can be improved?

miguelriemoliveira, 2020-04-24:
Hi  @aaguiar96 ,

it looks really nice. I would say lets try with this one and then if we need a more accurate detection we revisit the problem. 

For now I would say we need to integrate with the collect_and_label script (on me for now, #141) .

Then, the goal is to generate a dataset from the agrob bagfile. I would close this one and work on #141 from now on ...

aaguiar96, 2020-04-24:
> One question: you did a new feature which sounds nice. It should be integrated into https://github.com/miguelriemoliveira/RansacPlaneDetection
> I will add you as a collaborator, can you push?

Yes sure @miguelriemoliveira 

> For now I would say we need to integrate with the collect_and_label script (on me for now, #141) .

Ok, I agree.
I'll close the issue then. Thanks for the support @miguelriemoliveira

closed 2020-04-24 by aaguiar96