lardemua/atom #140: Add Agrob description to playback launch file

aaguiar96, 2020-04-20:


miguelriemoliveira, 2020-04-20:
The file is there as a copy of another calibration ... you must change it now.

aaguiar96, 2020-04-20:
@miguelriemoliveira should I also place the agrob xacro description on `AtlasCarCalibration/interactive_calibration/urdf/` folder and load it from the launch file?

miguelriemoliveira, 2020-04-21:
Hi André,

Take the atlascar as an example. I think the xacros should be in your
robot's package.

in the atlascar the xacro files are in the atlascar2 folder.

Hope it helps,
Miguel
*  Miguel Riem de Oliveira                                             *
email:    mriem@ua.pt    |    m.riem.oliveira@gmail.com
office: 22.2.18                 |    ext: 23887
Dep. de Engenharia Mecânica, Universidade de Aveiro
*Campus Universitário de Santiago, 3810-193  Portugal*

aaguiar96, 2020-04-21:
Hi @miguelriemoliveira and @eupedrosa 

I think it is done, you can check it.
The only difference to atlascar2 was that the velodyne PCL, when recorded, was on the `/velodyne` referential frame, and the `urdf` generates a link with the name `vlp16_frame`. So, I had to add a static dummy link from `/vlp16_frame` to `/velodyne` so that the PCL can be visualized on Rviz without the tf's generated by the bagfile.

aaguiar96, 2020-04-21:
Also, I've already written the `config.json` file for agrob.
I leave here the tf tree generated by the agrob `xacro`.

[image]

miguelriemoliveira, 2020-04-21:
Hi @aaguiar96 ,

I am in my teaching inferno. Will only be over tomorrow by the end of the afternoon.

I will take a look on Thursday.

miguelriemoliveira, 2020-04-21:
Great! I will take a look on Thursday.

See you,Miguel

*  Miguel Riem de Oliveira                                             *
email:    mriem@ua.pt    |    m.riem.oliveira@gmail.com
office: 22.2.18                 |    ext: 23887
Dep. de Engenharia Mecânica, Universidade de Aveiro
*Campus Universitário de Santiago, 3810-193  Portugal*

miguelriemoliveira, 2020-04-21:
Hi @aaguiar96 

> So, I had to add a static dummy link from /vlp16_frame to /velodyne 

and you do this in the xacro?

aaguiar96, 2020-04-22:
> Hi @aaguiar96
> 
> > So, I had to add a static dummy link from /vlp16_frame to /velodyne
> 
> and you do this in the xacro? Or with a static_transform_publisher?

I've done it with static_transform_publisher.

aaguiar96, 2020-04-24:
@miguelriemoliveira, I think we can close this one too. What do you think?

miguelriemoliveira, 2020-04-24:
Hi @aaguiar96 ,

we tested today and it seems to be working. 

Problems to solve in the next bag file:

1. camera_info topic matching the image topic
2. velodyne lidar reference frame

closed 2020-04-24 by miguelriemoliveira

miguelriemoliveira, 2020-04-24:
Telepathy :)