lardemua/atom #142: cv2.aruco_DetectorParameters object has no attribute 'doCornerRefinement'

miguelriemoliveira, 2020-04-24:
Hi @eupedrosa ,

for #141 we need to run data_collector_and_labeller using charucos, but I get and error as follows:

I had another error before which required the pip installation of **opencv-contrib-python**, but now this:

```
[ERROR] [1587740267.642691, 1584007733.026632]: bad callback: <bound method InteractiveDataLabeler.sensorDataReceivedCallback of <interactive_calibration.interactive_data_labeler.InteractiveDataLabeler instance at 0x7f4e4561a550>>
Traceback (most recent call last):
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/topics.py", line 750, in _invoke_callback
    cb(msg)
  File "/home/mike/catkin_ws/src/AtlasCarCalibration/interactive_calibration/src/interactive_calibration/interactive_data_labeler.py", line 185, in sensorDataReceivedCallback
    self.labelData()  # label the data
  File "/home/mike/catkin_ws/src/AtlasCarCalibration/interactive_calibration/src/interactive_calibration/interactive_data_labeler.py", line 306, in labelData
    result = self.pattern.detect(image)
  File "/home/mike/catkin_ws/src/AtlasCarCalibration/interactive_calibration/src/interactive_calibration/patterns.py", line 52, in detect
    param.doCornerRefinement = False
AttributeError: 'cv2.aruco_DetectorParameters' object has no attribute 'doCornerRefinement'

```

Can you help?

eupedrosa, 2020-04-24:
Hi @miguelriemoliveira.
Can you remove that line, line 52, and try again? It was some tests that I was doing, but that is only present in a higher version of opencv.

miguelriemoliveira, 2020-04-24:
Will try it and get back to you ...

miguelriemoliveira, 2020-04-24:
It works! I will push this version with that line commented. Thanks.

closed 2020-04-24 by miguelriemoliveira

Changhee-Kimm, 2022-01-04:
if you remove that line, it may decrease the accuracy of corner refinement.   it means the accuracy of finding corners is decreased.

'doCornerRefinement' is under version of opencv 3.2.0, and above 3.2.0, it is changed to  'cornerRefinementMethod'

so I recommend you to add  cv2.aruco.cornerRefinementMethod = something

commit 70a1f002 (2020-04-24): Fixes #142. Close.