lardemua/atom #144: Should we add a field in the config.json to indicate the type of message we have?

miguelriemoliveira, 2020-04-26:
Hi @TiagoaTavares @eupedrosa and @aaguiar96 ,

for now, our config  json files did not have information about how the data should be labelled and later on optimized.

The reason for that is that we assumed that it would be the **message type**, which is not in the json, it is inferred by the **data collector** from the **message topic**, which is in the config file.

An example of the json bellow.

Now since @aaguiar96 started working on the collection of 3d lidar data, we have a conflict between what @tiagomfmadeira was doing and the new stuff from @tiagomfmadeira . The problem is that both @tiagomfmadeira and @aaguiar96  use a PointCloud2 message type (the velodyne and the epth camera), so they should have the same code for collecting, labelling and computing the cost in the objective function.

My question is: how realistic do you think this is? My initial idea was to have as much automated stuff as possible, but now I am not entirely sure we can merge @tiagomfmadeira and @aaguiar96 's approaches.

The alternative would be adding some field to the json, (like the example bellow with the new calibration_type filed), but I would like to remove as much as possible, not add. This is a quesiton of software design ... I vote for the merging ... but would like to know what you think ... 

```json
{
    "sensors": {
        "right_camera": {
            "link": "zed_right_camera_optical_frame",
            "parent_link": "zed_camera_center",
            "child_link" : "zed_right_camera_frame",
            "topic_name" : "/zed/zed_node/right/image"
            "calibration_type" : "rgb_image"
        },

        "left_camera": {
            "link": "zed_left_camera_optical_frame",
            "parent_link": "zed_camera_center",
            "child_link" : "zed_left_camera_frame",
            "topic_name" : "/zed/zed_node/left/image"
            "calibration_type" : "rgb_image"
        },

	"vlp16": {
	   "link": "velodyne",
	   "parent_link": "tower_link",
	   "child_link": "vlp16_frame",
	   "topic_name": "/velodyne_points"
            "calibration_type" : "point_cloud"
	}
    },

    "anchored_sensor": "right_camera",
    "world_link": "base_link",
    "max_duration_between_msgs": 0.1,

    "calibration_pattern" : {
        "link": "chessboard_link",
        "parent_link": "base_link",
        "origin": [-0.37, 0.0, 0.0, 0, 0.0, 3.8],
        "fixed": true,
        "pattern_type": "charuco",
        "border_size": 0.5,
        "dimension": {"x": 13, "y": 8},
        "size": 0.029,
        "inner_size": 0.021
    }
}
```

aaguiar96, 2020-04-27:
Hi @miguelriemoliveira 

I am not entirely acquainted with the approach of @tiagomfmadeira but given that we have the same data types, at least the labeling procedure should be equal on both sides, in my opinion. In other words, each type of message should have a labeling procedure, independent of the type of sensor, don't you think?
Just a thought, I don't know if this is actually doable. :)

miguelriemoliveira, 2020-04-29:
Hi @aaguiar96 ,

thanks for the feedback. In fact @eupedrosa gave a great idea by phone: we take the best of both worlds. We will use the message type as you say, but also have the functionality of having a parameter which can override the message type.

I will implement once I have time (#145)

closed 2020-04-29 by miguelriemoliveira