lardemua/atom #147: Synchonize agrob zed-velodyne timestamps on next recordings

aaguiar96, 2020-05-01:
Since the ZED and velodyne are in different machines, the timestamps are different in the current bagfiles. 

Miguel's idea: use the `--clock` while executing `rosbag record` to use only the master's timestamp.

eupedrosa, 2020-05-01:
This is the major weakness of ROS. Non synchronized clock. Different machines, different clocks.
The most common solution is to use the ROS_MASTER as an NTP server. The others computers should synchronize at boot.

This changes require access to the hardware, something that right now is not possible. Miguel´s  idea requires two things: something publishing the clock, and the data to be produce with the parameter `use_sim_time` set to `true`.

This is a tricky problem...

miguelriemoliveira, 2020-05-22:
Hi @aaguiar96 ,

just a question: can you connect the zed camera's directly to the desktop, even if you just don't compute the disparity?

That could solve the time stamp issue...

aaguiar96, 2020-05-22:
> Hi @aaguiar96 ,
> 
> just a question: can you connect the zed camera's directly to the desktop, even if you just don't compute the disparity?
> 
> That could solve the time stamp issue...

I think so. Next time I'll give it a try! :)

aaguiar96, 2020-06-24:
Closing this since it is fixed.

closed 2020-06-24 by aaguiar96

miguelriemoliveira, 2020-06-24:
This is really nice that you fixed it. Can you post here the link you used for future reference? In any case, next time I try to do this I will talk to you :)

aaguiar96, 2020-06-24:
Here:

Install `ntp`, `ntpdate` and `chrony` from `atp` on both machines.

```
ntpdate -q <IP of your time server>
```

if you get *no server suitable for synchronization found*

choose one machine as time server. do on this machine: 

```
sudo vim /etc/chrony/chrony.conf
```

Add these lines:

```
# make it serve time even if it is not synced (as it can't reach out)
local stratum 8
# allow the IP of your peer to connect
allow <IP of your time-clinet>
```

Then do on the time client(machine that changes its time towards the server): 

```
sudo vim /etc/chrony/chrony.conf
```

Add these lines:

```
server <server_ip> minpoll 0 maxpoll 5 maxdelay .05
```

change maxdelay as you wish. 0.05 worked for me!

restart both machines

check with

```
ntpdate -q <IP of your time server>
```

if they still have a big offset do:

```
sudo /etc/init.d/chrony stop
sudo /etc/init.d/chrony start
```

check again. The offset should now get better.

hope that helps!

miguelriemoliveira, 2020-06-24:
Great. Thanks for the documentation.