lardemua/atom #148: Show agrob decription during the optimization procedure

aaguiar96, 2020-05-04:


miguelriemoliveira, 2020-05-10:
I am on it.

aaguiar96, 2020-05-10:
Thanks Miguel! Starting to work on this again tomorrow! :)

miguelriemoliveira, 2020-05-10:
Ok, we can talk if you want, but I will be busy from 16h onward, so it has to be before.

aaguiar96, 2020-05-10:
I'll try to develop the cost function in the morning. We can talk at 15h if you want.

miguelriemoliveira, 2020-05-10:
ok.

See you tomorrow.

Miguel

*  Miguel Riem de Oliveira                                             *
email:    mriem@ua.pt    |    m.riem.oliveira@gmail.com
office: 22.2.18                 |    ext: 23887
Dep. de Engenharia Mecânica, Universidade de Aveiro
*Campus Universitário de Santiago, 3810-193  Portugal*

miguelriemoliveira, 2020-05-11:
Working on it ... it is already shows the robot but we still havea lot of loose ends ... tomorrow .

[image]

miguelriemoliveira, 2020-05-11:
OK, so now we are drawing the velodyne points. Here's an image:

[image]

miguelriemoliveira, 2020-05-11:
Tomorrow I will try to check why the chessboard is not being drawn.

miguelriemoliveira, 2020-05-15:
Ok, so now we have the chessboard. The visualization.py can be generalized to be the same for all calibration cases. However to undertake this endeavour, I will need a bagfile with some collections, cameras, 3D lidar and also 2D lidars. Need to have everything to then test the visualization.

@aaguiar96 would it be possible to have also in the next agorb bagfile also the 2D sick lidar data? That way we could use agrob as the testbed that has everything ... and we would also calibrate that one ...

For now this will wait for a better bagfile to continute development.

aaguiar96, 2020-05-15:
> Ok, so now we have the chessboard. The visualization.py can be generalized to be the same for all calibration cases. However to undertake this endeavour, I will need a bagfile with some collections, cameras, 3D lidar and also 2D lidars. Need to have everything to then test the visualization.

Nice @miguelriemoliveira  :)

> @aaguiar96 would it be possible to have also in the next agorb bagfile also the 2D sick lidar data? That way we could use agrob as the testbed that has everything ... and we would also calibrate that one ...

Hum... We do not have any planar laser available at this moment in agrob. Velodyne can work with a single ring, emulating a planar laser, but I think you need that from different sources right?

miguelriemoliveira, 2020-05-15:
Yes, different sensors would be better. Even to emulate all the stuff with the ros 2D lidar drivers.

I remembered before you had a yellow sick, that's why I asked.

Lets proceed with the velodyne and the zed cameras only. 

I will try to develop a general visualization code testing different bags.

aaguiar96, 2020-05-15:
Ok @miguelriemoliveira, sounds great.
I'll work on the cost function. I let you know as soon as I have new results.

Also, I'll call Tania Barbosa on Monday. :)

miguelriemoliveira, 2020-05-15:
Ok, look, I saw your code running. I think your cost function is well implemented (the beam error is being propoperly optimized, I think). 

The problem is, we may have to increase the complexity of the cost function. When you are available, let's chat for a moment.

aaguiar96, 2020-05-15:
I @miguelriemoliveira 

I was looking for it right now. I can't find any bug...
I have half an hour free now. Let's meet?

eupedrosa, 2020-05-15:
If you want, I can also participate (@miguelriemoliveira, @aaguiar96)

miguelriemoliveira, 2020-05-15:
Hi @eupedrosa 

only now I saw this. Sorry. I said to @aaguiar96 to try to implement the border distance metric.

eupedrosa, 2020-05-15:
So.. you did not want my participation :cry:

> only now I saw this. Sorry. I said to @aaguiar96 to try to implement the border distance metric

:+1:

miguelriemoliveira, 2020-05-15:
: - )

aaguiar96, 2020-05-15:
> If you want, I can also participate (@miguelriemoliveira, @aaguiar96)

Sorry @eupedrosa, I only saw it now too... :D

miguelriemoliveira, 2020-05-26:
Work on this will proceed in a general fashion for all robotic systems. 

Further developments on #156

closed 2020-05-26 by miguelriemoliveira