lardemua/atom #150: Create a collada file with your charuco

miguelriemoliveira, 2020-05-11:
Hi @aaguiar96 ,

we need a collada file (dae) of your charuco.

Check what @eupedrosa did for the eye in hand case:

package://interactive_calibration/meshes/charuco_5x5.dae

For now, this is hardcoded here (will try to make it dynamic later on)

https://github.com/miguelriemoliveira/OptimizationUtils/blob/27a6546d4b8ae302fe655103c2788167971576c0/test/sensor_pose_json_v2/visualization.py#L88

aaguiar96, 2020-05-11:
Hi @eupedrosa 

Did you write the collada file for you charuco from scratch?

eupedrosa, 2020-05-11:
> Did you write the collada file for you charuco from scratch?

NO :smile: I used blender. 
[charuco.zip](https://github.com/lardemua/AtlasCarCalibration/files/4610232/charuco.zip)

To generate the charuco image I used this: https://github.com/opencv/opencv_contrib/blob/3.4.0/modules/aruco/samples/create_board_charuco.cpp

But you must convert meters to pixels. You can use you own scale.

aaguiar96, 2020-05-11:
> But you must convert meters to pixels. You can use you own scale.

Blender exports the mesh in meters, and I have to manually convert them to pixels?

eupedrosa, 2020-05-11:
Yes. But then in blender, that scale must reflect the dimension of the 3D object.

aaguiar96, 2020-05-11:
That's hard... I don't know the real size of the chessboard (only approximately), and consequently, don't know the size of the markers in pixels...

I'll try to approximate it.

aaguiar96, 2020-05-20:
Hi @miguelriemoliveira  and @eupedrosa 

Now that I'm using yours charuco, should I change the calibration pattern settings to the ones present on `atlascar2_calibration.json`?

miguelriemoliveira, 2020-05-20:
Hi @aaguiar96 ,

no, that file was configured for usingthe chessboard. We received the charuco late january, and did not do a bag file on the atlascar with it yet.

Best starting point is either yours or @eupedrosa 's hand eye.

aaguiar96, 2020-05-26:
Hi @eupedrosa can you post our blender file again?
When I open it on blender I got nothing! :)

eupedrosa, 2020-05-26:
Which blender version are you using? I used blender 2.8

aaguiar96, 2020-05-26:
Got it, I had the wrong version! :)

aaguiar96, 2020-05-27:
Hi @miguelriemoliveira and @eupedrosa 

What do you think?

[image]

eupedrosa, 2020-05-27:
The border of the pattern is missing..

aaguiar96, 2020-05-27:
Better? 
[image]

eupedrosa, 2020-05-27:
Yes, better. But it looks washed. In the previous images it looked better.

aaguiar96, 2020-05-27:
No, it's because of rviz. Two topics publishing the meshes I guess...
I removed one of them now:

[image]

eupedrosa, 2020-05-27:
I guess we can close this issue, right?

aaguiar96, 2020-05-27:
Yes, thanks.

miguelriemoliveira, 2020-05-27:
Looks really nice. Yes, the fog like effect is because of drawing several overlapping meshes .

I used the "single_pattern" argument for this:

https://github.com/miguelriemoliveira/OptimizationUtils/blob/0c875343f48c7bb44b7d7f79bd685f137a65a29d/test/sensor_pose_json_v2/visualization.py#L85-L102

miguelriemoliveira, 2020-05-27:
Can you use the same @aaguiar96 ?

aaguiar96, 2020-05-27:
`-sp` right? The pattern does not appear with this argument...

miguelriemoliveira, 2020-05-27:
yes, -sp ... or --single_pattern

https://github.com/miguelriemoliveira/OptimizationUtils/blob/0c875343f48c7bb44b7d7f79bd685f137a65a29d/test/sensor_pose_json_v2/main.py#L78-L79

miguelriemoliveira, 2020-05-27:
How are you running the node? Can you paste here?

aaguiar96, 2020-05-27:
Like this:
```
clear && test/sensor_pose_json_v2/main.py -json /home/andre/Documents/rosbags/agrob_second_dataset/data_collected.json -sp -si -rv -ssf "lambda name: name in ['left_camera','right_camera','vlp16']"
```

miguelriemoliveira, 2020-05-27:
I am on it...

miguelriemoliveira, 2020-05-27:
This is the problem I think ... rviz is giving this error:

> ERROR] [1590587449.530670546]: Could not load resource [package://interactive_calibration/meshes/charuco.dae]: Unable to open file "package://interactive_calibration/meshes/charuco.dae".

eupedrosa, 2020-05-27:
This is something that we overlooked. What model should be loaded?? Diferent projects have different patterns. This may be problematic.

aaguiar96, 2020-05-27:
Sorry, I forgot to commit to OptimizationUtils. You can pull and the error will disapear.

aaguiar96, 2020-05-27:
> This is something that we overlooked. What model should be loaded?? Diferent projects have different patterns. This may be problematic.

Maybe an argument specifying the robotic platform that is being optimized? Or an argument with the path to the mesh file?

miguelriemoliveira, 2020-05-27:
I am working on this now, I am adding a new parameter on the config,yml. Check the meshfile param.

```
  # This section describes the properties of the calibration pattern used
    # for calibration.
    "calibration_pattern" : {
        # The frame id (or link) of the pattern.
        # This link transformation will be optimized.
        "link": "chessboard_link",
        # The parent frame id (or link) of the pattern.
        # For example, in hand-eye calibration the parent link
        # of the pattern can be the end-effector or the base of the arm
        "parent_link": "base_link",
        # Defines if the pattern link is the same in all collections (i.e. fixed=true),
        # or each collection will have its own estimative of the link transformation.
        "fixed": false,
        # The type of pattern used for the calibration.
        # Suported pattern are:
        # - chessboard
        # - charuco
        "pattern_type": "chessboard",
        # If the pattern type is "charuco" you need to define
        # the aruco dictionary used by the pattern.
        # See https://docs.opencv.org/trunk/dc/df7/dictionary_8hpp.html
        "dictionary": "DICT_5X5_100",
        # Mesh file (collada.dae or stl) for showing pattern on rviz
        "mesh_file": "/home/mike/catkin_ws/src/calibration/AtlasCarCalibration/interactive_calibration/meshes/charuco_5X5_800x600.dae",
        # The border width from the edge corner to the pattern physical edge.
        # This is used for 3D sensors and lidars.
        "border_size": 0.5,
        # The number of corners the pattern has in the X and Y dimentions.
        # Note: The charuco detector uses the number of squares per dimension in
        # its detector. Internally we add a +1 to Y and X dimensions to account for
        # that. Therfore, the number of corners should be used even for the charuco
        # pattern.
        "dimension": {"x": 9, "y": 6},
        # The length of the a square edge.
        "size": 0.101,
        # The length of the charuco inner marker.
        "inner_size": 0.014
    },
```

miguelriemoliveira, 2020-05-27:
I will push a huge commit briefly ... don't touch this stuff for now ...

aaguiar96, 2020-05-27:
> This is the problem I think ... rviz is giving this error:
> 
> > ERROR] [1590587449.530670546]: Could not load resource [package://interactive_calibration/meshes/charuco.dae]: Unable to open file "package://interactive_calibration/meshes/charuco.dae".

Did my commit solved the problem?

miguelriemoliveira, 2020-05-27:
I think it did, but now I am adding the mesh_file parameter, so it is back not working again. It already finds the chessboard, but I am still having problems with the single_pattern flag. 

Will need more time ...

miguelriemoliveira, 2020-05-30:
Hi @aaguiar96 ,

we can close this one, right?

aaguiar96, 2020-05-30:
Yes @miguelriemoliveira, closing it!

closed 2020-05-30 by aaguiar96