lardemua/atom #151: Fix the first_agrob dataset

miguelriemoliveira, 2020-05-11:
Hi @aaguiar96 ,

we have a problem in the first agrob dataset: collection "1" does not have the pattern detected from the right camera, take a look at the json;

```
3224             {
3225               "id": 68,
3226               "x": 365.572937,
3227               "y": 394.874512
3228             },
3229             {
3230               "id": 69,
3231               "x": 335.898895,
3232               "y": 397.446625
3233             }
3234           ]
3235         },
3236         "right_camera": {                                                                                                                        
3237           "detected": false,
3238           "idxs": []
3239         },
3240         "vlp16": {
3241           "detected": true,
3242           "idxs": [
3243             10760,
3244             10768,
3245             10770,
3246             10776,
3247             10778,
3248             10779,
3249             10780,

```

one fix would be to manually find the chessboard points and add them to the json file, so we would have the minimum two collections for testing the optimization.

I am not sure the pixel valie should be the center of the arucos or the chessboard corners, @eupedrosa  which one is it?

Can you do it? You know how?

aaguiar96, 2020-05-11:
I @miguelriemoliveira 

Maybe it is better to try to record a new collection, don't you think?
I think I will have to do it anyway because I realized today that I need the `PointCloud2` message information to read the pcl properly using `ros_numpy`.

I'm now trying to add these fields to the collection, just like you did for `camera_info` topic.

miguelriemoliveira, 2020-05-11:
Hum, I think it is already there is it not? lines 84-132 o the data_collected.json  (for the case of collection '0', I think)

Take a look

[image]

aaguiar96, 2020-05-11:
Oh... Lost so many time on this!
Thanks @miguelriemoliveira, you saved me a lot of time.

Regarding the fix of the agrob dataset, I think I'll try to create a new dataset. If I can't, I'll follow your suggestion.

aaguiar96, 2020-05-11:
@miguelriemoliveira, the `Save Collection` works with `bag_rate` close to 1.

I'll work with what we have for now.
I think we can close this issue, right?

miguelriemoliveira, 2020-05-11:
Agree!
*  Miguel Riem de Oliveira                                             *
email:    mriem@ua.pt    |    m.riem.oliveira@gmail.com
office: 22.2.18                 |    ext: 23887
Dep. de Engenharia Mecânica, Universidade de Aveiro
*Campus Universitário de Santiago, 3810-193  Portugal*

closed 2020-05-11 by aaguiar96