lardemua/atom #152: Add comments to the config.json file

miguelriemoliveira, 2020-05-17:
Hi @eupedrosa and @aaguiar96 ,

I am writting the hand eye paper describing the configuration of our calibration procedures and came to the conclusion that a configuration file must have some comments to be more understandable.

The problem is, standard json does not allow for comments.  

I took a look around and apparently there are others which also feel the same and have actually come up with a solution:

http://web.archive.org/web/20100629021329/http://blog.getify.com/2010/06/json-comments/

which is to have comments // or /* ... */ and to remove them before giving them to the parser (to minify the json, they call it)

I implemented a test for the atlascar config json. Here it is:

```
{
    /* Specify all sensors that are used in the calibration: Each sensor should have:
        <sensor_name>: {"link": <The frame of the sensor's data. The value of the sensor's message.header.frame_id.>,
                        "parent_link": <Parent frame of the transformation to be calibrated.>,
                        "child_link" : <Child frame of the transformation to be calibrated.>,
                        "topic_name" : <Name of the ros message topic that contains this sensor's data.>
        }
    */
    "sensors": {
        "right_camera": {
            "link": "zed_right_camera_optical_frame",
            "parent_link": "zed_camera_center",
            "child_link" : "zed_right_camera_frame",
            "topic_name" : "/zed/zed_node/right/image"
        },

        "left_camera": {
            "link": "zed_left_camera_optical_frame",
            "parent_link": "zed_camera_center",
            "child_link" : "zed_left_camera_frame",
            "topic_name" : "/zed/zed_node/left/image"
        },

        "vlp16": {
           "link": "velodyne",
           "parent_link": "tower_link",
           "child_link": "vlp16_frame",
           "topic_name": "/velodyne_points"
        }
    },

    /* Parameterization of the calibration pattern.
       @Eurico, can you add here a description for each field?
    */

    "calibration_pattern" : {
        "link": "chessboard_link",
        "parent_link": "base_link",
        "origin": [-0.37, 0.0, 0.0, 0, 0.0, 3.8],
        "fixed": false,
        "pattern_type": "charuco",
        "dictionary": "DICT_4X4_50",
        "border_size": 0.12,
        "dimension": {"x": 10, "y": 7},
        "size": 0.029,
        "inner_size": 0.024
    },

    /* Miscellaneous information */
    "anchored_sensor": "right_camera",     // Anchor a given sensor, making it immovable during the optimization.
    "world_link": "base_link",             // Frame of reference for the optimization procedure. Must exist in the
                                           //transformation chains of all the sensors which are being calibrated.
    "max_duration_between_msgs": 10000000  // Max time delta between sensor data messages when creating a collection.
}
```

and tested the minification, it works great.

So, two questions:

1. Do you agree that we add this functionality?
2. If so, lets use this atlascar json as the standard. Eurico, can you give me a hand in commenting the calibration pattern section?
3. Once the atlascar2 json is set, we should propagate to the hand-eye as well as the agrob jsons.

What do you think?

eupedrosa, 2020-05-17:
>So, two questions:
>
>1. Do you agree that we add this functionality?
>2. If so, lets use this atlascar json as the standard. Eurico, can you give me a hand in commenting the calibration pattern section?
>3. Once the atlascar2 json is set, we should propagate to the hand-eye as well as the agrob jsons.

1. I agree with the need for comments in the configuration file. :+1:
2. Yes, I can give you a hand in commenting the calibration pattern section.
3. Yes, we should propagate, like a virus, to the other projects.
4. Its was three (3) questions, not two (2) :smile:

miguelriemoliveira, 2020-05-17:
Thanks. If I only have three (3) questions, then how come you give four (4) answers : - )?

miguelriemoliveira, 2020-05-26:
This is implemented. Thank for the help @eupedrosa .

closed 2020-05-26 by miguelriemoliveira