lardemua/atom #153: Test agrob bags from 20-5-2020

miguelriemoliveira, 2020-05-20:
Hi @aaguiar96 ,

I was really anxious to try out the new bag file. 

I ran but could not see the images. 

Did you change the topic names? Perhaps you need to fix the playback republish topics?

Thanks for the help,
Miguel

miguelriemoliveira, 2020-05-20:
BTW, I ran like this:

`roslaunch interactive_calibration agrob_calibration.launch bag:=/home/mike/bagfiles/agrob/agrob_calibration_2020-05-20-12-2/calibration_2020-05-20-12-25-58_0.bag`

aaguiar96, 2020-05-21:
Hi @miguelriemoliveira 

Yes, we had that problem with the topics last time.
I fixed the issue this time. I committed the changes to agrob and atlascar right now.

aaguiar96, 2020-05-21:
Hi @miguelriemoliveira and @eupedrosa 

Apart from the dimensions, that I can measure, are these parameters correct for the charuco that I used yesterday?

```
    "calibration_pattern" : {
        "link": "chessboard_link",
        "parent_link": "base_link",
        "origin": [-0.37, 0.0, 0.0, 0, 0.0, 3.8],
        "fixed": false,
        "pattern_type": "charuco",
        "dictionary": "DICT_5X5_100",
        "border_size": 0.03,
        "dimension": {"x": 11, "y": 8},
        "size": 0.06,
        "inner_size": 0.045
    }
```

I'm trying to create a dataset, but the pattern detections are unstable...

eupedrosa, 2020-05-21:
You have the "x" and "y" swapped...

eupedrosa, 2020-05-21:
@aaguiar96, did you used a ruler to measure the `size` and `inner_size`?

aaguiar96, 2020-05-21:
> @aaguiar96, did you used a ruler to measure the `size` and `inner_size`?

Yes, but I also saw from calib.io that the square size is 60mm.

aaguiar96, 2020-05-21:
@miguelriemoliveira,

Can you execute the system to see the detections?
Do you think that is good enough?

miguelriemoliveira, 2020-05-21:
The pattern is far away from the cameras. Next bag file if you can bring it closed. 
Also the illumination sometimes creates reflections which hamper the detection.

I tweked a bit with the aruco detector's parameters. Now we can get detect about 50% of the time.

aaguiar96, 2020-05-21:
Thanks @miguelriemoliveira 

Did you commit the changes?
I'll try to save a dataset.

miguelriemoliveira, 2020-05-21:
Yes,

4ea46400a2209b5a2436377ddabbfbe76e3b6b76

aaguiar96, 2020-05-21:
Hi @miguelriemoliveira and @eupedrosa 

I was able to record a dataset with 19 collections.
Check it out! :)

[agrob_second_dataset.zip](https://github.com/lardemua/AtlasCarCalibration/files/4662894/agrob_second_dataset.zip)

miguelriemoliveira, 2020-05-21:
Hi,

Great!

Can you tell me how many collections have all 3 sensors detected?

aaguiar96, 2020-05-21:
All of them, I think! :)

aaguiar96, 2020-05-21:
Yes, all you them!

miguelriemoliveira, 2020-05-21:
It looks great. Now you can really test the optimization.

If you need help with something let me know.

closed 2020-05-21 by miguelriemoliveira

eupedrosa, 2020-05-21:
@aaguiar96, why is the `data_collected.json` so large (22MB)?
Do you save the (complete) velodine data?

miguelriemoliveira, 2020-05-21:
Hi @eupedrosa ,

yes, by design we save the complete sensor data, besides de labels.

eupedrosa, 2020-05-21:
Should we follow the same strategy as the images? save on a separate file?

miguelriemoliveira, 2020-05-21:
We could. I thought of saving into pcl or ply. It would have to be a format
we could read by itself.

The bottomline was, nobody had time to work on it ...

Regards,
Miguel

*  Miguel Riem de Oliveira                                             *
email:    mriem@ua.pt    |    m.riem.oliveira@gmail.com
office: 22.2.18                 |    ext: 23887
Dep. de Engenharia Mecânica, Universidade de Aveiro
*Campus Universitário de Santiago, 3810-193  Portugal*

eupedrosa, 2020-05-21:
PLY is a good format. I like that idea.

aaguiar96, 2020-05-21:
> I volunteer @aaguiar96 ...

Ahahah, sure, I'll work on it (https://github.com/lardemua/AtlasCarCalibration/issues/155)

:D