lardemua/atom #154: Can't see the images running bag file eye_to_base_2020-02-05-10-47-48.bag

miguelriemoliveira, 2020-05-20:
Hi @eupedrosa ,

cannot see the images. I think the problem is that you are not decompressing, since I only see the compressed image topics.

Bag file is:
`eye_to_base_2020-02-05-10-47-48.bag`

rostopic list gives:

```
/astra/rgb/camera_info
/astra/rgb/image_raw/compressed
/clicked_point
/clock
/data_labeler/feedback
/data_labeler/update
/initialpose
/interactive_first_guess/feedback
/interactive_first_guess/update
/joint_states
/kinect/rgb/camera_info
/kinect/rgb/image_color/compressed
/move_base_simple/goal
/rosout
/rosout_agg
/tf
/tf_static

```

and I run with :

`roslaunch interactive_calibration eye_calibration.launch bagfile:=/home/mike/bagfiles/ur10/eye_to_base_2020-02-05-10-47-48.bag rviz:=true`

Perhaps the playback republish node has the wrong topic names?

eupedrosa, 2020-05-21:
Yeh, you need to add 
```xml
<node pkg="image_transport" type="republish" name="republish_kinect" output="screen"
args="compressed in:=/kinect/rgb/image_color raw out:=/kinect/rgb/image_color"/>

<node pkg="image_transport" type="republish" name="republish_astra" output="screen"
args="compressed in:=/astra/rgb/image_raw raw out:=/astra/rgb/image_raw"/>
```
to the launch file.

We need to improve this. For now, we have a specific launch file for each use-case. There must be a better way.

miguelriemoliveira, 2020-05-21:
I was trying to fix the launch files but there are several inconsistencies and I was not able to do it.

One problem is the files. You have a single launch file for all use cases, but the topic names change throughout the bag files. I am using:

both_eyes_2_2020-02-05-17-51-24.bag
eye_in_hand5_2020-01-23-18-19-36.bag
eye_to_base_2020-02-05-10-47-48.bag

Do you have anything more recent?

Can we talk tomorrow? We could establish a method for this. I could fix it while I am doing the videos ...

We can do a general launch file, but I think that means we have to impose a strucutre (file names), otherwise we will be using way to many command line arguments. Look at my command now to run the eye_to_base 

`roslaunch interactive_calibration eye_calibration.launch bagfile:=/home/mike/bagfiles/ur10/eye_to_base_2020-02-05-10-47-48.bag rviz:=true xacro_file:=/home/mike/catkin_ws/src/AtlasCarCalibration/interactive_calibration/calibrations/ur10e/eye_to_base/ur10e.urdf.xacro rviz_config:=/home/mike/catkin_ws/src/AtlasCarCalibration/interactive_calibration/calibrations/ur10e/eye_to_base/rviz.rviz xacro_file_first_guess:=/home/mike/catkin_ws/src/AtlasCarCalibration/interactive_calibration/calibrations/ur10e/eye_to_base/first_guess.urdf.xacro read_first_guess:=false`

miguelriemoliveira, 2020-05-21:
One other inconsistency I noticed was with the names of the sensors: we use astra, camera, world_camera all for the same thing. I suggest we use "hand_camera"

We can talk tomorrow.

miguelriemoliveira, 2020-05-22:
I just started a dedicated issue for a more detailed restructuring. We can close this one.

closed 2020-05-22 by miguelriemoliveira