lardemua/atom #155: Save 3D LiDAR data into .ply and read it on the optimization side

aaguiar96, 2020-05-21:


miguelriemoliveira, 2020-05-22:
Because @eupedrosa was so mean to you I will give a help : - )

https://github.com/miguelriemoliveira/OptimizationUtils/blob/a2055032709a80556670ae989228cdedb7cef87f/test/camera_pose_oc_dataset.py#L511-L599

miguelriemoliveira, 2020-06-12:
I will start working on this because now with velodyne data the datasets become very large, and very difficult to handle (for example we cannot show them in the online json viewer)

aaguiar96, 2020-06-12:
> @aaguiar96 , do you mind that I work on this?

Of course not.
Btw, we can see them in the json viewer, you just have to change to tree mode which is not set by default since the file is very large. In any case, I think this we should do this anyways.

miguelriemoliveira, 2020-07-18:
I was working on this since yesterday. HAve tried with both plyfile and pypcd. 

It turned out to be more difficult than expected. I am abandoning this for now.

aaguiar96, 2020-07-28:
Reopening issue since **i think** I can do it with [python_pcd](https://github.com/cburbridge/python_pcd/blob/master/src/python_pcd/io.py)

@miguelriemoliveira what is the data that I can exclude from the json? The pointcloud data, but also the labelled idxs? Or only the pcl data?

miguelriemoliveira, 2020-07-28:
Hi @aaguiar96 ,

only the pcl data. The idxs can stay put.

aaguiar96, 2020-07-28:
Implemented! @miguelriemoliveira I'll create a pull request ok?

Check the code first to see if the things are where they should be.
I had to make a correction to the original code of python_pcd, but it works.

The data_collector write the pcd files and the calibrate loads them.

miguelriemoliveira, 2020-07-28:
Thats great! And it works only for xyz point clouds?

If so we should have a check in the collection for that ...

*  Miguel Riem de Oliveira                                             *
email:    mriem@ua.pt    |    m.riem.oliveira@gmail.com
office: 22.2.18                 |    ext: 23887
Dep. de Engenharia Mecânica, Universidade de Aveiro
*Campus Universitário de Santiago, 3810-193  Portugal*

aaguiar96, 2020-07-28:
> Thats great! And it works only for xyz point clouds?

Don't know yet. I'll check.

Know something is going wrong when saving the calibration file.
If I comment this, the calibration ends with success.

aaguiar96, 2020-07-28:
> Know something is going wrong when saving the calibration file.

It prints the velodyne data and an error message:

```
(...)
192, 64, 155, 6, 255, 63, 4, 0, 0, 0, 0, 0, 128, 66, 15, 0, 0, 0, 112, 92, 194, 212, 127, 127, 0, 0]
is_dense: False is not JSON serializable
```

aaguiar96, 2020-07-28:
Here is a dataset with 15 collections I took if you want to test only the calibration module.

data_collection.json -> 863KB

[dataset_pcd_test.zip](https://github.com/lardemua/atom/files/4990006/dataset_pcd_test.zip)

miguelriemoliveira, 2020-07-31:
Hi @aaguiar96 ,

I think you code in calibrate breaks the usage of older datasets.

```
rosrun atom_core calibrate -json /home/mike/datasets/agrob/agrob_01_07_2020/data_collected.json -csf 'lambda x: int(x) < 66' -ssf 'lambda name: name in ["left_camera","right_camera", "vlp16"]' -oi -rv -si -sr 0.3

Deleted collections: []
Removing collection 22 -> pattern was not found in sensor left_camera (must be found in all sensors).
Deleted sensors: []
Loaded dataset containing 3 sensors and 55 collections.
Traceback (most recent call last):
  File "/home/mike/catkin_ws/src/calibration/atom/atom_core/scripts/calibrate", line 669, in <module>
    main()
  File "/home/mike/catkin_ws/src/calibration/atom/atom_core/scripts/calibrate", line 273, in main
    filename = os.path.dirname(json_file) + '/' + collection['data'][sensor_key]['data_file']
KeyError: 'data_file'
```

We can put an if this key exists, load, if not, assume the point cloud data is there already, right? Can you do it now or should I?

eupedrosa, 2020-07-31:
> We can put an if this key exists, load, if not, assume the point cloud data is there already, right? Can you do it now or should I?

I think it is a little bit more than that, I just did that and it does not work.

aaguiar96, 2020-07-31:
I'm on it!

miguelriemoliveira, 2020-07-31:
It should ... Lets see what @aaguiar96 says ...

aaguiar96, 2020-07-31:
Fixed. Just added this:

```
                if 'data_file' in collection['data'][sensor_key].keys():
                    # Read pointcloud from pcd file
                    filename = os.path.dirname(json_file) + '/' + collection['data'][sensor_key]['data_file']
                    frame_id = collection['data'][sensor_key]['header']['frame_id']
                    cloud_msg = read_pcd(filename, frame_id, get_tf=False)
                    collection['data'][sensor_key]['data'] = cloud_msg
                else:
                    from rospy_message_converter import message_converter
                    cloud_msg = message_converter.convert_dictionary_to_ros_message("sensor_msgs/PointCloud2",
                                                                                    collection['data'][sensor_key])
                    collection['data'][sensor_key]['data'] = cloud_msg
```

eupedrosa, 2020-07-31:
You can use `collection['data'][sensor_key].has_key("data_file")`

aaguiar96, 2020-07-31:
@eupedrosa and @miguelriemoliveira can you try both versions now?

> You can use `collection['data'][sensor_key].has_key("data_file")`

Ok @eupedrosa I'll change on a next commit.

miguelriemoliveira, 2020-07-31:
That 

`key in dictionary.keys() 
`
versus

`dictionary.has_key(key)
`
is what was taking so long in the optimization utils. It was a large dictionary and the first method takes a long time. It makes sense if we think about it. The second method uses the hash key to quickly locate the key-value in the dictionary, but the second method first converts the dictionary keys to a list and then searches for the key in that list, which means it has to compare against all the elements in the list.

aaguiar96, 2020-07-31:
> is what was taking so long in the optimization utils. It was a large dictionary and the first method takes a long time. It makes sense if we think about it. The second method uses the hash key to [...]

Makes sense. Fixed!

eupedrosa, 2020-07-31:
@aaguiar96, It now fails when writing the final json file. Just remove the line `collection['data'][sensor_key]['data'] = cloud_msg`

aaguiar96, 2020-07-31:
> Just remove the line `collection['data'][sensor_key]['data'] = cloud_msg` from the `else` block.

Hmm... I use that in visualization. It breaks because of that?

miguelriemoliveira, 2020-07-31:
Hi @aaguiar96 ,

another comment:

You should do this loading of the point cloud immediately after loading the json, not when you're about to label the point clouds. 

The place to do it is this:

https://github.com/lardemua/atom/blob/40e692d1ca85b40e8f516ff116102e1b4a1d374a/atom_core/scripts/calibrate#L166-L190

Notice we are doing it also there for the camera images.

aaguiar96, 2020-07-31:
> @aaguiar96, It now fails when writing the final json file. Just remove the line `collection['data'][sensor_key]['data'] = cloud_msg` from the `else` block.

@miguelriemoliveira I already told you that it was breaking when saving the json. Now it breaks also with the old dataset so, it might be some error that I introduced... 

@eupedrosa says this: `collection['data'][sensor_key]['data'] = cloud_msg` is the problem, but it is exactly what you do for cameras right?

miguelriemoliveira, 2020-07-31:
Actually, even better, you should have a function to load and another to save the json file.

**The function to load should:**

check for data_file fields and they exist load the respective file into the data field.

**The function to write should:** 

check for the data field and, if it exists, save a new image or point cloud 
Do not save if the field data_file exists and points to an image or point cloud that is in the filesystem already
Then, delete the field data

After, it should

miguelriemoliveira, 2020-07-31:
> @miguelriemoliveira I have told you before that it was breaking when saving the json. Now it breaks also with the old dataset so, it might be some error that I introduced...
> 
> @eupedrosa says this: collection['data'][sensor_key]['data'] = cloud_msg is the problem, but it is exactly what you do for cameras right?

Yes, it could be that the cloud_msg is a np array or contains one inside. In any case, if you do as I said above (in particular point 3. ) you wont have this problem when saving.

aaguiar96, 2020-07-31:
> > @miguelriemoliveira I have told you before that it was breaking when saving the json. Now it breaks also with the old dataset so, it might be some error that I introduced...
> > @eupedrosa says this: collection['data'][sensor_key]['data'] = cloud_msg is the problem, but it is exactly what you do for cameras right?
> 
> Yes, it could be that the cloud_msg is a np array or contains one inside. In any case, if you do as I said above (in particular point 3. ) you wont have this problem when saving, because you won't save the **data** field..

I do not understand this...

eupedrosa, 2020-07-31:
>>check for the data field and, if it exists, save a new image or point cloud
>>Do not save if the field data_file exists and points to an image or point cloud that is in the filesystem already

>I do not understand this...

What @miguelriemoliveira  is saying is that we should process the data before writing the json file. This way we prevent unecessary data to be written to file.

I guess the *loading* and *saving* of the json file is getting bigger. Maybe it is time to move it to the utilities? or other place?

aaguiar96, 2020-07-31:
> What @miguelriemoliveira is saying is that we should process the data before writing the json file. This way we prevent unecessary data to be written to file.

Thanks.

I'm dealing with a strange bug. When I do this:
```
cloud_msg = message_converter.convert_dictionary_to_ros_message("sensor_msgs/PointCloud2", collection['data'][sensor_key])
```
at the beginning (while reading data from json) it works. The exact same line when I'm writing to the json gives:
```
TypeError: 'PointCloud2' object is not iterable
```

Any idea why is this happening?

miguelriemoliveira, 2020-07-31:
Hi @aaguiar96 ,

I will call you in 30 mins or 1 hour.

eupedrosa, 2020-07-31:
Yes, it is an object of type 'PointCloud2' that does have an __iter__() function.

miguelriemoliveira, 2020-07-31:
Hi @aaguiar96 ,

but I am confused ... when you write should you not do the inverse operation, i.e. 

convert_ros_message_to_dictionary

instead of 

convert_dictionary_to_ros_message

Perhaps I am misunderstanding ...

aaguiar96, 2020-07-31:
> Yes, it is an object of type 'PointCloud2' that does have an **iter**() function.

But it does not happen anymore with the exact same arguments...

> convert_ros_message_to_dictionary
> 
> instead of
> 
> convert_dictionary_to_ros_message
> 
> Perhaps I am misunderstanding ...

No. I use it while reading the json file if the data is saved in the json. To write the json file I have to use it to write to the pcd file, since the function that writes uses a ROS message and convert it to pcd.

aaguiar96, 2020-07-31:
Hi @miguelriemoliveira  and @eupedrosa 

I solved it. It was a rookie mistake, I was confused about the variable types.
Now everything is working.

* Read dataset without pcd files 
* Read dataset with pcd files
* Write dataset that did not have pcd files, removing the data field from json and creating a pcd file for each tested collection
* Write dataset that already had pcd files

Can you also test please?

eupedrosa, 2020-07-31:
It is working for me.

miguelriemoliveira, 2020-07-31:
Hi all,

I confirm its working. I used data collector to save, and calibrate to read. It works.

Could not test for an rgbd point cloud because I don't have a bag file with those. But I think it will work. 

Great work @aaguiar96, closing issue.

miguelriemoliveira, 2020-08-01:
I will organize this into function to put in utilities as @eupedrosa suggested.

miguelriemoliveira, 2020-08-02:
Hi @aaguiar96 .

This took longer than expected. I did a significant overhaul from what we had before, based on your code.
The goal was to have a function for loading and another for saving the results json, and that all scripts used these functions. What I did:

1. a loadResults json and a save results json. For now they are in utilities but I will suggest some changes in a proposal of better organization.

2. Checked and the pipeline of saving -> loading -> saving was not turning out the same results. The reason was that the data field sometimes was the result of converting from ros message to dictionary, some others from saving a pcd or image file.  Now I implemented a strict conversion pipeline, for images file -> cv_image -> ros_image_msg -> dicitonary, for point clouds file -> point_cloud_msg -> dictionary. 

3. Tested and we can save, load, save again, etc. Always the same result. Some additional conversions had to be made, but I tried to cache those for faster processing.

4. Used the same functions in calibrate (load and save) as well as in collect_and_label (save)

5. Missing the usage of these functions in script range_sensor_to_camera_evaluation.py @aaguiar96 , can you do this please?

closed 2020-08-02 by miguelriemoliveira