lardemua/atom #156: Reorganization of the calibration files for a robotic system

miguelriemoliveira, 2020-05-22:
Hi @eupedrosa and @aaguiar96 ,

About this, I thought about it over sleep and I have a proposal. Let me know if it makes sense.

Here's a list of the files required for each calibration (names sometimes vary between hand eye, atlascar2 and agrob, we should fix it):

- a bag file
- config.json 
- robot_description.urdf.xacro
- first_guess.urdf.xacro
- optimized.urdf.xacro
- rviz.rviz
- view_optimization.rviz
- robot_calibration.launch
- robot_view_optimization.launch
- ... I think I am not missing anything

And we 3 stages of calibration: 

1. First guess:
2. Data collection and labelling
3. Optimization

I like the idea of @eupedrosa of having a single **robot_calibration.launch**  file that starts up the calibration is great, but there are some things which are specific for each robotic system, namely the part about the bag file playback, including the republish decompressing stuff. The reason is that each bag has different topic names, and may or may not require decompression of images. Here's and example:

```xml
<launch>

    <!-- %%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%-->
    <!-- Parameters-->
    <!-- %%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%-->
<!--    <arg name="bag" default="$(find interactive_calibration)/calibrations/atlascar2/27-05-2019_with_chessboard.bag"/>-->
<!--    <arg name="bag" default="$(find interactive_calibration)/calibrations/atlascar2/calibration_31_05_2019_fixed.bag"/>-->
<!--    <arg name="bag" default="/tmp/bag_file_2019-09-24-01-13-19.bag"/>-->
    <arg name="read_first_guess" default="false"/>
    <arg name="read_optimized" default="false"/>
<!--    <arg name="bag" default="$(env HOME)/bagfiles/bag_file_2019-09-24-01-20-41.bag"/>-->
    <arg name="bag" default="$(env HOME)/bagfiles/calibration_5_dez_2019.bag"/>
    <arg name="bag_start" default="0"/>
    <arg name="xacro_file" default="$(find atlascar2_description)/urdf/atlascar2.urdf.xacro"/>
    <arg name="xacro_file_first_guess" default="$(find interactive_calibration)/calibrations/atlascar2/first_guess.urdf.xacro"/>
    <arg name="xacro_file_optimized" default="$(find interactive_calibration)/calibrations/atlascar2/optimized.urdf.xacro"/>
    <arg name="rviz_config_file" default="$(find interactive_calibration)/calibrations/atlascar2/calibration.rviz"/>

    <!-- %%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%-->
    <!-- Load robot description file based on whether or not we are reading the first guess or optimized xacro file.-->
    <!-- %%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%-->
    <group if="$(arg read_optimized)">
        <param name="robot_description" command="$(find xacro)/xacro $(arg xacro_file_optimized)"/>
    </group>

    <group unless="$(arg read_optimized)">
        <group if="$(arg read_first_guess)">
            <param name="robot_description" command="$(find xacro)/xacro $(arg xacro_file_first_guess)"/>
        </group>

        <group unless="$(arg read_first_guess)">
            <param name="robot_description" command="$(find xacro)/xacro $(arg xacro_file)"/>
        </group>
    </group>

    <!-- %%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%-->
    <!-- Robot and joint state publisher for generating tfs -->
    <!-- %%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%-->
    <node name="robot_state_publisher" pkg="robot_state_publisher" type="state_publisher"/>

    <node name="joint_state_publisher" pkg="joint_state_publisher" type="joint_state_publisher">
        <param name="use_gui" value="true"/>
    </node>

    <!-- %%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%-->
    <!-- Visualization -->
    <!-- %%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%-->
    <node name="rviz" pkg="rviz" type="rviz" args="-d $(arg rviz_config_file)" required="false"/>

    <!-- %%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%-->
    <!-- Playing back bag files -->
    <!-- %%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%-->
    <!--To put together the ros bag and the robot state publisher one must use sim_time=true and &#45;&#45;clock-->
    <param name="/use_sim_time" value="true"/>
    <!--    <node pkg="rosbag" type="play" name="rosbag_play" output="screen" args="$(arg bag) -r .2 -l /tf:=/tf_dev_null &#45;&#45;clock"/>-->
    <!--<node pkg="rosbag" type="play" name="rosbag_play" output="screen" args="$(arg bag) -r 1 -l &#45;&#45;clock"/>-->
    <include file="$(find atlascar2_bringup)/launch/playback_sensor_data.launch">
        <arg name="bag" value="$(arg bag)" />
        <arg name="bag_start" value="$(arg bag_start)" />
    </include>
</launch>
```

Everything else could be generalized. My suggestion is, as is done in this example, to define a playback.launch file (also inside the calibration folder) which is responsible for playing back the bag files of each specific robotic system.

The visualization part (rviz config files) cannot be generalized for now I think, the only way would be the development of an rviz plugin (@eupedrosa , I intend to propose an M.Sc. thesis for next year on this alone, you want to co-supervise?).

If we put all these files with standard names in a folder specific for each robotic system, we can then create a general calibration launch file which only needs to get as input the folder  and then self configures.

What do you think?

miguelriemoliveira, 2020-05-22:
BTW, related ToDo issues: #60, #152, #100, #125, #145,

eupedrosa, 2020-05-22:
Do you know how this kind of problems is solved in ROS 2 ???
Launch files are python scripts!

Maybe, we can do something similar. Create a script that generates launch files that depens on some configuration. Programming is a tool, lets use it!!!

eupedrosa, 2020-05-22:
>The visualization part (rviz config files) cannot be generalized for now I think, the only way would be the development of an rviz plugin (@eupedrosa , I intend to propose an M.Sc. thesis for next year on this alone, you want to co-supervise?).

Yes, I do.

miguelriemoliveira, 2020-05-22:
> > The visualization part (rviz config files) cannot be generalized for now I think, the only way would be the development of an rviz plugin (@eupedrosa , I intend to propose an M.Sc. thesis for next year on this alone, you want to co-supervise?).

> 
> Yes, I do.

Great. Thanks. Will contact  you in a few months.

miguelriemoliveira, 2020-05-22:
Concerning the python script (I like the idea): 

Do you mean a python script which launches the ros nodes, or do you mean a python script that generates a launch file?

Can we talk a bit in the afternoon?

eupedrosa, 2020-05-22:
A script that generates a launch file. Managing the nodes by ourselfs is not a good idea.

Yes, we can talk in the afternoon. Can it be after 16h00 ? I have a meeting before that.

miguelriemoliveira, 2020-05-22:
Hi  Eurico,

16h ok.

See you,
Miguel

*  Miguel Riem de Oliveira                                             *
email:    mriem@ua.pt    |    m.riem.oliveira@gmail.com
office: 22.2.18                 |    ext: 23887
Dep. de Engenharia Mecânica, Universidade de Aveiro
*Campus Universitário de Santiago, 3810-193  Portugal*

miguelriemoliveira, 2020-05-22:
Ok, so after the discussion the objective was to create a script with create a <robotic system>calibration ros package, with all the necessary files. This will need to have two scripts.

miguelriemoliveira, 2020-05-23:
Ok, so the first script for the generation of the ros calibration package is working. 

@eupedrosa , can you try?

rosrun interactive_calibration create_calibration_ros_package.py -n /home/mike/catkin_ws/src/test_calibration_pkg

Also, I need you to finish #152 in the template config.json (we will derive all from there). It is here

https://github.com/lardemua/AtlasCarCalibration/blob/master/interactive_calibration/templates/config.json

miguelriemoliveira, 2020-05-23:
Tomorrow if I have time I will start with the script for creating the launch files from the config.json.

eupedrosa, 2020-05-23:
Hi @miguelriemoliveira, I have a suggestion to make.

After openening the json file to comment, something did not feel right. I felt the same by reading the file on GitHub and on my computer using VIM. 

Here is an example:
```json
{
  /* Miscellaneous information */
    "anchored_sensor": "right_camera",     // Anchor a given sensor, making it immovable during the optimization.
    "world_link": "base_link",             // Frame of reference for the optimization procedure. Must exist in the
                                           //transformation chains of all the sensors which are being calibrated.
    "max_duration_between_msgs": 10000000  // Max time delta between sensor data messages when creating a collection.
}
```

Do you see how ugly it looks? Comments are not supported in JSON. And no syntax highlight will ever support it. Let me summarize the arguments against comments in JSON.

* Not supported by the original JSON spec.
* Syntax highligh makes it look ugly.
* Add an extra layer of abstraction to load the file
* JSON files is not common in ROS for configurations files.

That said, my proposal is to change to yaml. Here is the same example in yaml:
```yaml
{
    # YAML uses # for comments
    # Miscellaneous information 
    "anchored_sensor": "right_camera",     # Anchor a given sensor, making it immovable during the optimization.
    "world_link": "base_link",             # Frame of reference for the optimization procedure. Must exist in the
                                           # transformation chains of all the sensors which are being calibrated.
    "max_duration_between_msgs": 10000000  # Max time delta between sensor data messages when creating a collection.
}
```

In our code, we only have to change
```python
# Assuming there is no comments
data = json.load(fileobj)
```
to
```python
data = yaml.load(fileobj, Loader=yaml.CLoader)
```
Both code snippets create the same python dictionary. The `config.json` will now be `config.yml`. The content is the same, we only have to use \# as comments. This way we be inline with ROS that uses YAML files for configuration.

What do you think @miguelriemoliveira ??

miguelriemoliveira, 2020-05-23:
I agree. Comments in json were solved functionaly with the minifyer but the looks of it was terrible. 

Lets change to yaml. You say everything is the same. Lets try. Can you create a template yaml and put it in templates?

eupedrosa, 2020-05-23:
Yes, I can create the template.

miguelriemoliveira, 2020-05-23:
Great. Once you have it let me know and I will work on the script to create the launch files.

eupedrosa, 2020-05-23:
Done, check the file. The name is `config.yml` and is in the `template` folder.

miguelriemoliveira, 2020-05-23:
Looks great. For now lets use the jsons. I will test the yml config files in the new script I am doing to create the launch files. Then we talk about changing all scripts to support yml.

I am having a lot of fun : - )

miguelriemoliveira, 2020-05-23:
Seems to be working great. Will continue. And the paper? : -)

eupedrosa, 2020-05-23:
What paper?

miguelriemoliveira, 2020-05-23:
The one you should be writing incessantly right now ...

*  Miguel Riem de Oliveira                                             *
email:    mriem@ua.pt    |    m.riem.oliveira@gmail.com
office: 22.2.18                 |    ext: 23887
Dep. de Engenharia Mecânica, Universidade de Aveiro
*Campus Universitário de Santiago, 3810-193  Portugal*

miguelriemoliveira, 2020-05-23:
Hi @eupedrosa  and @aaguiar96 ,

Just ended a functional **configure_calibration_pkg** .

It works like this:

1. Create a calibration package:

`rosrun interactive_calibration create_calibration_pkg.py -n ur10e_eye_to_base`

2. Edit the config.yml for the robotic system you are working on.

3. Configure the calibration package

`rosrun interactive_calibration configure_calibration_pkg.py -n ur10e_eye_to_base -b ~/bagfiles/ur10/eye_to_base_2020-02-05-10-47-48.bag -d ~/catkin_ws/src/calibration/AtlasCarCalibration/interactive_calibration/calibrations/ur10e/eye_to_base/ur10e.urdf.xacro`

which generates a launch file (for now just the first guess one, called it bringup_dg.launch)

```xml
<launch>
<!-- This file was generated automatically by the script configure_calibration_pkg.py on the 23/05/2020 16:37:59-->

<!-- %%%%%%%%%%%%% ATOMIC Framework %%%%%%%%%%%%%%%%%%%%%-->
<!-- Runs bringup first guess for initial parameter setting through rviz interactive markers-->

<!-- %%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%-->
<!-- Parameters-->
<!-- %%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%-->
<!-- bag_file: you can give another bagfile to the launch file using this arg-->
<arg name="bag_file" default="/home/mike/bagfiles/ur10/eye_to_base_2020-02-05-10-47-48.bag"/>
<!-- bag_start: starting time for playing back the bag file. Check "rosbag play -h"-->
<arg name="bag_start" default="0"/>
<arg name ="bag_rate" default ="1" />
<arg name="description_file" default="$(find ur10e_eye_to_base)/urdf/description.urdf.xacro"/>
<arg name="rviz_file" default="$(find ur10e_eye_to_base)/rviz/config.rviz"/>

<!-- %%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%-->
<!-- Load robot description and tf generators -->
<!-- %%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%-->
<param name = "robot_description" command = "$(find xacro)/xacro $(arg description_file)" />
<node name="robot_state_publisher" pkg="robot_state_publisher" type="state_publisher"/>

<!-- %%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%-->
<!-- Playback the bag file -->
<!-- %%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%-->
<param name="/use_sim_time" value="true"/>
<node pkg="rosbag" type="play" name="rosbag_play" output="screen" args=" $(arg bag_file) --clock -r $(arg bag_rate) -l -s $(arg bag_start) /tf:=/tf_dev_null /tf_static:=/tf_static_dev_null"/>

<node pkg="image_transport" type="republish" name="republish_base_camera" output="screen" args="compressed in:=/kinect/rgb/image_color raw out:=/kinect/rgb/image_color"/>

<!-- %%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%-->
<!-- Visualization -->
<!-- %%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%-->
<node name="rviz" pkg="rviz" type="rviz" args="-d $(arg rviz_file)" required="false"/>

</launch>
```

Looks good no?  You can run it to test:

`roslaunch ur10e_eye_to_base bringup_fg.launch`

The missing part is the automatic creation of an rviz configuration ... not sure if I should jump into it. Take a look at an rviz file:

```rviz
Panels:
  - Class: rviz/Displays
    Help Height: 78
    Name: Displays
    Property Tree Widget:
      Expanded:
        - /Global Options1
        - /Status1
      Splitter Ratio: 0.5
    Tree Height: 540
  - Class: rviz/Selection
    Name: Selection
  - Class: rviz/Tool Properties
    Expanded:
      - /2D Pose Estimate1
      - /2D Nav Goal1
      - /Publish Point1
    Name: Tool Properties
    Splitter Ratio: 0.5886790156364441
  - Class: rviz/Views
    Expanded:
      - /Current View1
    Name: Views
    Splitter Ratio: 0.5
  - Class: rviz/Time
    Experimental: false
    Name: Time
    SyncMode: 0
    SyncSource: ""
Preferences:
  PromptSaveOnExit: true
Toolbars:
  toolButtonStyle: 2
Visualization Manager:
```rviz

it would take some time to do ... and the rviz plugin we wanted to do could deprecate this entirely ...

Please test and give me our opinion on this and the rest.

eupedrosa, 2020-05-23:
Ok, I'm gonna test it first.

eupedrosa, 2020-05-23:
I was able to generate the launch files. It launched but I got this error:
```
[ERROR] [1590259585.266834676, 1580899674.703422070]: Robot semantic description not found. Did you forget to define or remap '/robot_description_semantic'?
```

Furthermore, I have TF and robot description but they do not match.

[image]

This should be working right? Something is missing.

miguelriemoliveira, 2020-05-23:
Strange ... I will check ...

miguelriemoliveira, 2020-05-23:
It works perfect for me. Can you try to use my calibration package? 

https://github.com/lardemua/AtlasCarCalibration/tree/master/ur10e_eye_to_base_calibration

Just try to run the configure (your update idea)

`rosrun ur10e_eye_to_base_calibration configure`

and then you can use the launch file

`roslaunch ur10e_eye_to_base bringup_fg.launch`

does it work? Same error?

eupedrosa, 2020-05-24:
Hi @miguelriemoliveira, it was not a smooth experience. The `launch` and `urdf` directory were missing and the `update` script failed with silence. After having a proper `config.yml` it worked.

There is a minor issue in the launch file:
```
[ WARN] [1590325307.124026545]: The 'state_publisher' executable is deprecated. Please use 'robot_state_publisher' instead
```
Just use `robot_state_publisher` instead of `state_publisher`. The latter may not work in ROS Noetic.

miguelriemoliveira, 2020-05-24:
hi @eupedrosa ,

uff, sorry. But at least it works! The silence of the configure script also occurred if you used directly the 

rosrun interactive_calibration configure_calibration_pkg -n ...

?

Will change to state_publisher.

eupedrosa, 2020-05-24:
> The silence of the configure script also occurred if you used directly the
rosrun interactive_calibration configure_calibration_pkg -n ...

Yes.

> Will change to state_publisher.

You have to change to `robot_state_publisher`.

miguelriemoliveira, 2020-05-24:
Perhaps we can talk tonight? And try to see what goes wrong with your code?

to robot_state_publisher, right.

Miguel
*  Miguel Riem de Oliveira                                             *
email:    mriem@ua.pt    |    m.riem.oliveira@gmail.com
office: 22.2.18                 |    ext: 23887
Dep. de Engenharia Mecânica, Universidade de Aveiro
*Campus Universitário de Santiago, 3810-193  Portugal*

eupedrosa, 2020-05-24:
Yes, we can talk tonight. Just give me a call when you are available.

miguelriemoliveira, 2020-05-24:
Ok, great. BTW, I think I will have good news about rviz. The .rviz is in
yaml, so its easy to parse and add stuff.

Miguel
*  Miguel Riem de Oliveira                                             *
email:    mriem@ua.pt    |    m.riem.oliveira@gmail.com
office: 22.2.18                 |    ext: 23887
Dep. de Engenharia Mecânica, Universidade de Aveiro
*Campus Universitário de Santiago, 3810-193  Portugal*

aaguiar96, 2020-05-27:
Hi @miguelriemoliveira and @eupedrosa 

I'm trying to generate the agrob package.
I'm getting this error:
```
Verified package path in /home/andre/Source/catkin_ws/src/AtlasCarCalibration/agrob
Executing command: ln -fs /home/andre/Source/catkin_ws/src/agrob/agrob_description/urdf/agrob.urdf.xacro /home/andre/Source/catkin_ws/src/AtlasCarCalibration/agrob/urdf/description.urdf.xacro
Executing command: xacro /home/andre/Source/catkin_ws/src/AtlasCarCalibration/agrob/urdf/description.urdf.xacro -o /tmp/description.urdf
warning: inconsistent namespace redefinitions for xmlns:xacro:
 old: http://ros.org/wiki/xacro
 new: http://www.ros.org/wiki/xacro (/opt/ros/melodic/share/lms1xx/urdf/sick_lms1xx.urdf.xacro)
warning: inconsistent namespace redefinitions for xmlns:xacro:
 old: http://www.ros.org/wiki/xacro
 new: http://ros.org/wiki/xacro (/home/andre/Source/catkin_ws/src/husky/husky_description/urdf/husky.urdf.xacro)

Unknown attribute "type" in /robot[@name='agrob']/transmission[@name='front_left_wheel_trans']
Unknown attribute "type" in /robot[@name='agrob']/transmission[@name='front_right_wheel_trans']
Unknown attribute "type" in /robot[@name='agrob']/transmission[@name='rear_left_wheel_trans']
Unknown attribute "type" in /robot[@name='agrob']/transmission[@name='rear_right_wheel_trans']
Topic /zed/zed_node/left/image is in compressed format (/zed/zed_node/left/image/compressed). Will setup a decompressor.
Topic /zed/zed_node/right/image is in compressed format (/zed/zed_node/right/image/compressed). Will setup a decompressor.
Setting up launch files ...
/zed/zed_node/left/image
/zed/zed_node/right/image
Traceback (most recent call last):
  File "/home/andre/Source/catkin_ws/src/AtlasCarCalibration/interactive_calibration/scripts/configure_calibration_pkg.py", line 274, in <module>
    rviz = yaml.load(open(rviz_file_template), Loader=yaml.FullLoader)
AttributeError: 'module' object has no attribute 'FullLoader'
```

Did you got it as well?

eupedrosa, 2020-05-27:
Beta software....
By the way @miguelriemoliveira, maybe you should use
```python
rviz = yaml.load(open(rviz_file_template), Loader=yaml.SafeLoader)
```
The yaml spec allows the execution of arbitrary code, see https://github.com/yaml/pyyaml/wiki/PyYAML-yaml.load(input)-Deprecation

miguelriemoliveira, 2020-05-27:
It was your fault : - ) Check above:

> In our code, we only have to change
> 
> # Assuming there is no comments
> data = json.load(fileobj)
> to
> 
> data = yaml.load(fileobj, Loader=yaml.CLoader)
> Both code snippets create the same python dictionary. The config.json will now be config.yml. The content is the same, we only have to use # as comments. This way we be inline with ROS that uses YAML files for configuration.
> 
> What do you think @miguelriemoliveira ??

I will change.

eupedrosa, 2020-05-27:
And now it is _my_ **fault** that we are also improving the code.

miguelriemoliveira, 2020-05-27:
@aaguiar96 , you are using beta functionalities that I am still developing. 

It is good that you are using the configure script already, but then perhaps it is better that I guide you through the usage first. This is a big change from what we had before, and I was first trying to develop a full version before telling you both to try it out ...

Lets meet this afternoon and I help you ok? 15h?

eupedrosa, 2020-05-27:
For me, 15h00 is OK, but I have to leave around 16h00.

miguelriemoliveira, 2020-05-27:
> And now it is my fault that we are also improving the code.

Ok, ok : - )

See you at 15h.

aaguiar96, 2020-05-27:
I have a meeting at 14h30. I'll join you as soon as it ends.

aaguiar96, 2020-05-27:
My meeting was postponed to 15h00... Can we meet at 15h30? If not, I'll join you when my meeting ends.

miguelriemoliveira, 2020-05-27:
15h30 ok by me.

Eurico, 15h30?

*  Miguel Riem de Oliveira                                             *
email:    mriem@ua.pt    |    m.riem.oliveira@gmail.com
office: 22.2.18                 |    ext: 23887
Dep. de Engenharia Mecânica, Universidade de Aveiro
*Campus Universitário de Santiago, 3810-193  Portugal*

eupedrosa, 2020-05-27:
Ok, 15h30.

miguelriemoliveira, 2020-05-28:
These three are working for any robotic system, provided the config.yml is well configured.

```
roslaunch atlascar2_calibration playbag.launch
roslaunch atlascar2_calibration set_initial_estimate.launch
roslaunch atlascar2_calibration collect_data.launch
```

@eupedrosa , @aaguiar96 can you test for eye_to_base and agrob respectively?

eupedrosa, 2020-05-28:
I alredy have 2 breaking issues to report:

1. The script `create_calibration_pkg.py` does not check if `-n pkg-name` already exist. It will overwrite existing files.

2. It does not respect the selected output directory. For some reason it tries to create directories under `AtlasCarCalibration`.

### Output of the script
```
Creating package on current directory.
Creating package ur10e_eye_to_base_calibration on path /home/melodic-dev
Executing command: cd /home/melodic-dev && catkin_create_pkg ur10e_eye_to_base_calibration rospy -a "Miguel Riem Oliveira" -m "Miguel Riem Oliveira" -D "This package was created automatically by the create_calibration_pkg.py script. Check the ATOMIC framework webpage for more information."
Created file ur10e_eye_to_base_calibration/package.xml
Created file ur10e_eye_to_base_calibration/CMakeLists.txt
Created folder ur10e_eye_to_base_calibration/src
Successfully created files in /home/melodic-dev/ur10e_eye_to_base_calibration. Please adjust the values in package.xml.
Executing command: rospack profile && rospack profile && rospack profile
Full tree crawl took 0.014562 seconds.
Directories marked with (*) contain no manifest.  You may
want to delete these directories.
To get just of list of directories without manifests,
re-run the profile with --zombie-only
-------------------------------------------------------------
0.013574   /opt/ros/melodic/share
0.000062 * /opt/ros/melodic/share/doc
0.000011 * /opt/ros/melodic/share/doc/liborocos-kdl
Full tree crawl took 0.014099 seconds.
Directories marked with (*) contain no manifest.  You may
want to delete these directories.
To get just of list of directories without manifests,
re-run the profile with --zombie-only
-------------------------------------------------------------
0.013204   /opt/ros/melodic/share
0.000061 * /opt/ros/melodic/share/doc
0.000010 * /opt/ros/melodic/share/doc/liborocos-kdl
Full tree crawl took 0.013660 seconds.
Directories marked with (*) contain no manifest.  You may
want to delete these directories.
To get just of list of directories without manifests,
re-run the profile with --zombie-only
-------------------------------------------------------------
0.012769   /opt/ros/melodic/share
0.000063 * /opt/ros/melodic/share/doc
0.000010 * /opt/ros/melodic/share/doc/liborocos-kdl
Executing command: mkdir /home/melodic-dev/src/AtlasCarCalibration/ur10e_eye_to_base_calibration/calibration
mkdir: cannot create directory ‘/home/melodic-dev/src/AtlasCarCalibration/ur10e_eye_to_base_calibration/calibration’: File exists
Executing command: mkdir /home/melodic-dev/src/AtlasCarCalibration/ur10e_eye_to_base_calibration/launch
mkdir: cannot create directory ‘/home/melodic-dev/src/AtlasCarCalibration/ur10e_eye_to_base_calibration/launch’: Read-only file system
Executing command: mkdir /home/melodic-dev/src/AtlasCarCalibration/ur10e_eye_to_base_calibration/rviz
mkdir: cannot create directory ‘/home/melodic-dev/src/AtlasCarCalibration/ur10e_eye_to_base_calibration/rviz’: Read-only file system
Executing command: mkdir /home/melodic-dev/src/AtlasCarCalibration/ur10e_eye_to_base_calibration/urdf
mkdir: cannot create directory ‘/home/melodic-dev/src/AtlasCarCalibration/ur10e_eye_to_base_calibration/urdf’: Read-only file system
Executing command: mkdir /home/melodic-dev/src/AtlasCarCalibration/ur10e_eye_to_base_calibration/scripts
mkdir: cannot create directory ‘/home/melodic-dev/src/AtlasCarCalibration/ur10e_eye_to_base_calibration/scripts’: File exists
Traceback (most recent call last):
  File "/home/melodic-dev/src/AtlasCarCalibration/interactive_calibration/scripts/create_calibration_pkg.py", line 59, in <module>
    f = open(configure_file, 'w')
IOError: [Errno 30] Read-only file system: '/home/melodic-dev/src/AtlasCarCalibration/ur10e_eye_to_base_calibration/scripts/configure
```

## Remarks
Ignore the read-only error, I was testing your code in a docker container to emulate a fresh install.
But is  also tell us that it is openning a file to write where is should not.

miguelriemoliveira, 2020-05-28:
Thanks for the testing. I will try to solve it and get back.

miguelriemoliveira, 2020-05-28:
Hi @eupedrosa ,

I think I solved all issues with the create_calibration_pkg.py.

Please try again.

miguelriemoliveira, 2020-05-30:
Check #47 #55 and #60 should also be included in this sweep ...

miguelriemoliveira, 2020-05-30:
All issues reported by @eupedrosa  are solved, I think.

1. paths now support uri format;
2. launch files come from templates;
3. several bugs fixed.

Can you test again @eupedrosa  and @aaguiar96 (eye_to_base, agrob)?

Step by step:

Create a calibration package:
rosrun interactive_calibration create_calibration_pkg.py -n atlascar2_calibration

Edit the config.yml, and after reconfigure your calibration package
`rosrun atlascar2_calibration configure`

set initial estimate of the sensor poses
`roslaunch atlascar2_calibration set_initial_estimate.launch`

create a dataset
`roslaunch atlascar2_calibration collect_data.launch output_folder:=~/datasets/test
`

Thanks in advance.

miguelriemoliveira, 2020-06-06:
We can close this one. Will continue with more specific issues.

closed 2020-06-06 by miguelriemoliveira