lardemua/atom #159: Improve pattern limit points calculation

aaguiar96, 2020-05-26:
Hi  @miguelriemoliveira 

During the meeting with @eupedrosa we concluded that the misalignment of the corners might be related to the way pattern limit points is being calculated.

@eupedrosa suggested using line equations instead of representing the lines as a discrete number of points.
I can do this, but I think is better for us to talk first because these changes will affect the calibrations of the other sensor configurations.

@eupedrosa and @miguelriemoliveira, let's meet this week?

miguelriemoliveira, 2020-05-26:
Hi @aaguiar96 ,

I can meet Wednesday and Thursday afternoon, Friday all day.

Can you describe the problem you were having in more detail?

At the present moment I disagree with using line equations because of my philosophy: keep it simple until it works. 

I mean, the discrete approach (using points) should work in theory, right? To me, we should put the more simple solution working, and only after increase the complexity of the system.

Moreover, I think there is a second, perhaps more cumbersome problem with using "point-to-line distance", which is that what we really want is a "point-to-line-segment" distance, and this is much harder to define (unless I am missing something). Actually, to the best of my knowledge, the best way to do that is a discrete approach.

eupedrosa, 2020-05-26:
Let me pitch in..

I believe that the discrete approach is actually more complex. @miguelriemoliveira you talk about not using the line equation, but you are using it to calculate the discrete points.

The suggestion is to use vectors as the edges of the pattern. Calculating the distance from a point to 4 line segments (define by vectors ) is more clean than a brute force against all points. Distance to line segment is not difficult.

We will change your knowledge... ;)

miguelriemoliveira, 2020-05-27:
But what is the distance from a point to a line segment? Can you draw a schematic?

aaguiar96, 2020-05-27:
Hi @miguelriemoliveira 

"the distance from a point to a line is the shortest distance from a given point to any point on an infinite straight line", i.e., is the distance from the source point to the closest point in the target line.

[image]

aaguiar96, 2020-05-27:
> I can meet Wednesday and Thursday afternoon, Friday all day.

Me too. @eupedrosa what about you?

eupedrosa, 2020-05-27:
Another diagram, but now using vectors...

[image]

The distance can be calculated using vectors and the `dot` product, which gives the projection.
It always returns the closes point to the line segment. If the projection is negative, the distance is between the query point (C) and the segment point used as origin (A) (middle diagram). If the projection value is higher than the line segment length, then the distance is the query point (C) to to the other line segment point (B) (right diagram).

miguelriemoliveira, 2020-05-27:
Hi @aaguiar96 and @eupedrosa ,

@aaguiar96 , your solution is for distance from point to line , not from point to line segment, am I right?

@eupedrosa , I like the solution. It works for point to line segment, and I think the cases left and right will output the distance that we would have computed using the discrete approach. So with this approach I agree. It is elegant.

One advise though: draw the distance lines in the visualization.py  so that we can check that the functionality is operating as intended. You can check the draw beams  for an example.

miguelriemoliveira, 2020-05-27:
Just another question: @aaguiar96 was using the corners of the pattern, so for that the distance we would be talking about would be the point to point distance no? Are you using all border points and not just the corner ones?

aaguiar96, 2020-05-27:
> @aaguiar96 , your solution is for distance from point to line , not from point to line segment, am I right?

Yes @miguelriemoliveira, in this case, we would use some line equation

> Are you using all border points and not just the corner ones?

For now, I'm using only the corners.
What's happening is that the pattern corners and the LiDAR corners on the pattern do not match. I still don't know why, but we thought that the `dataset_chessboards['limit_points']` might not be adapted for charuco. Of course, there is the possibility of existing some bug on my code too! :)

miguelriemoliveira, 2020-05-27:
Ok. But if you will still use the corners of the velodyne point cloud, you will compute the distance to all 4 line segments? ... and select the smallest?

aaguiar96, 2020-05-27:
> Ok. But if you will still use the corners of the velodyne point cloud, you will compute the distance to all 4 line segments? ... and select the smallest?

I think we can use only the two horizontal or vertical lines and compute the smallest distance from each LiDAR corner to each line. In theory, the corresponding pattern points will be the pattern corners. No?

miguelriemoliveira, 2020-05-27:
Yes, makes sense.

*  Miguel Riem de Oliveira                                             *
email:    mriem@ua.pt    |    m.riem.oliveira@gmail.com
office: 22.2.18                 |    ext: 23887
Dep. de Engenharia Mecânica, Universidade de Aveiro
*Campus Universitário de Santiago, 3810-193  Portugal*

miguelriemoliveira, 2020-06-12:
this is complete no? Work now is on #149 ?

aaguiar96, 2020-06-12:
Yes @miguelriemoliveira 

I'll close the issue

closed 2020-06-12 by aaguiar96