lardemua/atom #160: Problem with roslaunch and raw_input

miguelriemoliveira, 2020-05-26:
Hi @eupedrosa ,

when redesigning the code I have the following problem: using raw_input (as we do in the data_collector and labeller) seems to be  incompatible with roslaunch.

Can you call with a roslaunch script a node that uses raw_input? I get the error:

> /home/mike/datasets/test6Traceback (most recent call last):
>   File "/home/mike/catkin_ws/src/calibration/AtlasCarCalibration/interactive_calibration/scripts/collect_and_label_data.py", line 133, in <module>
>     data_collector = DataCollectorAndLabeler(args, server, menu_handler)
>   File "/home/mike/catkin_ws/src/calibration/AtlasCarCalibration/interactive_calibration/src/interactive_calibration/data_collector_and_labeler.py", line 38, in __init__
>     answer = raw_input(str(args['output_folder']))
> EOFError

Not sure how to solve this, we can also think about redesigning the functionality:

1. give a path to the dataset
2. if the path exists abort, unless a "overwrite flag has been set to true"

What do you think?

eupedrosa, 2020-05-26:
There are several behaviors that we can adopt:
* Always fail if a directory already existe, use  exit(1)
* Always replace the directory
* Ask the user, if the `-y` is passed as argument assume all answers to be 'Yes'
* Add a suffix to the diretory name, like in ROS bags, append the current date.
* Move the current directory to a backup name, see https://stackoverflow.com/a/47087513

Pick one? maybe.

### Extra: 
In python you can use `os.stdout.istty()` to check if the program ir running on a terminal. This will return false, for example, using roslaunch. Anyway, you have to decide what to do, replace, fail?

miguelriemoliveira, 2020-05-26:
Hi  @eupedrosa ,

thanks. I think the best is to:

> Ask the user, if the -y is passed as argument assume all answers to be 'Yes'

I will proceed with this implementation.

miguelriemoliveira, 2020-05-27:
Well, I cant't get the  os.stdout.istty() to work (perhaps its python 3). Googled and found several alternatives: 

```
print('istty said = ' + str(sys.stdout.isatty()))
print(os.isatty(0))
print(sys.flags.interactive)

```

None of them is able to distinguish between calling the node from a terminal or from a launch file. I guess it is also interactive since we give the output="screen" flag in the launch file, but then fails with the error detailed in the beginning of the issue.

I will move to a new alternative:

> Move the current directory to a backup name, see https://stackoverflow.com/a/47087513

Everytime the dataset exists, I jsut move it to the /tmp folder with a suffix with the date or something.

This decision deprecates #161

eupedrosa, 2020-05-27:
My fault, I wrote it by memory.
Did you try both stdin and sdout? like `interactive = sys.stdin.isatty() and sys.stdout.isatty()`

miguelriemoliveira, 2020-05-27:
Hi @eupedrosa ,

still the same, I get also true for when this is called from a launch file... any other suggestions? 

If not I just move to :

> Move the current directory to a backup name, see https://stackoverflow.com/a/47087513

eupedrosa, 2020-05-27:
I think  we can move to a backup name. This is what I actualy did for the hand-eye, but manually.

miguelriemoliveira, 2020-05-28:
I think it is working, can you try it out @eupedrosa ?

miguelriemoliveira, 2020-05-28:
Please close if it works for you.

closed 2020-06-04 by miguelriemoliveira