lardemua/atom #162: Failsafe plan for not finding mesh_file ...

miguelriemoliveira, 2020-05-27:
```
  # TODO If no mesh is given, or if mesh_file does not exist, isse a warning and create a drawing of the
        #  pattern with lines m.mesh_resource = 'package://interactive_calibration/meshes/charuco_5x5.dae'
        m.mesh_resource = dataset['calibration_config']['calibration_pattern']['mesh_file']
        m.mesh_use_embedded_materials = True
        markers.markers.append(m)
```

miguelriemoliveira, 2020-05-29:
Hum, not sure about this. I think the drawing of the meshes strucutre (i.e. the lines and points that we use for optimization should always be there, not just as a backup to not being able to draw the mesh with texture.)

aaguiar96, 2020-05-30:
Hi @miguelriemoliveira 

Honestly, I did not understand your doubt. Sorry!

aaguiar96, 2020-05-30:
Ahh ok, I think I got it.

I think that the points used for optimization should always be drawn.
However, if no mesh file is provided, maybe the user is not interested in seeing the pattern? Should we draw a `charuco_5X5` even if the real pattern could be an arbritary chessboard?

miguelriemoliveira, 2020-05-30:
So the idea of the issue was to find a way to have a backup plan for drawing the pattern when the mesh file was not present. 

The backup plan would be to draw some lines and points, like a wireframe of the pattern, the limit points, etc.

After working with you on the visualization I think that drawing the lines and points and limit points of the pattern is not accessory and should not be drawn only when we do not have a mesh. It should always be draw (and of course it can be hidden from rviz).

You agree?

miguelriemoliveira, 2020-05-30:
I was late. Yes, that was it. We should only draw if we have the correct mesh, I think. So I will close this one since there is no need for a "failsafe plan".

Thanks

closed 2020-05-30 by miguelriemoliveira