lardemua/atom #167: Initial set up of atom_core

miguelriemoliveira, 2020-06-02:
First step, get the sensor_pose_v2 code from optimization utils into atom_core.

miguelriemoliveira, 2020-06-02:
Hi @aaguiar96 ,

I have made the import.

Here's how I run:

`roslaunch atom_calibration atlascar2_view_optimization.launch`

`rosrun atom_core main.py -json ~/datasets/calibration_30_mar_2020/data_collected.json -vo -si -rv`

there should still be some problems. I will work on them later. Now it is too late.

miguelriemoliveira, 2020-06-02:
Hi @aaguiar96 ,

did you manage to try it out? I should have some time tonight to correct possible errors. I will also work the automating the calibration step with a single launch file.

Also, I will try to draw the pattern in the visualization.py.

aaguiar96, 2020-06-02:
Hi @miguelriemoliveira 

Sorry, I did not... I am with tons of work on the INESC side... I'll reserve one hour today to try it out, at the end of the afternoon! :)

aaguiar96, 2020-06-02:
I'll also try to implement the hessian coefficients of the plane!

miguelriemoliveira, 2020-06-02:
Great.
*  Miguel Riem de Oliveira                                             *
email:    mriem@ua.pt    |    m.riem.oliveira@gmail.com
office: 22.2.18                 |    ext: 23887
Dep. de Engenharia Mecânica, Universidade de Aveiro
*Campus Universitário de Santiago, 3810-193  Portugal*

aaguiar96, 2020-06-02:
Hi @miguelriemoliveira 

I was able to run the system.
I just got this rviz error regarding the mesh visualization

```
Mesh resource marker [11/5000] could not load [file://('/home/andre/Source/catkin_ws/src/atom/atom_calibration/meshes/charuco_5X5_800x600.dae', 'charuco_5X5_800x600.dae', ParseResult(scheme='', netloc='', path='/home/andre/Source/catkin_ws/src/atom/atom_calibration/meshes/charuco_5X5_800x600.dae', params='', query='', fragment=''))]
```

miguelriemoliveira, 2020-06-02:
Hi @aaguiar96 ,

fixed.

[image]

@eupedrosa , I still get confused with this strange reference system from the charucos. The z axis goes away from the camera (unlike hte chessboard). And when we draw the pattern is painted on the wrong face.  Take a look at the image

We talked about this before: but could you not reverse the side of the texture in the charuco pattern's mesh?

miguelriemoliveira, 2020-06-02:
Hi again @eupedrosa ,

I know we (I don't remember who ) solved this for the hadn eye paper.

Any idea what we did? From the images in the paper it seems the charucos are pointing towards the camera (as if the reference system is fixed). Look at figs 4 and 5 in the paper...

eupedrosa, 2020-06-02:
I remember the "problem" but I do not know how you "hack it".

There is no problem with the charuco frame of reference. It uses the "right hand" convention for its axis - just like everything in ROS. The problem is the chessboard detector that uses a "left hand" convention.

miguelriemoliveira, 2020-06-03:
Hi @eupedrosa ,

chessboard reference frames are not left handed, I think: X for left to right, Y from top to bottom, Z pointing away from the camera.

This is right-handed. It is different from the charuco, but it is right handed. Am I seeing things wrong?

The point I think is: The pattern can have any frame, but the dae file must be consistent with that convention. 

For example, if the chessboard pattern has the Z pointing away from the camera, then the mesh should have the texture painted in the side that is observed when you are at X=0, Y=0, Z = -10 looking at the pattern (local pattern frame coordinates).

For the charuco (and this is what I think is wrong with the charuco, perhaps just the drawing), if the 
Z is pointing towards the camera, then the mesh should have the texture painted in the other side, i.e. the one that  is observed when you are at X=0, Y=0, Z = +10 looking at the pattern (local pattern frame coordinates).

So my question: don't your dae files have the texture painted on the wrong side?

eupedrosa, 2020-06-03:
Look, if the charuco dae is drawn correctly when charuco detection is used how can it be wrong?

The chessboard uses the u for x coordinate and v for y coordinate. Now, which direction z has? The same as the camera optical frame, right? Which is right handed convention. It also means that from the camera point o view and using the correct convention we are looking at the back of the chessboard object... If we assume that we are looking at the front of the chessboard, then it has to be a left hand convention. Obviously that in Ros there is no left hand convention. Therefore, because the model was drawn to the more sain charuco, it looks wrong.

To finish, the charuco dae is not incorrect.

miguelriemoliveira, 2020-06-03:
Hi @eupedrosa ,

> The chessboard uses the u for x coordinate and v for y coordinate. Now, which direction z has? The same as the camera optical frame, right? Which is right handed convention. It also means that from [...]

the "back" is an arbitrary concept. I could call it the front, right? In any case, let's assume that the chessboard has the back to the camera. All we have to do is to have a mesh with has the texture in the back, rather than the front.

> Look, if the charuco dae is drawn correctly when charuco detection is used how can it be wrong?

I will take a look at this and try to figure out how you do it.

eupedrosa, 2020-06-03:
> the "back" is an arbitrary concept. I could call it the front, right? In any case, let's assume that the chessboard has the back to the camera. All we have to do is to have a mesh with has the texture painted in the back, rather than the front.

If I do this, then it will fail for the charuco. And, front and back is not arbitrary (in 3D graphics) it is well establish that a face has a front and a back.

> I will take a look at this and try to figure out how you do it.

I did nothing special. The charuco detector defines an origin and I used that origin.
The diference for the chessboard is that the positive x goes from right to the left, this has the effect of flipping the Z axis when compared with the chessboard. This is observed when you use SolvePnP.

miguelriemoliveira, 2020-06-06:
This issue is done.  Will continue work on the visualization in #170

closed 2020-06-06 by miguelriemoliveira