lardemua/atom #169: Add BAGFILES and DATASETS enviroment varibales to your .zshrc

miguelriemoliveira, 2020-06-05:
Hi @aaguiar96 and @eupedrosa ,

so that we have a way of finding the bagfiles and datasets folder for each of use I suggest that we use these two env variables. From my .zshrc:

```
export ROSBAGS="/home/mike/bagfiles"
export DATASETS="/home/mike/datasets"              
```

then, instead of having the config.yml like agrob was before:

`"bag_file": "/home/andre/Documents/rosbags/agrob_calibration_2020-05-20-12-2/calibration_2020-05-20-12-25-58_0.bag",
`
we can now have:

`"bag_file": "$ROSBAGS/agrob_calibration_2020-05-20-12-2/calibration_2020-05-20-12-25-58_0.bag",`

I just pushed these changes to the agrob config.yml.

If you agree please add the variables to your shrc, or give me some alternatives.

eupedrosa, 2020-06-06:
Hi @miguelriemoliveira. Yes, that is an option.

However, we are yet to solve the problem of storing and sharing the bagfiles :\

eupedrosa, 2020-06-18:
With #183, I think it safe to use environment variable. Following ROS style, I think I will use `ROS_BAGS` for my bags directory.

miguelriemoliveira, 2020-06-18:
Wait @eupedrosa , we should all use the same environment variables otherwise it defeats the purpose of using them. We can use these? Can you add this to a section in the readme?

export ROS_BAGS="/home/mike/bagfiles"
export ATOM_DATASETS="/home/mike/datasets"

eupedrosa, 2020-06-18:
Ok. I will do that.

eupedrosa, 2020-06-19:
I think we all agree on this one. Closing.

closed 2020-06-19 by eupedrosa

miguelriemoliveira, 2020-06-19:
Did you add a section on this in the README?

eupedrosa, 2020-06-19:
Yes.

miguelriemoliveira, 2020-06-19:
Great