lardemua/atom #173: Bug in charuco detection ...

miguelriemoliveira, 2020-06-10:
when running:

roslaunch ur10e_eye_in_hand_calibration collect_data.launch bag_rate:=0.4 output_folder:=~/datasets/test overwrite:=true

I get after 15 secs an error in the charuco detector:

> [ERROR] [1591777705.338495, 1579803593.672204]: bad callback: <bound method InteractiveDataLabeler.sensorDataReceivedCallback of <atom_calibration.interactive_data_labeler.InteractiveDataLabeler instance at 0x7f1e3d0d7460>>
> Traceback (most recent call last):
>   File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/topics.py", line 750, in _invoke_callback
>     cb(msg)
>   File "/home/mike/catkin_ws/src/calibration/atom/atom_calibration/src/atom_calibration/interactive_data_labeler.py", line 213, in sensorDataReceivedCallback
>     self.labelData  # label the data
>   File "/home/mike/catkin_ws/src/calibration/atom/atom_calibration/src/atom_calibration/interactive_data_labeler.py", line 335, in labelData
>     result = self.pattern.detect(image, equalize_histogram=True)
>   File "/home/mike/catkin_ws/src/calibration/atom/atom_calibration/src/atom_calibration/patterns.py", line 116, in detect
>     ccorners = cv2.cornerSubPix(gray, ccorners, (5,5), (-1, -1), criteria)
> error: OpenCV(4.2.0) /io/opencv/modules/imgproc/src/cornersubpix.cpp:58: error: (-215:Assertion failed

Do you know what could be wrong? Did we not remove the subpix a couple of weeks ago?

miguelriemoliveira, 2020-06-10:
dba8c00 fixes it but removes the subpix computation part. We could wrap it around a try catch ... does it really make a difference to have subpixel accuracy for the detection of the corners?

eupedrosa, 2020-06-10:
> does it really make a difference to have subpixel accuracy for the detection of the corners?

It does, it improves the corner detection. It is always failing?

miguelriemoliveira, 2020-06-10:
No, in some situations ...

miguelriemoliveira, 2020-06-11:
@eupedrosa , can you add the try catch wrapper? Do you take this issue?

miguelriemoliveira, 2020-07-02:
@eupedrosa ,

should we close this one?

closed 2020-07-02 by miguelriemoliveira