lardemua/atom #183: Paths in `config.yml` with 'ENV' variables and '~' are saved as absolute paths in launch files.

eupedrosa, 2020-06-17:
Using environment variables and the home ~ provides flexibility when defining paths in `config.yml`. However, this paths are expanded to their absolute value when `configure` is used to generate the auxiliary launch files. This creates a problem when sharing the calibration project. For example the home directory is no the same for all.

eupedrosa, 2020-06-17:
@miguelriemoliveira, I tested this in my computer and it is working.
However, you should also test it to make sure everything is ok.

closed 2020-06-17 by eupedrosa

miguelriemoliveira, 2020-06-19:
I can't get this to work:

> roslaunch atlascar2_calibration calibrate.launch dataset_file:=~/datasets/atlascar2/calibration_30_mar_2020/data_collected.json csf:='lambda x: int(x)<20' 

and the result is:

> Traceback (most recent call last):
>   File "/home/mike/catkin_ws/src/calibration/atom/atom_core/scripts/calibrate", line 598, in <module>
>     main()
>   File "/home/mike/catkin_ws/src/calibration/atom/atom_core/scripts/calibrate", line 146, in main
>     f = open(json_file, 'r')
> IOError: [Errno 2] No such file or directory: '~/datasets/atlascar2/calibration_30_mar_2020/data_collected.json'

I think it was something you did in this issue about handling the ~ ... do you know what?

eupedrosa, 2020-06-19:
No, that is something different, it has to do with the changes in 0c6ee810e6c80052f095528853881e956b9ba661
That still works for the config file.

Did we not agree to use `$ATOM_DATASETS`? In your case you should use 
```bash
roslaunch atlascar2_calibration calibrate.launch dataset_file:=$ATOM_DATASETS/atlascar2/calibration_30_mar_2020/data_collected.json csf:='lambda x: int(x)<20'
```

But that is a easy change if you want to have the ~ back.

miguelriemoliveira, 2020-06-20:
Hi @eupedrosa ,

ok, I thought that we wanted to give the possibility to use ~, environment variables, uris etc in the config file and in the command line arguments.

I mean, the $ROS_BAGS and $ATOM_DATASETS can be used, but if someone just wants to use ~ I would say it should also be possible. 

Up to you.

eupedrosa, 2020-06-20:
But just the ~ or you also want the URIs?

miguelriemoliveira, 2020-06-20:
You decide. But I would say all (~ amd uris) or nothing

commit 033a8c7d (2020-06-17): Fix #183.