lardemua/atom #185: General effort to reduce code dependencies.

eupedrosa, 2020-06-18:
In my opinion, dependencies should be kept to a minimum. This helps reduce complexity and makes it easier to deploy our software.

It is true that a `requirements.txt` helps to keep track of 3rd party dependencies, but we should avoid redundant dependencies. There are also dependencies no longer in use.

eupedrosa, 2020-06-18:
@miguelriemoliveira, I know you a have a strong opinion about this. But let's discuss this issue.

miguelriemoliveira, 2020-06-18:
Sure. I would say we need a list of the dependencies you think we could eventually remove. Then we can go one by one ...

eupedrosa, 2020-06-18:
We can use this issue for that.. Any proposal for removal should go through a discussion here.

miguelriemoliveira, 2020-06-18:
Agreed!

eupedrosa, 2020-06-19:
**Package to remove:** open3d
**Reason:** Not in use. 

I believe this is straightforward the package is not being used.

miguelriemoliveira, 2020-06-19:
> Package to remove: open3d
> Reason: Not in use.
> 
> I believe this is straightforward the package is not being used.

Agree. But perhaps we may have to re-add that with #155. If we do need it , we readd.

eupedrosa, 2020-06-19:
If needed.. but using open3d just to read a ply file may be overkill. [python-ply](https://github.com/dranjan/python-plyfile) is simpler.

miguelriemoliveira, 2020-06-19:
I agree.

eupedrosa, 2020-06-24:
Remove
https://github.com/lardemua/atom/blob/00bd1cba6df9f534407b25a5f28534a48de5ca30/atom_core/src/atom_core/objective_function.py#L8

Merged by mistake.

miguelriemoliveira, 2020-08-05:
Hi @eupedrosa 

Removed dependency on **KeyPressManager**

You should remove the path for KeyPressManager from $PYTHONPATH which is installed with 
https://github.com/miguelriemoliveira/KeyPressManager/blob/master/install.sh

Check your .bashrc or .zhrc.

I will also remove the dependency from rospy_urdf_to_rviz_converter, later on I ask you to try it out using your docker magic OK?

miguelriemoliveira, 2020-08-05:
Also removed dependency from rospy_urdf_to_rviz_converter

Again, in the .shrc you should remove both these lines

```
export PYTHONPATH="${PYTHONPATH}:/home/mike/workingcopy/KeyPressManager"
export PYTHONPATH="${PYTHONPATH}:/home/mike/workingcopy/rospy_urdf_to_rviz_converter" 
```

@eupedrosa , it works from my side. Can you test with docker to see if a new installation runs smoothly?

miguelriemoliveira, 2020-08-05:
Is there any other dependency to remove?

eupedrosa, 2020-08-06:
> @eupedrosa , it works from my side. Can you test with docker to see if a new installation runs smoothly?

The installation is yet to run smoothly. But it is getting easier. I guess the worst part right now is the `OptimizationUtils`. Requiring the user to do
```
export PYTHONPATH="${PYTHONPATH}:/path/to/OptimizationUtils" 
```
is not the best, specially if it is the only alternative.  In my opinion there are two ways to solve this:
1. Change `OptimizationUtils` to a ros package.
2. Create a proper `setup.py` so that we can use `pip install`

> Is there any other dependency to remove?

Yes there is. For example, `pandas` is being used in `OptimizationUtils` just to print tables.

eupedrosa, 2020-08-06:
There is another thing that we should address, which is the removal of uncessary code and lines and lines of commented code. We use *git* for history, therefore there is no need to keep old code in comments.

miguelriemoliveira, 2020-08-06:
Hi @eupedrosa ,

About this:

> Change OptimizationUtils to a ros package.

Does not make sense because other projects use it too, including non ROS projects .

> Create a proper setup.py so that we can use pip install

Right now we have an install.sh, but that option would be better. Can you work on that?

> Yes there is. For example, pandas is being used in OptimizationUtils just to print tables.

I think pandas is not really a problem. You can install it using 

`pip install -r requirements.txt`

This is really simple to do, and you already have to do it for the other dependencies.

We have to find a middle ground here. I concedeed with KeyPressManager, rospy_urdf_to_rviz_converter, etc, can you give me pandas : - ). If you really insist in removing pandas you can create a function to print in tables as in pandas to replace that functionality.

miguelriemoliveira, 2020-08-06:
> There is another thing that we should address, which is the removal of uncessary code and lines and lines of commented code. We use git for history, therefore there is no need to keep old code in comments.

I can do that. I will try to do a cleaning.

eupedrosa, 2020-08-06:
> > Create a proper setup.py so that we can use pip install

> Right now we have an install.sh, but that option would be better. Can you work on that?

Yes, I can do that.

> I think pandas is not really a problem. You can install it using
>
> pip install -r requirements.txt
>
> This is really simple to do, and you already have to do it for the other dependencies.
>
> We have to [...]

I know it is easy to install pandas. I am talking about the *cost-benefit*  ratio. I remember (if I am not mistaken) that ROS used to declare *texlive* as a dependency just to create a PDF with the TF tree. That said, after removing *KeyPressManager* and *rospy_urdf_to_rviz_converter* you can have *pandas*.

miguelriemoliveira, 2020-08-07:
Hi @eupedrosa ,

Since we are releasing it to the public, wrote a large tutorial for OptimizationUtils. It is in the readme.md of the OptmizationUtils.

If you can take a look and replace add whatever you think.

Visualization function and starting up the optimization are still missing.

eupedrosa, 2020-08-07:
I did not understood your example for optimization. The `cat` and `dog` stuff. What are you trying to optimize???
Why not use the cosine example?

miguelriemoliveira, 2020-08-07:
The idea was to use an abstract example, in which you have the weight and height of a dog and a cat as parameters of the optimization.

The goal would be to get a dog and a cat that have the same weight and height. The good thing about this example is the fact that it is a sparse problem.

The cosine fitting has the sparse matrix with all ones, which is really not very helpful for explaining ... any other ideas? Or should I try to explain the dog cat better, something else ...

miguelriemoliveira, 2020-08-08:
I just finished with the reorganization of optimization utils. Now all that's left form that side is the readme. Will work on that once we agree on an example.

miguelriemoliveira, 2020-08-25:
Closing. Some more efforts should be endeavored but we should open specific issues for each.

closed 2020-08-25 by miguelriemoliveira