lardemua/atom #187: Verify config data structure has all the keys

miguelriemoliveira, 2020-06-19:
Compare the keys from the template with those of a datastructure.

miguelriemoliveira, 2020-06-19:
Completed. If the template config.yml has some keys which do not exist in the robot's config.yml the code aborts with an informative error:

> Traceback (most recent call last):47426   Duration: 2.764289 / 231.995038               
>   File "/home/mike/catkin_ws/src/calibration/atom/atom_calibration/scripts/collect_data", line 141, in <module>
>     data_collector = DataCollectorAndLabeler(args, server, menu_handler)
>   File "/home/mike/catkin_ws/src/calibration/atom/atom_calibration/src/atom_calibration/data_collector_and_labeler.py", line 69, in __init__
>     self.config = loadConfig(args['calibration_file'])
>   File "/home/mike/catkin_ws/src/calibration/atom/atom_calibration/src/atom_calibration/utilities.py", line 137, in loadConfig
>     ' rosrun <your_robot>_calibration configure' + Fore.RESET)
> ValueError: Your config file /home/mike/catkin_ws/src/calibration/atlascar2/atlascar2_calibration/calibration/config.yml appears to be corrupted. These mandatory parameters are missing: ['another_key']
> Perhaps you should re-run:
>  rosrun <your_robot>_calibration configure

@eupedrosa , you think this is enough? If so please close ...

eupedrosa, 2020-06-19:
Is it necessary to print the Traceback?

miguelriemoliveira, 2020-06-19:
Do yoy have a better way? What do you suggest?

eupedrosa, 2020-06-19:
I have to see the code of your proposal.

miguelriemoliveira, 2020-06-19:
```python
 if not all(key in config for key in template_config):
        missing_keys =  [key for key in template_config if key not in config]
        raise ValueError(Fore.RED + 'Your config file ' + filename +
                         ' appears to be corrupted. These mandatory parameters are missing: ' + Fore.BLUE + str(missing_keys)
                         + Fore.RED + '\nPerhaps you should re-run:\n' + Fore.BLUE +
                         ' rosrun <your_robot>_calibration configure' + Fore.RESET)
```

eupedrosa, 2020-06-19:
I was reading your commit.

I believe this will only validate the most top level `keys`. That is, `keys` in sub-dictionaries are not validated.

miguelriemoliveira, 2020-06-19:
This was a tricky one. Done, please close if nothing else to do.

> Traceback (most recent call last):
>   File "/home/mike/catkin_ws/src/calibration/atom/atom_calibration/scripts/collect_data", line 141, in <module>
>     data_collector = DataCollectorAndLabeler(args, server, menu_handler)
>   File "/home/mike/catkin_ws/src/calibration/atom/atom_calibration/src/atom_calibration/data_collector_and_labeler.py", line 69, in __init__
>     self.config = loadConfig(args['calibration_file'])
>   File "/home/mike/catkin_ws/src/calibration/atom/atom_calibration/src/atom_calibration/utilities.py", line 154, in loadConfig
>     ' rosrun <your_robot>_calibration configure' + Fore.RESET)
> ValueError: Your config file /home/mike/catkin_ws/src/calibration/atlascar2/atlascar2_calibration/calibration/config.yml appears to be corrupted. These mandatory parameters are missing: ['calibration_pattern/test']
> Perhaps you should re-run:
>  rosrun <your_robot>_calibration configure
>

closed 2020-06-28 by miguelriemoliveira