lardemua/atom #189: Merge general hand eye with calibrate

miguelriemoliveira, 2020-06-19:
Should do this to advance in the general approach. 

Defined with @eupedrosa that the best way is to try to put in calibrate the stuff from hand eye, i.e. to use the calibrate as base.

miguelriemoliveira, 2020-06-22:
Hi @eupedrosa and @aaguiar96 ,

I need some help from you. I was working throughout the weekend on this issue. The goal was to fuse this 

https://github.com/miguelriemoliveira/OptimizationUtils/blob/master/test/general_hand_eye/general_hand_eye.py

with the calibrate code 

https://github.com/lardemua/atom/blob/master/atom_core/scripts/calibrate

I have created this branch  to work on this

https://github.com/lardemua/atom/blob/test_189/atom_core/scripts/calibrate

The code from general_hand_eye is more recent, and it turned out that it has some more interesting ways of doing things.

Here's the list of  changes I can remember  I did to the calibrate in atom_core taking some of the general_hand_eye stuff:

1. Moved the place where the pattern's transforms are in the dataset structure to where all the others are, i.e. in datasets['collections'][collection_key]['transforms'] .  Before they were in dataset['patterns']['collections'][collection_key] . Now the initial estimate is stored in dataset['patterns']['transforms_initial']

2. Because of 1, it was possible to design a single getter and setter for the sensors and the patterns

https://github.com/lardemua/atom/blob/0c6ee810e6c80052f095528853881e956b9ba661/atom_core/src/atom_core/getters_and_setters.py#L21-L37

3. Because of #120, used the unique set of optimized transforms to define parameters. Now parameters represent transformations, rather than sensors. Changed parameter's names accordingly

4. Revised the creation of residuals to conform with new names in 3.

5. Some changes in the visualization, details

6. Objective function, Image msg_type. Here revised the computation of the transformation. REplaced the old formulation which would retrieve the sensor_to_root, root_to_pattern and then compute the sensor_to_pattern by multiplying both. Now a much simpler approach (similar to Eurico's idea of the tf tree) which uses a function which was already implemented, i.e. **optimization_utilities.getTransform(from_frame, to_frame, collection['transforms'])**

https://github.com/lardemua/atom/blob/1e2f7437cf9032e279bc39060d29ffb17fcf7b1e/atom_core/src/atom_core/objective_function.py#L194-L200

**PROBLEM** it seems as if the optimization works fine (we get below one pixel residual average) for single collections, but when using multiple collections the erros is high, 5 or 8 residual average.

**One example:**  I was with @aaguiar96  testing the old code optimizing the agorb's new dataset.

Only collection 1 -> New code 0.5, André 0.5 
Only collection 8 -> New code 2.4, André 2.5
Collections 1 thought 9 -> _New code 8, André 4.

Something is wrong when working with multiple collections. Ideas, tests, I am a bit lost...

eupedrosa, 2020-06-22:
Did you check if the partial detections are working correctly?

miguelriemoliveira, 2020-06-22:
I think i did. Also, the atlascar dataset has tge same problem.  No partial
detections there.

eupedrosa, 2020-06-22:
And is it working for the hand-eye? An error of 2 pixel for atlascar I can understand. But 8 pixels it is too much.. Sorry I can not help you right now. Still away from home.

miguelriemoliveira, 2020-06-22:
I tested and I think it is also not working. But I think it should be
better to first get the atlascar and agrob working, since those were the
ones that worked before in calibrate.

aaguiar96, 2020-06-22:
> I tested and I think it is also not working.

Hi @miguelriemoliveira 

If it is not working in hand-eye, the error should be in some portion that you extracted from it, right?
Have you tried a step-by-step debug?
For example, partially replace portions of the new code with portions of the old code.
I.e., for example, if you replace the new way of computing the transformation by the older one, does it work? If not, keep the new approach. Do that for every single modification until you find the bug...

miguelriemoliveira, 2020-06-22:
> If it is not working in hand-eye, the error should be in some portion that you extracted from it, right?

It should be that, yes. 

> Have you tried a step-by-step debug?
> For example, partially replace portions of the new code with portions of the old code.
> I.e., for example, if you replace the new way of computing the [...]

I will do that. 

Thanks to both for help.

miguelriemoliveira, 2020-06-23:
Finally, after 6 days of being lost I got it. 

I had changed this portion of the code:

https://github.com/miguelriemoliveira/OptimizationUtils/blob/8cc8e0c9ae54eb648c5ccbe3a8c01bfab48456b9/OptimizationUtils/OptimizationUtils.py#L304-L315

with this (for getting a nice printout of what the missing key would be)

```python
   def errorDictToList(self, errors):

        if type(errors) is list:
            error_list = errors
        elif type(errors) is dict:
            error_dict = errors
            error_list = []

            for error_dict_key in error_dict.keys():
                if error_dict_key not in self.residuals.keys():
                    raise ValueError('Objective function returned dictionary with residual ' + Fore.RED +
                                     error_dict_key + Fore.RESET +
                                     ' which does not exist. Use printResiduals to check the configured residuals')

                error_list.append(error_dict[error_dict_key])
```

The problem was that, in doing this change, I lost the order of the residuals in the output list, since I started iterating a dictionary which offers no guarantees that the order of keys is the same.

So now I will revert this and we are back optimizing for multiple collections.  I did not confirm yet, but I am confident this is it.

This is the hardest bug I can remember I had is some 5 or 6 years. Thanks to all for the help to @eupedrosa  and @aaguiar96, and also to @afonsocastro (who was with me yesterday until 3am yesterday). 

@afonsocastro , you have to come continue to work with us. We are going to have scholarships in the near future!  :)

eupedrosa, 2020-06-23:
Programming is the art of creating bugs  :sparkles:

@miguelriemoliveira, did you confirm you findings???

miguelriemoliveira, 2020-06-23:
In that case I am a great artist :)

I did. The problem does not occur anymore. Did not commit yet. Tonight I will push a proposal for the merging. It takes a lot from hand eye. Then you tell me what you think.

Right now I am going for a beer :)

eupedrosa, 2020-06-23:
Great :beers:

aaguiar96, 2020-06-23:
> Right now I am going for a beer :)

Cheers! :beer:

miguelriemoliveira, 2020-06-24:
Hi @eupedrosa and @aaguiar96 ,

I have just merged the branch used for testing this issue. I think we have a merged functional version. I did not test yet for the hand eye scenarios but I tested for the atlascar and the agrob (the new dataset is not ok).

The code is simpler now I think. Again, the list of changes is this:

1. Moved the place where the pattern's transforms are in the dataset structure to where all the others are, i.e. in datasets['collections'][collection_key]['transforms'] . Before they were in dataset['patterns']['collections'][collection_key] . Now the initial estimate is stored in dataset['patterns']['transforms_initial']

2. Because of 1, it was possible to design a single getter and setter for the sensors and the patterns

3. Because of #120, used the unique set of optimized transforms to define parameters. Now parameters represent transformations, rather than sensors. Changed parameter's names accordingly

4. Revised the creation of residuals to conform with new names in 3.

5. Some changes in the visualization, details

6. Objective function, Image msg_type. Here revised the computation of the transformation. Replaced the old formulation which would retrieve the sensor_to_root, root_to_pattern and then compute the sensor_to_pattern by multiplying both. Now a much simpler approach (similar to Eurico's idea of the tf tree) which uses a function which was already implemented, i.e. optimization_utilities.getTransform(from_frame, to_frame, collection['transforms'])

@eupedrosa , are you ok with this solution of using **optimization_utilities.getTransform(from_frame, to_frame, collection['transforms'])**? It simplifies the transformations as you wanted and it does not use outside code. Take a look and gives us your comment.

@aaguiar96 , about 6. If @eupedrosa  agrees, I would ask you to revise your PointCloud2 code. See examples in the laserscan and image branches of the objective function, for example https://github.com/lardemua/atom/blob/00bd1cba6df9f534407b25a5f28534a48de5ca30/atom_core/src/atom_core/objective_function.py#L153-L157

eupedrosa, 2020-06-24:
If this works, yes, we can remote the old code. 

I will test the merge for the hand-eye.

eupedrosa, 2020-06-24:
@miguelriemoliveira, why was this added as a third party import?
https://github.com/lardemua/atom/blob/00bd1cba6df9f534407b25a5f28534a48de5ca30/atom_core/src/atom_core/objective_function.py#L8

It is not used and it is a very weird import, `mercurial` is a tool like `git`... I bet your PyCharms did it automatically...

miguelriemoliveira, 2020-06-24:
I have no idea why it is there ... I think you can remove . This should go into #185

eupedrosa, 2020-06-24:
https://github.com/lardemua/atom/blob/00bd1cba6df9f534407b25a5f28534a48de5ca30/atom_core/scripts/calibrate#L383

Here, we are assuming that an anchor sensor must be defined and it exist. This is not true for the hand-eye. There is no anchor sensor because we do not need one. It throws an exception and the calibration fails.

eupedrosa, 2020-06-24:
Hi @miguelriemoliveira,
The optimization viewer is no longer publishing the pattern link.

miguelriemoliveira, 2020-06-24:
Hi @eupedrosa ,

> atom/atom_core/scripts/calibrate
> 
> Line 383 in 00bd1cb
> 
>  anchored_parent = dataset['sensors'][anchored_sensor]['calibration_parent'] 
> Here, we are assuming that an anchor sensor must be defined and it exist. This is not true for the hand-eye. There is no anchor sensor because we do not need one. It throws an exception and the calibration fails.

We have to discuss this. I would say the parameter "anchored_sensor" should exist in the yaml because we are checking for all the fields (except the ones in the sensors). See this

https://github.com/lardemua/atom/blob/563e3d2351457819dc53632473e74e3498b87b79/atom_calibration/src/atom_calibration/utilities.py#L120-L163

If you have no sensor that is anchored just use:

anchored_sensor: ""

You agree with this solution?

miguelriemoliveira, 2020-06-24:
> The optimization viewer is no longer publishing the pattern link.

Perhaps it has a different name. For example here for the agrob we have the c0_chessboard_link

[image]

eupedrosa, 2020-06-24:
I agree with you. That is why I used anchored_sensor: ''" 
And it fails with this.. You may have misunderstood my description of the issue, or I explained it poorly.

eupedrosa, 2020-06-24:
> Perhaps it has a different name

Yes, the pattern seems to have a frame_id without the prefix.. the frame name is "chessboard_link".

miguelriemoliveira, 2020-06-24:
> I agree with you. That is why I used anchored_sensor: ''"
> And it fails with this.. You may have misunderstood my description of the issue, or I explained it poorly.

Ah, now I get it. Solved.

miguelriemoliveira, 2020-06-24:
> Yes, the pattern seems to have a frame_id without the prefix.. the frame name is "chessboard_link".

This has to though out more carefully. Right now, if the pattern is fixed I just publish a link without appendix. 
But perhaps I should replicate the link since this is what is done for the cameras. I will work on that.

miguelriemoliveira, 2020-06-28:
@eupedrosa I tested eye_to_base and it seemed not to work. Can you test?

miguelriemoliveira, 2020-07-02:
Merged. Now only #194 remains...

closed 2020-07-02 by miguelriemoliveira