lardemua/atom #194: Check eye_to_base

miguelriemoliveira, 2020-06-29:


miguelriemoliveira, 2020-07-15:
Hi @eupedrosa ,

will you have time for this?

eupedrosa, 2020-07-15:
Sorry, not until next week. But the only thing that you need is the correct pattern parameters. And here they are:

```yaml
- paper Charuco DICT_4X4
  x: 8, y: 5
  size: 0.029
  inner: 0.021

- Charuco from Calib.io
  x: 13, y: 8
  size: 0.02
  inner: 0.015
  outter: 0.01

- paper Chessboard
  x: 9, y: 7
  size: 0.019
  outter: 0.02
```

miguelriemoliveira, 2020-07-15:
Oi,

não há problema. Não percebi bem a tua resposta, mas se estavas a tentar
passar trabalho já tenho muitos issues e se conseguires pegar nisso para a
semana já é muito bom :).

Abraço,
Miguel
*  Miguel Riem de Oliveira                                             *
email:    mriem@ua.pt    |    m.riem.oliveira@gmail.com
office: 22.2.18                 |    ext: 23887
Dep. de Engenharia Mecânica, Universidade de Aveiro
*Campus Universitário de Santiago, 3810-193  Portugal*

eupedrosa, 2020-07-23:
After #214, this is now working.

closed 2020-07-23 by eupedrosa

miguelriemoliveira, 2020-07-23:
Great!

eupedrosa, 2020-07-23:
You can see the result of the optimization here:
[image]

miguelriemoliveira, 2020-07-24:
Looks good!