lardemua/atom #196: Pointcloud2 objective function is too slow

miguelriemoliveira, 2020-07-02:
I have narrowed it down to this part which is the one that is taking longer ...

```
                for idx in range(0, len(points)):
                    # Compute the interception between the chessboard plane into the 3D sensor reference frame, and the
                    # line that goes through the sensor origin to the measured point in the chessboard plane
                    p1_in_lidar = points[idx, :]
                    pt_intersection = isect_line_plane_v3(p0_in_lidar, p1_in_lidar, p_co_in_lidar, p_no_in_lidar)

                    if pt_intersection is None:
                        raise ValueError('Error: pattern is almost parallel to the lidar beam! Please delete the '
                                         'collections in question.')

                    # Compute the residual: distance from sensor origin from the interception minus the actual range
                    # measure
                    rname = collection_key + '_' + sensor_key + '_oe_' + str(idx)
                    rho = np.sqrt(p1_in_lidar[0] ** 2 + p1_in_lidar[1] ** 2 + p1_in_lidar[2] ** 2)
                    r[rname] = abs(distance_two_3D_points(p0_in_lidar, pt_intersection) - rho)

                    if args['ros_visualization']:
                        marker.points.append(deepcopy(rviz_p0_in_lidar))
                        marker.points.append(Point(pt_intersection[0], pt_intersection[1], pt_intersection[2]))
```

miguelriemoliveira, 2020-07-02:
@aaguiar96 , can you confirm this is slow? Pycharm can help you profile the code.

One option is to use the orthogonal distance instead of the beam distance, which would be very fast to compute ...

aaguiar96, 2020-07-03:
I @miguelriemoliveira 

Closing this one since we fixed it.

closed 2020-07-03 by aaguiar96