lardemua/atom #197: Calibrated zed camera does not have distortion

miguelriemoliveira, 2020-07-02:
seq: 55
  stamp: 
    secs: 1593594493
    nsecs: 795850487
  frame_id: "c0_zed_right_camera_optical_frame"
height: 720
width: 1280
distortion_model: "plumb_bob"
D: [0.0, 0.0, 0.0, 0.0, 0.0]
K: [692.687073, 0.0, 636.413391, 0.0, 692.687073, 382.806641, 0.0, 0.0, 1.0]
R: [1.0, 0.0, 0.0, 0.0, 1.0, 0.0, 0.0, 0.0, 1.0]
P: [692.687073, 0.0, 636.413391, -83.122452, 0.0, 692.687073, 382.806641, 0.0, 0.0, 0.0, 1.0, 0.0]
binning_x: 0
binning_y: 0
roi: 
  x_offset: 0
  y_offset: 0
  height: 0
  width: 0
  do_rectify: False
---
^C%

aaguiar96, 2020-07-03:
I @miguelriemoliveira 

Did you fixed this one here?
> Also, we added the optimization of the intrinsics and the distortion components (distortion was not there) since there images are clearly in need of some distortion. After this, camera errors droped from 2.5 pixels avg to bellow 1 pixel.

eupedrosa, 2020-07-03:
This is not fixed per se. But it is "solved" by the intrinsic optimization. I think the Zed camera has its own calibration app. Maybe you should explore that.

@miguelriemoliveira, this can be closed, right?

miguelriemoliveira, 2020-07-03:
I think we should leave it open (or create a new one) to explore why the zed does not have distortion.

The camera_info should have some values different from zero in the D field.

@aaguiar96 did you use the ros monocular calibration tool for te left and right cameras?

aaguiar96, 2020-07-03:
> I think we should leave it open (or create a new one) to explore why the zed does not have distortion.
> 
> The camera_info should have some values different from zero in the D field.
> 
> @aaguiar96 did you use the ros monocular calibration tool for te left and right cameras?

Yes @miguelriemoliveira, I used the stereo calibration toolbox from ROS.

miguelriemoliveira, 2020-07-03:
OK, but perhaps if we use the monocular camera calibration we get the distortion

http://wiki.ros.org/camera_calibration/Tutorials/MonocularCalibration

Could you try it?

eupedrosa, 2020-07-03:
I belive the Zed ROS driver uses its own calibration files and ignores the calibration files generated by ROS.
From the [manual](https://support.stereolabs.com/hc/en-us/articles/360011828773-How-do-I-recalibrate-my-ZED-stereo-camera-) the calibration files are located at `/usr/local/zed/settings/` (be carefull when changing this files)

aaguiar96, 2020-07-03:
Wait, I did not express my self correctly.
We are using the factory calibration from ZED, that do not has distortion coefficients.

The other day, when I went to the lab, I used to stereo calibration toolbox from ROS and obtained the following calibration
https://github.com/lardemua/atom/issues/179#issuecomment-644062098

We are not using it yet, we can try...

eupedrosa, 2020-07-03:
@miguelriemoliveira, can you compare the intrinsic parameters from the optimization with https://github.com/lardemua/atom/issues/179#issuecomment-644062098 ?

miguelriemoliveira, 2020-07-03:
I can but not right now. This weekend I should find some time to do so.

aaguiar96, 2021-03-05:
Closing this.

closed 2021-03-05 by aaguiar96