lardemua/atom #198: Revise normalization code in objective function

miguelriemoliveira, 2020-07-02:
Hi @eupedrosa ,

we must finish this sometime soon ...

miguelriemoliveira, 2020-07-18:
Hi @eupedrosa ,

I wanted to complete this. Can we work on this? When are you available?

eupedrosa, 2020-07-19:
Hi @miguelriemoliveira, I will be available starting tomorrow.

miguelriemoliveira, 2020-07-28:
So working with @eupedrosa last night on the agrob dataset (cameras + lidar3d) we came to the conclusion that:

1. Using the error without any normalization does not converge
2. Using just the message type normalization (multiply by 200 ) makes the optimization converge (good values, 0.6 pixels in cameras and 0.006 in metes)
3. Using just the pair collection sensor intra  normalization and inter normalization seems to make no difference at all ...

Results are strange, especially because of 3. The idea is to use a z-score but I am not sure it will work. 

@eupedrosa we have to talk to this again.

eupedrosa, 2020-07-30:
Hi @miguelriemoliveira and @aaguiar96,
I have some news about this.

Yesterday @miguelriemoliveira give the ideia of normalizing the residuals by the mean per message type. This works. All the other ideas were plain bad.

The best results that I got was **0.7676447357290376** pixels for both cameras and **0.0069623491671051332** meters for the velodyne. The optimization has the following options:
```python
options = {'ftol': 1e-6, 'xtol': 1e-6, 'gtol': 1e-6, 'diff_step': None, 'x_scale': 'jac'}
```
Yes @miguelriemoliveira, it provides better results with `x_scale`.

Another important detail:exclamation: 
I had to remove the *stepping* from the image corners. By, for example, using a step of 0.3 it was discarding a lot of information. This makes sense for the orthogonal residuals in PointCloud2 data but not for Images, specially if you consider that most chessboard detection has partial data.

eupedrosa, 2020-07-30:
This issue is addressed in commit 60eccedb9f1f4d1767118f716263724e4001255e.
Please update to the newest version for a nice boost.

miguelriemoliveira, 2020-07-31:
Hi @eupedrosa ,

great commit and post. Now it seems the normalization is a settled matter. Thats a big breakthrough!

Great work. Thanks.

I noticed you went into a minimalistic frenzy and removed all the report from the objective function. I will put it back with a verbose flag ok? I think it is important for debugging.

About this:

> I had to remove the stepping from the image corners. By, for example, using a step of 0.3 it was discarding a lot of information. This makes sense for the orthogonal residuals in PointCloud2 data but not for Images, specially if you consider that most chessboard detection has partial data.

You're right, makes sense. We should fix this in #201 .

miguelriemoliveira, 2020-07-31:
Hi @eupedrosa ,

Unless I did something wrong, the code you pushed was not working. You removed the step from the function **getPointsInPatternAsNPArray** but not from the creation of the residuals.

Did you test before commiting?

eupedrosa, 2020-07-31:
Yes I tested. I'm gonna check if I forgot to commit something.

eupedrosa, 2020-07-31:
Strange, with the `step` only on residual creation it was working. It was failing to you?

miguelriemoliveira, 2020-07-31:
I think its because I am using -sr 0.3 and you used something else?

miguelriemoliveira, 2020-07-31:
In any case, its fixed. Don't worry about it,

miguelriemoliveira, 2020-08-04:
Hi @eupedrosa ,

I think your 64c426acca56fc47ad3702c4f6d357256c136a89 breaks the calibrate executable. 
Can you check it out? Here's the error log.

```
Traceback (most recent call last):
  File "/home/mike/catkin_ws/src/calibration/atom/atom_calibration/scripts/calibrate", line 582, in <module>
    main()
  File "/home/mike/catkin_ws/src/calibration/atom/atom_calibration/scripts/calibrate", line 530, in main
    opt.startOptimization(optimization_options=options)
  File "/home/mike/workingcopy/OptimizationUtils/OptimizationUtils/OptimizationUtils.py", line 375, in startOptimization
    errors = self.errorDictToList(self.objective_function(self.data_models))
  File "/home/mike/catkin_ws/src/calibration/atom/atom_calibration/src/atom_calibration/calibration/objective_function.py", line 167, in objectiveFunction
    normalizer = data['normalizer']
KeyError: 'normalizer'
```

Don't forget to git pull since the reorganization #222 is ongoing. calibrate and the objective function are now in atom_calibration. 

For this reason, the <your_robot>_calibration packages must be reconfigured to generate new launch files, but I already did that for agrob so you can just git pull.

eupedrosa, 2020-08-04:
Hi @miguelriemoliveira.
Your commit 045381fca72dfe633ffc7449d52a17dd59be5906 deleted the code for the normalization that I pushed in 64c426acca56fc47ad3702c4f6d357256c136a89.

I will commit it again.

closed 2020-08-04 by eupedrosa

miguelriemoliveira, 2020-08-04:
> Hi @miguelriemoliveira <https://github.com/miguelriemoliveira>.
> Your commit 045381f
> deleted the code for the normalization that I pushed in 64c426a
>
> I will commit it again.

Oopps ... sorry

commit ed7a5b70 (2020-08-04): Fix #198.