lardemua/atom #199: Correct velodyne position on agrob URDF

aaguiar96, 2020-07-03:


aaguiar96, 2020-07-08:
Hi @miguelriemoliveira and @eupedrosa 

Been busy these days...
Today I'll post here the ZED length and the displacement between the two sensors measured with a hammer.
Then we can check the URDF and calibration right?

miguelriemoliveira, 2020-07-08:
Hi @aaguiar96 ,

In the email I received as notification I got (the original version of your post):

"Today I'll post here the ZED length and the displacement between the two sensors measured with a **hammer**."

much more interesting, don't know why you replaced by a ruler ...

aaguiar96, 2020-07-08:
You catch my first demoniac ideia, muahahah! :laughing:

aaguiar96, 2020-07-08:
Measured with a *hammer* :rofl: :+1: 

* Total ZED lenght: 17.2 cm
* Between lens: 10.5 cm
* Vertical displacement between sensors: 4.5 cm
* Horizontal displacement from velodyne center to ZED's extrema: 8 cm

miguelriemoliveira, 2020-07-08:
I just rerun a calibration. Same result:

[image]

I ran the command

`rosrun tf tf_echo c0_zed_right_camera_optical_frame c0_vlp16_frame`

and got:

```
At time 1594213972.402
- Translation: [-0.038, -0.063, -0.106]
- Rotation: in Quaternion [0.749, -0.190, 0.149, 0.617]
            in RPY (radian) [1.790, -0.476, -0.113]
            in RPY (degree) [102.547, -27.293, -6.492]

```

@aaguiar96 , are these numbers in agreement with your measurements?

aaguiar96, 2020-07-08:
Hi @miguelriemoliveira 

That depends. We have to look at the results carefully.
The frame `c0_zed_right_camera_optical_frame` is aligned to the right lens?

If so, from the measures we have that.
* Vertically:
   * From velodyne center do the more distant extrema of ZED we have 8 cm. ZED lenght = 17.2, between lens = 10.5. So from the velodyne center to the most distant lens we should have 8 - (17.2 - 10.5) / 2 = ~4.7 cm. 
* Horizontally:
   * Distance from zed to velodyne = 4.5 cm

The translation component is in what reference frame? x corresponds to horizontal displacement and y the vertical?

eupedrosa, 2020-07-08:
> Horizontal displacement from velodyne center to ZED's extrema: 8 cm

This means the velodyne is not centered.

aaguiar96, 2020-07-08:
> This means the velodyne is not centered.

Yes, it is not exactly centered.

miguelriemoliveira, 2020-07-08:
I am sorry, I get lost with all these numbers. Can't you create a collaborative way of drawing all these measurementes on top of a photograph?

miguelriemoliveira, 2020-07-08:
Hi @aaguiar96 ,

I think the translations are in the zed_camera_right optical frame, so z forward, x left to right.

> The frame c0_zed_right_camera_optical_frame is aligned to the right lens?

yes

aaguiar96, 2020-07-08:
A college of mine was doing the measures and I misunderstand them.
The velodyne and zed are centered horizontally.

Here are the correct measures

[image]

eupedrosa, 2020-07-08:
Are you sure this are the measurements for the Zed camera? The distance baseline is 12 cm not 10.5cm.

aaguiar96, 2020-07-09:
Hi @eupedrosa 

You're right. I think my college measured from the interior border of the lens.

eupedrosa, 2020-07-09:
@miguelriemoliveira, I have bad news for you!

I did a collection and I got this results:
[image]

At the lower right you have the transformation between cameras. Not bad, I believe.

And here is from another perspective:
[image]

Now at the lower right you have the transformation between the right camera and the velodyne. @aaguiar96, can you check the values? See if is has realistic values.

I say this is bad news for @miguelriemoliveira because, to get this calibration I removed the orthogonal residuals and the normalization. The only thing that I keept was a `meter_to_pixel_factor = 10`.

Now seriously, @miguelriemoliveira  and @aaguiar96, is this a resonable result?

aaguiar96, 2020-07-09:
@miguelriemoliveira  and I started  a meeting now. Do you want to join?

If you can't, we are going to see those results and come back to you later.

aaguiar96, 2020-07-09:
New small dataset for URDF tests.

[datasets.zip](https://github.com/lardemua/atom/files/4897815/datasets.zip)

aaguiar96, 2020-07-09:
Closing. Working on #203.

closed 2020-07-09 by aaguiar96