lardemua/atom #201: Sampling residuals?

miguelriemoliveira, 2020-07-08:
Hi @eupedrosa and @aaguiar96 ,

I was thinking about the long time it takes to optimize agrob. One simple idea is to sample the residuals.

For the velodyne with 2 cameras + lidar we now have 6000+ residuals ... that may be too much. 

What do you think about a parameter that just samples the total of residuals selecting only 50% or 10% of the total to compute the objetive function?

A more complex idea would be to have this sampling be reduced as the optimization progresses (similar to the effect of the learning rate in deep learning). But for now, a simple sample equal for all could work...

What do you think?

eupedrosa, 2020-07-08:
Hi @miguelriemoliveira,

I agree with the *sampling* idea. Note that what we previously did is decimation, not sampling. I understand sampling as random selection.

> A more complex idea would be to have this sampling be reduced as the optimization progresses (similar to the effect of the learning rate in deep learning). But for now, a simple sample equal for all could work...

At start we should have more or less samples? And how do we qualify the optimization progress?

miguelriemoliveira, 2020-07-08:
Hi @eupedrosa ,

thanks for the imput.

> At start we should have more or less samples? 

Less.

> And how do we qualify the optimization progress?

Number of iterations

miguelriemoliveira, 2020-07-18:
Right now sampling is being done by selecting a step with which to iterate through the labelled points.

This is not interesting because it is not really random. Will try to create a randomized process which occurs at startup, and then only the randomly selected labels are used.

miguelriemoliveira, 2020-08-03:
@eupedrosa said he will work on this.

eupedrosa, 2020-08-03:
Hi @miguelriemoliveira.
I already have a working version. But there is a catch. Being "random" the results are not deterministic, it this Ok for you or should I add the possibility to set a *seed* for when a deterministic outcome is desired?

miguelriemoliveira, 2020-08-03:
Setting the seed is helpfull for debugging purposes.

closed 2020-08-04 by eupedrosa

miguelriemoliveira, 2020-08-04:
Hi @eupedrosa ,
I tested again several combinations of agrob with varying number of collections and other parameters. It seems to be working fine. I did not observe again 100+ iterations.

Lets not worry about it.

commit c2373c08 (2020-08-04): Use random sampling to select orthognal residuals. (Closes #201)