lardemua/atom #202: Reformulate lidar labelling code and objective function to use dictionaries instead of np arrays

miguelriemoliveira, 2020-07-08:
Hi @aaguiar96 ,

in the labelling you created np arrays inside the dataset dictionary (not sure, it seems here you are using lists ... ).

https://github.com/lardemua/atom/blob/d26fb97991af6c94b43160af7fa102f0fd7066f0/atom_core/scripts/calibrate#L245-L312

I think this breaks the writting to a json results file later on. Can you change (and double check everything continues to work) to a dictionary as we do here?

https://github.com/lardemua/atom/blob/d26fb97991af6c94b43160af7fa102f0fd7066f0/atom_core/src/atom_core/patterns.py#L122-L128

and then in the objective function we convert to nparray here 

https://github.com/lardemua/atom/blob/d26fb97991af6c94b43160af7fa102f0fd7066f0/atom_core/src/atom_core/objective_function.py#L140-L151

You can put the maximum number of function evaluations to 1 to quicly test the writting of the json file ...

aaguiar96, 2020-07-08:
Ok @miguelriemoliveira 

I'll do that as soon as I can.

miguelriemoliveira, 2020-07-08:
I just tested, we do get this error when writting the json.

```
ile "/usr/lib/python2.7/json/encoder.py", line 184, in default
    raise TypeError(repr(o) + " is not JSON serializable")
TypeError: array([ 1.77318919, -0.02414092,  0.40941092,  1.        ]) is not JSON serializable

```

aaguiar96, 2020-07-10:
Hi @miguelriemoliveira and @eupedrosa 

Done! :)

[image]

[results.zip](https://github.com/lardemua/atom/files/4903776/results.zip)

Can I close it?

miguelriemoliveira, 2020-07-10:
Great.  And the optimization continues to work? Did you test? If so you can close.

aaguiar96, 2020-07-10:
Yes.
Closing then.

closed 2020-07-10 by aaguiar96