lardemua/atom #203: Run tests of agrob calibration with new xacro

miguelriemoliveira, 2020-07-09:
After #199, run new tests

aaguiar96, 2020-07-09:
New agrod dataset for URDF tests.

[datasets.zip](https://github.com/lardemua/atom/files/4897817/datasets.zip)

eupedrosa, 2020-07-09:
Any new conclusion?

aaguiar96, 2020-07-10:
Hi @eupedrosa 

We fixed the URDF and add a new parameter to sample the residuals. Then we collected a new dataset with this fix.
We've done several tests and concluded that in almost all the tests, the cloud points start on the left of the patterns. This leads to a solution where the velodyne is always on the right of the ZED camera.
We did not figured out the problem yet.

aaguiar96, 2020-07-10:
Hi @eupedrosa and @miguelriemoliveira 

@eupedrosa at the end of the last meeting, me and @miguelriemoliveira were a bit lost regarding the next step. What to do to overcome these error of centimeters?...
I think the three of us should schedule a meeting to analyze the results with detail and decide what's the next step.

What do you say?

miguelriemoliveira, 2020-07-10:
Can we talk in 20 mins?

aaguiar96, 2020-07-10:
Yes!

eupedrosa, 2020-07-10:
So. it should start now, right?

miguelriemoliveira, 2020-07-10:
I need an extra 5 to 8 mins ... sorry.

miguelriemoliveira, 2020-07-17:
This is done.

closed 2020-07-17 by miguelriemoliveira