lardemua/atom #204: Better error handling of missing urdf file from config.yml

eupedrosa, 2020-07-10:
The `description_file` is not validated for existence and when it happens two things occurs:
1. It throws a parsing error.
2. If a `/tmp/description.urdf` already exists, the script uses this file, which results in undefined behavior.

miguelriemoliveira, 2020-07-10:
Hi,

I will try to pick this up later on.
*  Miguel Riem de Oliveira                                             *
email:    mriem@ua.pt    |    m.riem.oliveira@gmail.com
office: 22.2.18                 |    ext: 23887
Dep. de Engenharia Mecânica, Universidade de Aveiro
*Campus Universitário de Santiago, 3810-193  Portugal*

miguelriemoliveira, 2020-07-18:
Hi @eupedrosa ,

looked into this. We are checking if the file exists inside the urireader function, take a look

https://github.com/lardemua/atom/blob/076890dc6a76037c9a34e205c33df22dfbc9c98c/atom_core/src/atom_core/utilities.py#L176

I am not sure if you meant something else ... should I close?

closed 2020-07-28 by miguelriemoliveira