lardemua/atom #205: Results to get for agrob

miguelriemoliveira, 2020-07-13:
https://docs.google.com/document/d/1f6X_QPLspYcNilqBSexrJ4Vvfr0tOQhFCofg6hkGYj4/edit?usp=sharing

aaguiar96, 2020-07-13:
*Extrinsic Calibration between Camera and LiDAR Sensors by Matching Multiple 3D Planes*

* Extrinsic calibration of an omnidirectional visual system composed of 6 cameras and a 3D LiDAR (velodyne)
Experiments done on simulated and real data;
* Using simulated data:
   * The calibration results were automatically compared with the ground truth to generate the simulation
* Using real data:
   * The framework was not compared with any other present in the literature
   * The calibration results were compared with the ones measured manually on the physical system
   * The tables presented present this comparison, as well as the results for different camera focal lengths and distances from the sensors to the chessboard

aaguiar96, 2020-07-13:
*Automatic extrinsic calibration between a camera and a 3D Lidar using 3D point and plane correspondences*

* Monocular-Camera-to-LiDAR calibration (uses velodyne).
* Experiments done on simulated and real data;
* Using simulated data:
   * The calibration results were automatically compared with the ground truth to generate the simulation
* Using real data:
   * The authors do not use any quantitative metric
   * Instead, the LiDAR ranges are projected onto the image using the obtained extrinsic calibration, and the projections are visually qualified by the fitting of the ranges on the image objects, namely, on the chessboard pattern.

aaguiar96, 2020-07-13:
*LiDAR-Camera Calibration using 3D-3D Point correspondences*

* Multi-Camera-to-LiDAR calibration (uses velodyne and ZED).
* Experiments done on real data
   * Extrinsic calibration compared with tape measurements. Here rotations are not evaluated.
   * Extrinsic calibration also evaluated using the average error between the camera PCL and LiDAR PCL
   * ICP and Kabsch calibration result exposed.
* The authors do not provide any discussion about the performance of their method nor any critical comparison with ICP and Kabsch.

aaguiar96, 2020-07-13:
*A Novel Multi-Planar LIDAR and Computer Vision Calibration Procedure Using 2D Patterns for Automated Navigation*

TODO

aaguiar96, 2020-07-13:
*An Interactive LiDAR to Camera Calibration*

TODO

aaguiar96, 2020-07-13:
*Reflectance Intensity Assisted Automatic and Accurate Extrinsic Calibration of 3D LiDAR and Panoramic Camera Using a Printed Chessboard*

TODO

aaguiar96, 2020-07-13:
*Extrinsic calibration between a multi-layer lidar and a camera*

TODO

eupedrosa, 2020-07-23:
Hi @aaguiar96. Any news on this front?

aaguiar96, 2020-07-23:
Hi @eupedrosa 

Unfortunately no... This week I have my PhD Thesis Plan defense, and also a lot of work on the INESC side.
I'll try to focus more on this next week.

eupedrosa, 2020-07-23:
Ok, keep with the good work and good luck.
But I can guess (from previous experiences) the defense will go great.

aaguiar96, 2020-07-23:
> But I can guess (from previous experiences) the defense will go great.

Thanks! Hope so. :)

miguelriemoliveira, 2020-07-27:
#206 - ILCC
#211 - Lidar_camera_calibration_package
#216 - Autoware - Lidar to camera

aaguiar96, 2020-07-27:
*Automatic Extrinsic Calibration for Lidar-Stereo Vehicle Sensor Setups*

TODO

aaguiar96, 2020-09-02:
Hi @miguelriemoliveira and @eupedrosa 

Can you confirm this table?
This is the summary of the comparisons that will be done on results. Did I miss anything?
This table does not account for the characterization.

[image]

miguelriemoliveira, 2020-09-03:
I think that's it. I would call it full system instead of simultaneous.

aaguiar96, 2020-09-03:
Ok thanks @miguelriemoliveira 

I recorded a dataset with 58 collections yesterday.
Today I'll test it.

aaguiar96, 2020-09-08:
Hi @eupedrosa and @miguelriemoliveira 

What do you think of this template for the characterization experiments?
Here is an example of the experiments to evaluate the impact of the number of collections.
The idea is to fix all the other parameters except the one we will evaluate.

[image]

aaguiar96, 2020-09-09:
The tables are actually looking like this now:

[image]

However (as represented on table 2), I want to vary the number of collections with no dependencies on other parameters.
To do so, I activate the `remove partial detections` parameter.
With this, I'm getting this error:

```
Removing collection 8 -> pattern was not found in sensor left_camera (must be found in all sensors).
Partial detection removed: label from collection 11, sensor left_camera
Partial detection removed: label from collection 11, sensor right_camera
Partial detection removed: label from collection 10, sensor left_camera
Partial detection removed: label from collection 10, sensor right_camera
Partial detection removed: label from collection 13, sensor right_camera
Partial detection removed: label from collection 1, sensor left_camera
Partial detection removed: label from collection 0, sensor left_camera
Partial detection removed: label from collection 3, sensor left_camera
Partial detection removed: label from collection 5, sensor left_camera
Partial detection removed: label from collection 5, sensor right_camera
Partial detection removed: label from collection 4, sensor left_camera
Partial detection removed: label from collection 7, sensor left_camera
Partial detection removed: label from collection 7, sensor right_camera
Partial detection removed: label from collection 6, sensor left_camera
Partial detection removed: label from collection 6, sensor right_camera
Partial detection removed: label from collection 9, sensor left_camera
Partial detection removed: label from collection 9, sensor right_camera
Collection 11: Chessboard not detected by sensor left_camera
Collection 11: Chessboard not detected by sensor right_camera
Traceback (most recent call last):
  File "/home/andreaguiar/Source/catkin_ws/src/atom/atom_calibration/scripts/calibrate", line 606, in <module>
    main()
  File "/home/andreaguiar/Source/catkin_ws/src/atom/atom_calibration/scripts/calibrate", line 137, in main
    dataset['patterns'] = patterns.createPatternLabels(args, dataset)  # TODO: Solve this strange dependency.
  File "/home/andreaguiar/Source/catkin_ws/src/atom/atom_calibration/src/atom_calibration/calibration/patterns_config.py", line 223, in createPatternLabels
    raise ValueError('Collection ' + collection_key + ' could not find pattern.')
ValueError: Collection 11 could not find pattern.
```

Do you know what's happening @miguelriemoliveira and @eupedrosa?

eupedrosa, 2020-09-09:
The labels of collection 11 are remove, then collection 11 is used to detect the pattern (maybe for first guess?). Obviously, it fails.

aaguiar96, 2020-09-09:
> The labels of collection 11 are remove, then collection 11 is used to detect the pattern (for pattern characterization). Obviously, it fails.

Ok, but so this is a bug since removed collections should not be used further, right?

eupedrosa, 2020-09-09:
Yes, it is a bug. Create an issue and assign it to @miguelriemoliveira ;)

miguelriemoliveira, 2020-09-09:
Thank you very much @eupedrosa ... you are very nice :-)

I can take a look tonight (@aaguiar96 please create the issue), but perhaps now you could try to discard that collection 11 using the -csf flag ...

miguelriemoliveira, 2020-09-09:
The tables look nice. I like them.

eupedrosa, 2020-09-09:
> Thank you very much @eupedrosa ... you are very nice :-)

No problemo...

About the tables.. I think that a column for the standard deviation may not be the best way. Usually **means** are presented like this: `mean +- std` (you can use  \pm in latex for the +-) . This way you remove two unecessary columns while increasing readability.

aaguiar96, 2020-09-09:
> About the tables.. I think that a column for the standard deviation may not be the best way. Usually **means** are presented like this: `mean +- std` (you can use \pm in latex for the +-) . This way you remove two unecessary columns while increasing readability.

Makes sense! I'll change that.

> I can take a look tonight (@aaguiar96 please create the issue), but perhaps now you could try to discard that collection 11 using the -csf flag ...

I cannot @miguelriemoliveira. If I remove collection 11, it happens with other collections.

eupedrosa, 2020-09-09:
> I cannot @miguelriemoliveira. If I remove collection 11, it happens with other collections.

@aaguiar96, are you sure there is collections with full detections, i.e. without partial detections ?

miguelriemoliveira, 2020-09-09:
Yes, that could be a problem ... how many non-partial detections we have?

aaguiar96, 2020-09-09:
I think atom outputs all the non partial collections detected right?

```
Partial detection removed: label from collection 11, sensor left_camera
Partial detection removed: label from collection 11, sensor right_camera
Partial detection removed: label from collection 10, sensor left_camera
Partial detection removed: label from collection 10, sensor right_camera
Partial detection removed: label from collection 13, sensor right_camera
Partial detection removed: label from collection 1, sensor left_camera
Partial detection removed: label from collection 0, sensor left_camera
Partial detection removed: label from collection 3, sensor left_camera
Partial detection removed: label from collection 5, sensor left_camera
Partial detection removed: label from collection 5, sensor right_camera
Partial detection removed: label from collection 4, sensor left_camera
Partial detection removed: label from collection 7, sensor left_camera
Partial detection removed: label from collection 7, sensor right_camera
Partial detection removed: label from collection 6, sensor left_camera
Partial detection removed: label from collection 6, sensor right_camera
Partial detection removed: label from collection 9, sensor left_camera
Partial detection removed: label from collection 9, sensor right_camera
```

aaguiar96, 2020-09-15:
Hi @miguelriemoliveira and @eupedrosa 

I'm at the lab today, and I recorded two rosbags with all the info we wanted.
Here:

```
topics:      /camerat265/fisheye1/camera_info               8992 msgs    : sensor_msgs/CameraInfo
             /camerat265/fisheye1/image_raw/compressed      8993 msgs    : sensor_msgs/CompressedImage
             /camerat265/fisheye2/camera_info               8993 msgs    : sensor_msgs/CameraInfo
             /camerat265/fisheye2/image_raw/compressed      8993 msgs    : sensor_msgs/CompressedImage
             /tf                                          104592 msgs    : tf2_msgs/TFMessage          (7 connections)
             /tf_static                                        4 msgs    : tf2_msgs/TFMessage          (4 connections)
             /velodyne_points                               2971 msgs    : sensor_msgs/PointCloud2
             /zed/zed_node/left_raw/camera_info             3813 msgs    : sensor_msgs/CameraInfo
             /zed/zed_node/left_raw/image/compressed        2579 msgs    : sensor_msgs/CompressedImage
             /zed/zed_node/point_cloud/cloud_registered      628 msgs    : sensor_msgs/PointCloud2
             /zed/zed_node/right_raw/camera_info            3813 msgs    : sensor_msgs/CameraInfo
             /zed/zed_node/right_raw/image/compressed       3243 msgs    : sensor_msgs/CompressedImage
```

I will now test, and start generating the results.

aaguiar96, 2020-09-15:
Some notes and questions:

* I will have to add some dummy code to the data collector to save a .pcd file extracted from `/zed/zed_node/point_cloud/cloud_registered`, right?
* To evaluate the impact of the initial guess in the calibration, how should I do it? Can I do it in a single training dataset?

aaguiar96, 2020-09-15:
Here is a snapshot of one of the rosbags (30 secs): https://drive.google.com/file/d/14IWXlIE-rCdrrwqrKzgANvkNZZUkRRB_/view?usp=sharing

eupedrosa, 2020-09-15:
Hello @aaguiar96.

> I will have to add some dummy code to the data collector to save a .pcd file extracted from /zed/zed_node/point_cloud/cloud_registered, right?

As I understand, you want to register this point cloud with the point cloud generated by the velodyne, right?
Maybe it should not be part of the data collector. It is an external tool.

> To evaluate the impact of the initial guess in the calibration, how should I do it? Can I do it in a single training dataset?

I think so, maybe you could manipulate the initial pose of the sensor in the calibrate script.

---

By the way. I **hacked** an autocollector. The data is synchronized with a maximum difference of 10 milliseconds.
Yout can get it [here](https://drive.google.com/file/d/1M79ePiqr341QugLUdvIEjb2Xa3uEaK8p/view?usp=sharing) !!

Can you check if the results are good?

aaguiar96, 2020-09-15:
> By the way. I **hacked** an autocollector. The data is synchronized with a maximum difference of 10 milliseconds.
> Yout can get it [here](https://drive.google.com/file/d/1M79ePiqr341QugLUdvIEjb2Xa3uEaK8p/view?usp=sharing) !!

Hi @eupedrosa 
What do you mean by "hacked an autocollector"?
Did you get the 44 collections from the 30 secs of bagfile?

eupedrosa, 2020-09-15:
> What do you mean by "hacked an autocollector"?

A script that collects the data without our intervention. It receives a rosbag and a config file. An then it collects.

> Did you get the 44 collections from the 30 secs of bagfile?

Nop, from a 5 minutes bagfile. From the recent bagfile of 30 seconds I got 0 collections....
The CameraInfo messages have no stamp...

aaguiar96, 2020-09-15:
> A script that collects the data without our intervention. It receives a rosbag and a config file. An then it collects.
> 
> >

Great! 👍 

> Nop, from a 5 minutes bagfile. From the recent bagfile of 30 seconds I got 0 collections....
> The CameraInfo messages have no stamp..

Ok, I'll repeat the data collection and see what's going on. Thanks!

miguelriemoliveira, 2020-09-15:
Hi ,

> To evaluate the impact of the initial guess in the calibration, how should I do it? Can I do it in a single training dataset?

It is not a question of if you can, you must do it. Because otherwise you could not ensure you were using the same images and point clouds and could not say that the difference in results is solely due to the change in the initial guess.

I suggest you get one dataset with a good first guess (the best you can get).

Then you add noise to the adequate transformations in the json (or when loading) so that you have something like:

perfect first guess, perfect first guess with 1% noise, perfect first guess 2% noise, ... , perfect first guess with 20% noise.

What do you think?

miguelriemoliveira, 2020-09-15:
> I will have to add some dummy code to the data collector to save a .pcd file extracted from /zed/zed_node/point_cloud/cloud_registered, right?
> 
> As I understand, you want to register this point cloud with the point cloud generated by the velodyne, right?

Well, the problem is it must be the data collector to save these zed point clouds to make sure they are from the same time as the collections. I will work on it tonight. We will need this functionality for the rgbd cameras I think.

> Maybe it should not be part of the data collector. It is an external tool.

aaguiar96, 2020-09-21:
Closing this issue.
Opening a fresh one to discuss the paper results.

closed 2020-09-21 by aaguiar96