lardemua/atom #206: Test https://github.com/mfxox/ILCC and compare with our approach

aaguiar96, 2020-07-15:


aaguiar96, 2020-07-28:
I @miguelriemoliveira  and @eupedrosa 

This is not a ROS package. It requires the pointcloud data as `.pcd` files.
Can you confirm this? Is this a problem?...

miguelriemoliveira, 2020-07-28:
Hi @aaguiar96 ,

From their readme:

> Make a folder for example named as DATA and make the image and point cloud folders DATA/img and DATA/pcd respectively.
> 
> Put panoramic images into DATA/img and point cloud files into DATA/pcd. The files should be named like 00XX.png or 00XX.csv.
> 
> cd DATA and copy config.yaml to DATA and modify config.yaml according to your situation.
> 
> Corner detection from images.

So I guess they do need the pcd files. If you want I can try to generate them ...

aaguiar96, 2020-07-28:
It would be interesting to extract the pcd files from our json files. In this way we would have exacly the same data...

I can try to do it. :)

miguelriemoliveira, 2020-07-28:
OK, I worked on this two weeks ago #155 but abandoned the idea.

But I think me problem was with loading back the point cloud, which you don't need.

You can also think of saving a ply and then run the pcl_ply2pcd executable.

aaguiar96, 2020-07-28:
So, at the end of the calibration I should generate N .ply /.pcd files right? One for each collection.

miguelriemoliveira, 2020-07-28:
Right.
*  Miguel Riem de Oliveira                                             *
email:    mriem@ua.pt    |    m.riem.oliveira@gmail.com
office: 22.2.18                 |    ext: 23887
Dep. de Engenharia Mecânica, Universidade de Aveiro
*Campus Universitário de Santiago, 3810-193  Portugal*

aaguiar96, 2020-07-28:
@miguelriemoliveira  do you know this? https://github.com/cburbridge/python_pcd

miguelriemoliveira, 2020-07-28:
Hi @aaguiar96 ,

yes. There is a problem with that one which is all point clouds must be xyz only, no rgb.

In any case for the lidar perhaps it is enough.

aaguiar96, 2020-07-29:
Hi @miguelriemoliveira  and @eupedrosa 

So, ILCC has three main scrips:

1. One to detect the corners of the pattern on images
2. One to detect the corners of the pattern on the pointcloud
3. The optimization script

I already run 1. and 2. with success.
I'm stuck at three... It requires `opengv` but they use python2. I can only install opengv for python3...
Any ideas? Does someone know how to install opengv for python2 or any other workaround?

eupedrosa, 2020-07-29:
How did you try to install `opengv`, apt or pip?

aaguiar96, 2020-07-29:
> How did you try to install `opengv`, apt or pip?

None of that, I think it has to be from source.
But I came up with a solution. I made a version of the optimization script for python3, and it works...

eupedrosa, 2020-07-29:
Great, so ILLC is working, right?

aaguiar96, 2020-07-29:
> Great, so ILLC is working, right?

It is, but the optimization results are really bad. I am implementing some visualization tools to see what's going on.

eupedrosa, 2020-07-29:
Try running with their dataset. They have one on GitHub.

aaguiar96, 2020-07-29:
The marker cannot be found by the pointcloud corner detector...
I think we need a chessboard.

I can try to go to the lab and record the bagfile with the pointcloud of zed for ICP and with both patterns.
I just have to print the chessboard. What do you think?

miguelriemoliveira, 2020-07-29:
I think @eupedrosa is right. You should try first with their dataset just to see if it runs well.

aaguiar96, 2020-07-30:
Tested the calibration with ILCC's perspective dataset and it works.
Here's some pictures of the reprojection:

[image]
[image]
[image]

aaguiar96, 2020-07-30:
So, what should I do next?...

eupedrosa, 2020-07-30:
This means the algorithm is working. Now you should try making it work with our datasets. If I remember correctly it was not working right? 

@miguelriemoliveira and @aaguiar96, should we meet to discuss this?

aaguiar96, 2020-07-30:
> If I remember correctly it was not working right?

Yes. I'm almost certain that this is happening because the framework does not support charuco.

> @miguelriemoliveira and @aaguiar96, should we meet to discuss this?

Fine by me!

eupedrosa, 2020-07-30:
> Yes. I'm almost certain that this is happening because the framework does not support charuco.

This is a problem... We need partial detections to make it work, otherwise it is a pain-in-the-a** to detect the chessboard.

aaguiar96, 2020-07-30:
> This is a problem... We need partial detections to make it work, otherwise it is a pain-in-the-a** to detect the chessboard.

Eheh :-)

Can't we have a charuco and a chessboard side-by-side simultaneously in the same dataset? I know it is not ideal, but...

eupedrosa, 2020-07-30:
OpenCVs's chessboard detector works with a charuco board, therefore, the same pattern can be detected by both detectors. The problem is that the chessboard detector needs to detect all corners for a valid detection. Unless, the chessboard detector also returns partial detections.

aaguiar96, 2020-07-30:
PointCloud loaded from pcd file generated on atom. It has intensity.

[image]

So, I'll look into ILCC's code to see what's going one.
@eupedrosa can you send me the 50 collection dataset please?

eupedrosa, 2020-07-30:
> @eupedrosa can you send me the 50 collection dataset please?

Check your email.

aaguiar96, 2020-07-30:
> Check your email.

Thanks! :)

eupedrosa, 2020-07-30:
@aaguiar96, don't forget to update your `OptimizationUtils` repo, it has changes that speeds ups the optimization.

eupedrosa, 2020-07-30:
@aaguiar96 and @miguelriemoliveira, I believe this solution will not work with the charuco. If it uses the intensity to find the chessboard it will not work, because the white spaces are occupied with aruco markers. It will never detect the corners with the charuco, it has black where it expects white.

aaguiar96, 2020-07-31:
> , I believe this solution will not work with the charuco. If it uses the intensity to find the chessboard it will not work, because the white spaces are occupied with aruco markers. It will never detect the corners with the charuco, it has black where it expects white.

Makes sense... So, should we use a chessboard?

aaguiar96, 2020-09-21:
Can we close this for now?

miguelriemoliveira, 2020-09-21:
Yes

closed 2020-09-21 by miguelriemoliveira