lardemua/atom #213: Objective Function: Profiling

eupedrosa, 2020-07-21:
Sorry @miguelriemoliveira, I did not resist on doing a profiling analysis of my own.

I did a profiling using a *joint hand base* dataset using the *old* `general_hand_eye.py` and the *new* `calibrate`.
Both solution were able to solve the problem, with the same termination criteria and visualization disabled.
Here are the result (just for the *objective function*)

1. `general_hand_eye.py`
[image]

2. `calibrate`
[image]

### Short Analysis of *new* vs *old*
- Is almost 6x slower
- A lot of time is used on the str `split` method
- `deepcopy` (orange block) and `mean` (cyan block) represents almost 14% of consumed time.
- More than 50% of time is spent doing cyles, if blocks, list comprehension, etc. In contrast, the *old* version uses ~10% of the time doing this.
- The `getTransform` is faster than `tf.py` used in the *old* version.

The is huge room for improvement.

miguelriemoliveira, 2020-07-21:
Hi Eurico,

I will take a look and try to improve. I will try to work on this tonight.

Miguel

eupedrosa, 2020-07-22:
Hi @miguelriemoliveira.
Commits a7b97200838fe6f3521da152a78ca58e4ee7d57c and c76b5c74dc504661988f73f4b962259bf480d8a6 address this issue.

I'm sorry if I am stealing your joy.  But I needed to do some programming ...

miguelriemoliveira, 2020-07-22:
Thanks @eupedrosa . Just to be clear. We still need to improve the data format in the dataset dictionary right?

I will do a profiling which is by code line, it is easier to optimize looking at that information.

I will work on this Friday. 

Another important thing: can you please test the eye-to-base? Because we still don't know if it works after the fusion of codes. If we go about optimizing a code now, then if it does not work we don't know if it is the optimization's fault or the fusion's fault.

eupedrosa, 2020-07-22:
> Just to be clear. We still need to improve the data format in the dataset dictionary right?

In my opinion, yes! 

I was able to get a speedup of ~4x, but there is room for more.
[image]

> Another important thing: can you please test the eye-to-base? Because we still don't know if it works after the fusion of codes. If we go about optimizing a code now, then if it does not work we don't know if it is the optimization's fault or the fusion's fault.

This is the only one that is missing, the other 2 are working. Eye-on-hand and Joint-hand-base. My educated guess is that it will work. But, I will do the calibration...

miguelriemoliveira, 2020-07-22:
About this: 

> We still need to improve the data format in the dataset dictionary right?

1. The problem is with the information being explicit and clear (although this is a fuzzy criteria) 
2. The ability to read / write from / to json files.

eupedrosa, 2020-07-22:
> The problem is with the information being explicit and clear (although this is a fuzzy criteria)

We need to do a profiling session and talk we the data in front of us. Do you think it is possible to do it next week?

> The ability to read / write from / to json files

I am not worried about this, there are ways to solve it.

miguelriemoliveira, 2020-07-22:
Next week yes, but perhaps Friday night?

*  Miguel Riem de Oliveira                                             *
email:    mriem@ua.pt    |    m.riem.oliveira@gmail.com
office: 22.2.18                 |    ext: 23887
Dep. de Engenharia Mecânica, Universidade de Aveiro
*Campus Universitário de Santiago, 3810-193  Portugal*

eupedrosa, 2020-07-22:
For me, friday night is OK.

miguelriemoliveira, 2020-07-28:
I think now we have a good tool to see how much time every step of the objective function takes.

To run the profile just run the usual call to calibrate adding --profile_objective_function or -pof.

Here's an example of the output.

```
First call of Objective Function (no cache)

Timer unit: 1e-06 s

Total time: 0.585836 s
File: /home/mike/catkin_ws/src/calibration/atom/atom_core/src/atom_core/objective_function.py
Function: objectiveFunction at line 143

Line #      Hits         Time  Per Hit   % Time  Line Contents
==============================================================
   143                                           def objectiveFunction(data):
   144                                               """
   145                                               Computes the vector of residuals. There should be an error for each stamp, sensor and chessboard tuple.
   146                                               The computation of the error varies according with the modality of the sensor:
   147                                                   - Reprojection error for camera to chessboard
   148                                                   - Point to plane distance for 2D laser scanners
   149                                                   - (...)
   150                                                   :return: a vector of residuals
   151                                               """
   152                                               # print('Calling objective function.')

```

miguelriemoliveira, 2020-07-28:
Closing. Profile works fine. We can now thing about optimizing but will open an issue for that.

closed 2020-07-28 by miguelriemoliveira