lardemua/atom #216: Test autoware package

aaguiar96, 2020-07-27:
https://github.com/Autoware-AI/utilities/tree/master/autoware_camera_lidar_calibrator

aaguiar96, 2020-07-28:
I started with `autoware` package.
It's being difficult to make it run...

I was able to install all the dependencies and got the intrinsics calibration running - *charuco is not detected*.
Now, I am trying to run the extrinsic calibrator, but it is not easy...

I'm stuck here:
```
OpenCV Error: Unknown error code -49 (Input file is empty) in cvOpenFileStorage, file /home/andre-criis/opencv_build/opencv/modules/core/src/persistence.cpp, line 4422      
terminate called after throwing an instance of 'cv::Exception'
what():  /home/andre-criis/opencv_build/opencv/modules/core/src/persistence.cpp:4422: error: (-49) Input file is empty in function cvOpenFileStorage
```

I'll let you know if I can make it work.

miguelriemoliveira, 2020-07-28:
Hum, did you not have to give a file as input? Its complaining the file is empty ...

aaguiar96, 2020-07-28:
Yes, I did... I created a dummy yaml file with the ZED left camera calibration parameters.

Still not working...

aaguiar96, 2020-09-21:
Can we close this for now?

closed 2020-09-21 by aaguiar96