lardemua/atom #219: Cache operations in objective function

miguelriemoliveira, 2020-07-28:
@eupedrosa that's for you. 

Ideally we should have a single function which would cache all the stuff we need ...

miguelriemoliveira, 2020-07-28:
Hi @eupedrosa 

I found this:

https://levelup.gitconnected.com/faster-code-with-python-caching-8da6e8a92ae9

I think we could use it in our objective function. Do you want to discuss it?

miguelriemoliveira, 2020-07-28:
I think this is going much faster. @eupedrosa  can you test and tell me what you think?

miguelriemoliveira, 2020-07-29:
Ok, after many twists and turns I think I got it. @eupedrosa you want to take a look?

miguelriemoliveira, 2020-07-30:
To profile:

https://github.com/rkern/line_profiler

miguelriemoliveira, 2020-07-30:
Objective function profile for a agrob full sensor calibration with 20 collections, i.e.:

`calibrate -json /home/mike/datasets/agrob/agrob_01_07_2020/data_collected.json -csf 'lambda x: int(x) < 20' -ssf 'lambda name: name in ["left_camera","right_camera", "vlp16"]' -oi -rv -si -sr 0.3`

Right now, caching is only being done for functions **getPointsInPatternAsNPArray** and **getPointsInPatternAsNPArray**.

**WITHOUT CACHING**

```
Timer unit: 1e-06 s

Total time: 142.933 s
File: /home/mike/catkin_ws/src/calibration/atom/atom_core/src/atom_core/objective_function.py
Function: objectiveFunction at line 171

Line #      Hits         Time  Per Hit   % Time  Line Contents
==============================================================
   171                                           @profile
   172                                           def objectiveFunction(data):
   173                                               """
   174                                               Computes the vector of residuals. There should be an error for each stamp, sensor and chessboard tuple.
   175                                               The computation of the error varies according with the modality of the sensor:
   176                                                   - Reprojection error for camera to chessboard
   177                                                   - Point to plane distance for 2D laser scanners
   178                                                   - (...)
   179                                                   :return: a vector of residuals
   180                                               """
   181                                               # print('Calling objective function.')
   182                                           
   183                                               # Get the data from the model
   184       246        884.0      3.6      0.0      dataset = data['dataset']
   185       246        631.0      2.6      0.0      patterns = data['dataset']['patterns']
   186       246        550.0      2.2      0.0      args = data['args']
   187       246        607.0      2.5      0.0      if args['view_optimization'] or args['ros_visualization']:
   188       246        551.0      2.2      0.0          dataset_graphics = data['graphics']
   189                                           
   190       246        534.0      2.2      0.0      r = {}  # Initialize residuals dictionary.
   191      5166      15191.0      2.9      0.0      for collection_key, collection in dataset['collections'].items():
   192     19680      56647.0      2.9      0.0          for sensor_key, sensor in dataset['sensors'].items():
   193                                           
   194     14760      44709.0      3.0      0.0              if not collection['labels'][sensor_key]['detected']:  # chess not detected by sensor in collection
   195                                                           continue
   196                                           
   197     14760      42203.0      2.9      0.0              if sensor['msg_type'] == 'Image':
   198      9840      46361.0      4.7      0.0                  tic()
   199                                           
   200                                                           # Get the pattern corners in the local pattern frame. Must use only corners which have -----------------
   201                                                           # correspondence to the detected points stored in collection['labels'][sensor_key]['idxs'] -------------
   202      9840      42897.0      4.4      0.0                  step = int(1 / float(args['sample_residuals']))
   203      9840    6703139.0    681.2      4.7                  pts_in_pattern = getPointsInPatternAsNPArray(collection_key, sensor_key, dataset, step)
   204                                           
   205                                                           # Transform the pts from the pattern's reference frame to the sensor's reference frame -----------------
   206      9840      31083.0      3.2      0.0                  from_frame = sensor['parent']
   207      9840      34939.0      3.6      0.0                  to_frame = dataset['calibration_config']['calibration_pattern']['link']
   208      9840    6638577.0    674.7      4.6                  sensor_to_pattern = opt_utilities.getTransform(from_frame, to_frame, collection['transforms'])
   209      9840      50156.0      5.1      0.0                  pts_in_sensor = np.dot(sensor_to_pattern, pts_in_pattern)
   210                                           
   211                                                           # Project points to the image of the sensor ------------------------------------------------------------
   212                                                           # TODO still not sure whether to use K or P (probably K) and distortion or not
   213      9840      44939.0      4.6      0.0                  w, h = collection['data'][sensor_key]['width'], collection['data'][sensor_key]['height']
   214      9840      91119.0      9.3      0.1                  K = np.ndarray((3, 3), buffer=np.array(sensor['camera_info']['K']), dtype=np.float)
   215                                                           # P = np.ndarray((3, 4), buffer=np.array(sensor['camera_info']['P']), dtype=np.float)
[... cut ...]
   654       246       3881.0     15.8      0.0      pp = pprint.PrettyPrinter(indent=2)
   655       246     643033.0   2614.0      0.5      pp.pprint(report)
   656                                           
   657                                               # print("Normalization took " + tocs())
   658                                               # print(r)
   659                                               # exit(0)
   660       246        740.0      3.0      0.0      return rn  # Return the residuals

miguelriemoliveira, 2020-07-30:
Take a look at line 395 (one of the functions being optimized)

miguelriemoliveira, 2020-07-30:
Here's a video of the calibration of an Agrob dataset with 55 collections. Took 2 or 3 minutes.

https://youtu.be/QNEyUrZm8eg

eupedrosa, 2020-07-30:
The `objective_function` has more commented code that executing code.

miguelriemoliveira, 2020-07-30:
Cleaning on the way, don't worry.

closed 2020-07-31 by miguelriemoliveira