lardemua/atom #221: Generate camera-to-camera error metrics script

aaguiar96, 2020-07-29:
Hi @miguelriemoliveira  and  @eupedrosa 

This also has to done right?
Should I reuse some code? Which script?

miguelriemoliveira, 2020-07-29:
Hi @aaguiar96 ,

I think you can use this:

https://github.com/miguelriemoliveira/OptimizationUtils/blob/master/test/sensor_pose_json_v2/results_visualization.py

and 

https://github.com/miguelriemoliveira/OptimizationUtils/blob/master/test/sensor_pose_json_v2/stereocalib_v2.py

aaguiar96, 2020-08-24:
Hi @miguelriemoliveira, thanks for the tips.

This evening I was implementing the projection. 
I'm getting a result that I do not know if it's right and the calibration is bad, or if the projection has some error.

Here (projections in blue):

[image]
[image]
[image]

I'll try to debug some more. If you can, take a look please!

aaguiar96, 2020-08-24:
Also, I will record the final train and test datasets and try another calibration to see if the error comes from the calibration itself.

aaguiar96, 2020-08-24:
Ok, I think it was error on the calibration.
I calibrated only the two cameras:
[image]

and now I got this:

[image]

[image]

aaguiar96, 2020-08-24:
Also, I recorded a new training dataset.

Nr of collections: 29
Bagfile: calibration_2020-07-01-10-08-10_0.bag
File: [train_dataset.zip](https://github.com/lardemua/atom/files/5119188/train_dataset.zip)

I tested a full calibration with all the collections, and the cameras do not seem to be well calibrated.
Can you try and give me your feedback @miguelriemoliveira?

miguelriemoliveira, 2020-08-24:
Hi @aaguiar96 ,

I will take a look at this tonight and try to help.

aaguiar96, 2020-08-24:
Ok @miguelriemoliveira 

With the last commit, the script is already working and outputting the metrics! Check it out please.
The only thing that is still not supported are the partial detections. This case is a little tricky because we have to associate the corners, and some of them will not have any association. Did you solve this before, or did you only consider full detections?

miguelriemoliveira, 2020-08-25:
Hi @aaguiar96 ,

Sorry, fell asleep last night. Will look into this this afternoon.

For the partial detections your must use match the id field of each point, to make sure you are comparing only the same indices. 
Indices that are only detected in one camera should be discarded as it is not possible to compute the error in that case.

We do that in the objective function but in a somewhat cryptic fashion

https://github.com/lardemua/atom/blob/dca09cbcf2c7d475f9e02a0aa7f83ebaf2f27df1/atom_calibration/src/atom_calibration/calibration/objective_function.py#L109-L120

For the evaluation, since time is not a problem, I would do it more explicitly. For each index id in the src camera, search for the same index id in the target camera, if so compute error.

aaguiar96, 2020-08-25:
Ok thanks @miguelriemoliveira 

I did not know that the indexes correspond, i.e., I thought they could have some random order.
But nice, that way is simple. I'll add that feature and then I think the script is ready. :)

aaguiar96, 2020-08-25:
I @miguelriemoliveira 

Implemented with the last commit. I think we're ready to generate the results.

eupedrosa, 2020-08-25:
Hello @aaguiar96 and @miguelriemoliveira, I guess I am (almost) back from my vacations.

I found a bug here:
https://github.com/lardemua/atom/blob/d7e626a90ccac847d7e18b5a70a4dbcb84176ae5/atom_evaluation/scripts/camera_to_camera_evalutation.py#L254

The axis should be `axis=0`, otherwise the standard deviation is not correctly calculated.

I used your dataset for trainning and my dataset with more than 50 images for testing, here are the results:
```
-------------------------------------------------------------------------------------------------------------
  #           X Error                  Y Error           X Standard Deviation     Y Standard Deviation
-------------------------------------------------------------------------------------------------------------
 54           5.7511                   1.7663                   6.2182                   1.8697
 42           7.0982                   3.2030                   0.9983                   0.4932
 43           6.6598                   2.8408                   0.8080                   0.4986
 49           2.2894                   5.2392                   1.9269                   0.8069
 52           3.6468                   5.2492                   3.7900                   1.7335
 53           5.1667                   1.6159                   5.6693                   1.6909
 24           5.3479                   1.6536                   4.7051                   1.8185
 25           4.7444                   1.5354                   4.6289                   1.7569
 26           8.0171                   1.9010                   4.8605                   1.8729
 27           5.2404                   1.2090                   4.8904                   1.2887
 20           6.5219                   2.2910                   1.0997                   0.8057
 21           6.5238                   2.0392                   1.2697                   0.3466
 48           2.3025                   5.0672                   1.9444                   0.7269
 23           6.3368                   2.0256                   1.1430                   0.6026
 46           3.3004                   4.8201                   1.3524                   0.7226
 47           1.9667                   4.6219                   1.5722                   0.8283
 44           6.6573                   2.2553                   0.9803                   0.6954
 45           7.7538                   2.9348                   0.6626                   1.1151
 28           4.6999                   1.1877                   4.9847                   1.2966
 29           4.3689                   4.0608                   5.1104                   3.1619
 40           7.7306                   1.1311                   2.1663                   0.9575
 41           5.2821                   1.9849                   1.9730                   0.9085
  1           5.1953                   3.6100                   1.4839                   0.7886
  0           5.6302                   3.5849                   1.1780                   0.6402
  3           4.1723                   9.3534                   4.5715                   1.9422
  2           5.0222                   3.7298                   1.3684                   0.7913
  5           3.7866                   2.8271                   4.4987                   1.9566
  4           6.8114                   7.5987                   3.9623                   2.2950
  7           3.3017                   4.9965                   3.8465                   2.2647
  6           3.3832                   3.4959                   3.9409                   2.0986
  9           3.1919                   4.2145                   3.7906                   1.6877
  8           3.2611                   3.6077                   3.7699                   2.0326
 51           3.2890                   5.8059                   3.3395                   1.5103
 39           5.0751                   1.7213                   2.1830                   0.9789
 38           4.2060                   2.9835                   1.9649                   0.9875
 11           6.1454                   2.5530                   6.5961                   2.4942
 10           3.2936                   3.1273                   3.8643                   1.6227
 13           4.1863                   1.7116                   5.2072                   1.7295
 12           4.3962                   1.2819                   5.3176                   1.5886
 15           4.9770                   2.1636                   5.9885                   1.8305
 14           3.9819                   2.1874                   4.8022                   1.9183
 17           5.0505                   2.0958                   6.2320                   1.9358
 16           4.8478                   1.9330                   6.0006                   1.8708
 19           5.7478                   1.6025                   1.2072                   0.7680
 18           6.5262                   2.9165                   1.4179                   0.5472
 31           3.9577                   3.2792                   4.9727                   1.8250
 30           4.1003                   3.1310                   5.1299                   2.0556
 37           5.7667                   2.0569                   1.8045                   0.9479
 36           6.5020                   2.8225                   0.7943                   0.6118
[... cut ...]
 34           6.2808                   1.7304                   0.6919                   0.8107
 33           3.7719                   2.7901                   4.4504                   2.2412
 55           5.5590                   1.8052                   6.1271                   1.8110
 32           3.5284                   2.9110                   4.2852                   1.9570
 50           3.8231                   5.1536                   3.6647                   1.9566
```

Can we say these are good results?

miguelriemoliveira, 2020-08-25:
Hi @aaguiar96 ,

tried it. Really nice. I like the print. Some suggestions:

1. You should add a key to stop showing images and just go to the end and print everything.
2. You could sort the collections before testing, just so you go from 0 to end.

My concern now is the problem you mentioned that we had bad results when calibrating also with the lidar. @eupedrosa , can you test just training with the images like @aaguiar96 did?

miguelriemoliveira, 2020-08-25:
Hi @eupedrosa , nice to have you back, but I can't resist remembering you saying "I will work the entire August " : - ).

aaguiar96, 2020-08-25:
Hi @eupedrosa 

Thanks!

> I found a bug here:

Ok nice. Did you push that fix?

Ok, you used a different train and test dataset and it works, that's nice.
I do not have done that test yet, but I think that the number of collections on the train dataset must be equal or bigger of the number of collections in the test dataset, otherwise the script will break. This is because we use the transformations of the training json while iterating each collection on the test json. Is this a problem?

Also, @eupedrosa your dataset is old so the velodyne xacro is not updated and the velodyne has a rotation. I don't know if this has influence on the evaluation...

miguelriemoliveira, 2020-08-25:
Not that I am complaining ... I also did almost nothing these last few weeks : - )

aaguiar96, 2020-08-25:
> Hi @aaguiar96 ,
> 
> tried it. Really nice. I like the print. Some suggestions:
> 
>     1. You should add a key to stop showing images and just go to the end and print everything.
> 
>     2. You could sort the collections before testing, just so you go from 0 to end.

Thanks @miguelriemoliveira 
I'll add that features.

miguelriemoliveira, 2020-08-25:
Hi @aaguiar96 ,

> I do not have done that test yet, but I think that the number of collections on the train dataset must be equal or bigger of the number of collections in the test dataset, otherwise the script will break

This should not occur ... Can you pinpoint the line that causes this?

aaguiar96, 2020-08-25:
> This should not occur ... Can you pinpoint the line that causes this?

I do not test yet, but the way I implemented I think this will occur.
How did you avoid this @miguelriemoliveira? I go through the collections present in the test dataset to compute the error. For each collection of the test dataset, I extract the calibration result for the same collection in the training dataset.

miguelriemoliveira, 2020-08-25:
Well, I think that all we want from the train dataset (the results.json) is the transformations that were marked to be calibrated.

In the train.json, you have this:

```json
"calibration_config": {
    "anchored_sensor": "right_camera", 
    "bag_file": "$ROS_BAGS/agrob/calibration_2020-07-01-10-08-10_0.bag", 
    "calibration_pattern": {
      "border_size": {
        "x": 0.040000, 
        "y": 0.030000
      }, 
      "dictionary": "DICT_5X5_100", 
      "dimension": {
        "x": 11, 
        "y": 8
      }, 
      "fixed": false, 
      "inner_size": 0.045000, 
      "link": "chessboard_link", 
      "mesh_file": "package://atom_calibration/meshes/charuco_5X5_800x600.dae", 
      "parent_link": "base_link", 
      "pattern_type": "charuco", 
      "size": 0.060000
    }, 
    "description_file": "package://agrob_description/urdf/agrob.urdf.xacro", 
    "max_duration_between_msgs": 1000, 
    "sensors": {
      "left_camera": {
        "child_link": "zed_left_camera_frame", 
        "link": "zed_left_camera_optical_frame", 
        "parent_link": "zed_camera_center", 
        "topic_name": "/zed_nano/zed_node/left/image"
      }, 
      "right_camera": {
        "child_link": "zed_right_camera_frame", 
        "link": "zed_right_camera_optical_frame", 
        "parent_link": "zed_camera_center", 
        "topic_name": "/zed_nano/zed_node/right/image"
      }, 
      "vlp16": {
        "child_link": "vlp16_frame", 
        "link": "velodyne", 
        "parent_link": "tower_link", 
        "topic_name": "/velodyne_points"
      }
    }, 
    "world_link": "base_link"
  }, 
```

which means you can create a list of all transformations being optimized, as we do here:

https://github.com/lardemua/atom/blob/d7e626a90ccac847d7e18b5a70a4dbcb84176ae5/atom_calibration/scripts/calibrate#L291-L298

You will also have to think about the intrinsics, but lets think about that later.

Once you have a list of optimized transforms, you go to your **test dataset** and for all collections, replace only those optimized transformations by the ones that were contained in the **train dataset**. This means you are using the calibrated transformations.

Then you can happily iterate the collections of your **test dataset** and compute the reprojection using homography for each.

Makes sense?

miguelriemoliveira, 2020-08-25:
Just calibrated @eupedrosa 's 55 collection dataset and evaluation gives poor results ...

Calibration seemed to run just fine, reporting camera errors below 1 pixel...

```bash
Optimization finished: `xtol` termination condition is satisfied.

Final errors:
Errors per sensor:
  left_camera 0.749841552146
  right_camera 0.84655484869
  vlp16 0.00686746135618
Sensor left_camera 0.749841552146
Sensor right_camera 0.84655484869
Sensor vlp16 0.00686746135618
Saving the json output file to /home/mike/datasets/agrob/agrob_01_07_2020/atom_calibration.json, please wait, it could take a while ...
Completed.
```

So I guess we have a bug ... but where? In the calibrate or the evaluation?

aaguiar96, 2020-08-25:
> Well, I think that all we want from the train dataset (the results.json) is the transformations that were marked to be calibrated.

Yes makes sense @miguelriemoliveira, but we only have transformations per collection right? Which collection to choose in the training dataset?

aaguiar96, 2020-08-25:
> So I guess we have a bug ... but where? In the calibrate or the evaluation?

Should be on evaluation due to the problem we're discussing or this: 

> Also, @eupedrosa your dataset is old so the velodyne xacro is not updated and the velodyne has a rotation. I don't know if this has influence on the evaluation...

miguelriemoliveira, 2020-08-25:
Hi @aaguiar96,

All transformations marked for optimization should be replaced in all collections of the training dataset. Notice that since the transformations are being optimized, we are sure they are fixed, i.e. constant throughout all collections.

So for example,  if you have optimized a **base_to_camera1** transformation, then you can copy from your training dataset (any collection will be fine, they are all the same, in these cases we use the selected collection key (**) ) ... copy from there to all the collections in the test dataset.

(**) 
https://github.com/lardemua/atom/blob/d7e626a90ccac847d7e18b5a70a4dbcb84176ae5/atom_calibration/scripts/calibrate#L246-L249 )

aaguiar96, 2020-08-25:
> Notice that since the transformations are being optimized, we are sure they are fixed, i.e. constant throughout all collections.

I was missing this. Now I got it. It's an easy fix, I'll commit in few moments. 
Thanks @miguelriemoliveira

miguelriemoliveira, 2020-08-25:
Yes, on evaluation. This is good news : - )

Also, another thing: I calibrated with only cameras (again good final errors reported) and in the end I get the same large errors. This is good.

Are we using intrinsics and distortion ? Not sure ... I will call you .

aaguiar96, 2020-08-25:
> Are we using intrinsics and distortion ? Not sure ... I will call you .

Yes, we are.

eupedrosa, 2020-08-25:
For @miguelriemoliveira 
> Hi @eupedrosa , nice to have you back, but I can't resist remembering you saying "I will work the entire August " : - ).

I have no recollection of that.... 

> So I guess we have a bug ... but where? In the calibrate or the evaluation?

Maybe not. Remeber, we are projecting from one camera to the other.
The errors adds all the way and we have several sources of error:
- corner detection in the source
- SolverPnP that is influenced by the corner detection
- transformation chain
- cornder detection in the target 

Another error that is not accounted is the image synchronization. The only way to eliminate this source of error would be to have some sort of support for the chessboard. @aaguiar96 needs more hours of gym to reduce its jitter. OR, improve the data collector to improve the synchronization. A 300 ms betwwen two images with a moving target can easily create a problem.

For @aaguiar96 
> Ok nice. Did you push that fix?

No, I did not. Do you whant me to do it?

> I do not have done that test yet, but I think that the number of collections on the train dataset must be equal or bigger of the number of collections in the test dataset, otherwise the script will [...]

I do not know. It work without problems with the datasets that I used.

> Also, @eupedrosa your dataset is old so the velodyne xacro is not updated and the velodyne has a rotation. I don't know if this has influence on the evaluation..

It may not influence.

aaguiar96, 2020-08-25:
> No, I did not. Do you whant me to do it?

No, I'll do that.

> I do not know. It work without problems with the datasets that I used.

I'm fixing this.

> Another error that is not accounted is the image synchronization. The only way to eliminate this source of error would be to have some sort of support for the chessboard. @aaguiar96 needs more hours [...]

I agree with this @eupedrosa. I think I could try gym, but it will delay the paper... :-(  So, if we are sure that this is having impact, maybe trying to have a support? A chair should work right?

miguelriemoliveira, 2020-08-25:
I also agree a static pattern will improve things ... but I think this is not the fundamental problem here. 

Look, during calibration we get sub pixel errors, if we run evaluation with that same train and test dataset then we must see the same magnitude of errors (if not the same). Do you agree?

aaguiar96, 2020-08-25:
> Well, I think that all we want from the train dataset (the results.json) is the transformations that were marked to be calibrated.

Fixed with last commit.

aaguiar96, 2020-08-25:
> Look, during calibration we get sub pixel errors, if we run evaluation with that same train and test dataset then we must see the same magnitude of errors (if not the same). Do you agree?

Yes, I think so. Did you test that @miguelriemoliveira ?

eupedrosa, 2020-08-25:
> Look, during calibration we get sub pixel errors, if we run evaluation with that same train and test dataset then we must see the same magnitude of errors (if not the same). Do you agree?

Yes, I agree, that is why I always had reservations about the Homography.
Can I "my" method to the evaluation script?

aaguiar96, 2020-08-25:
>  Can I add "my" method to the evaluation script?

Sure @eupedrosa.

miguelriemoliveira, 2020-08-25:
Sure. Do it in a new script please.

eupedrosa, 2020-08-25:
Before pushing any code. 
Here is some results with "my" approach:
- train dataset provided by @aaguiar96 
- teste dataset provided by me
```-------------------------------------------------------------------------------------------------------------
  #           X Error                  Y Error           X Standard Deviation     Y Standard Deviation
-------------------------------------------------------------------------------------------------------------
 54           0.2617                   0.3420                   0.3159                   0.3251
 42           0.4199                   0.5529                   0.5163                   0.4030
 43           0.2295                   0.2858                   0.2119                   0.3585
 49           0.1471                   0.3551                   0.1727                   0.2304
 52           0.2320                   0.1961                   0.2479                   0.2236
 53           0.2392                   0.2872                   0.3072                   0.3261
 24           4.8181                   0.7806                   0.8365                   0.6522
 25           0.4280                   0.5191                   0.3555                   0.5501
 26           0.3764                   1.1898                   0.4407                   0.4284
 27           0.7264                   0.4445                   0.3882                   0.4985
 20           0.2631                   0.3558                   0.2635                   0.4528
 21           0.1725                   0.2994                   0.1908                   0.1731
 48           0.5385                   0.2870                   0.1848                   0.1879
 23           0.2572                   0.3194                   0.3036                   0.2141
 46           0.7165                   0.2685                   0.3968                   0.2877
 47           0.8130                   0.3093                   0.3387                   0.3461
 44           0.3877                   0.5995                   0.3574                   0.5891
 45           0.8460                   0.7635                   0.6977                   0.9080
 28           0.6657                   0.3443                   0.3414                   0.3163
 29           2.0350                   2.0034                   2.4142                   2.2456
 40           1.9443                   0.6447                   0.2524                   0.2586
 41           0.6505                   0.3320                   0.5002                   0.2347
  1           0.1526                   0.2695                   0.1825                   0.2419
  0           0.1502                   0.2689                   0.1840                   0.2277
  3           3.7560                   3.7877                   0.3131                   0.6546
  2           0.1851                   0.2518                   0.2123                   0.2118
  5           0.3539                   0.6694                   0.4437                   0.3976
  4           4.6895                   2.5686                   1.5311                   0.9191
  7           0.5277                   1.4186                   0.6402                   0.9399
  6           0.2207                   0.2161                   0.2700                   0.2567
  9           0.4867                   1.0028                   0.2116                   0.3504
  8           0.3006                   0.2753                   0.3526                   0.3258
 51           0.2380                   0.3795                   0.2759                   0.2480
 39           0.6307                   0.2295                   0.3997                   0.2688
 38           1.9998                   1.3551                   1.5322                   1.2012
 11           2.2710                   3.2346                   0.5918                   0.4389
 10           0.3041                   0.1883                   0.2079                   0.2065
 13           0.3500                   0.4168                   0.3918                   0.4542
 12           1.2913                   1.2668                   0.9304                   0.8930
 15           0.2378                   0.4562                   0.2951                   0.2682
 14           0.4912                   0.2743                   0.3563                   0.2656
 17           0.4445                   0.3189                   0.3543                   0.3118
 16           0.3584                   0.4867                   0.4066                   0.3120
 19           0.6877                   0.9570                   0.3093                   0.2270
 18           0.5625                   0.5844                   0.6628                   0.6412
 31           0.4144                   0.4974                   0.2711                   0.2369
 30           0.3285                   0.3760                   0.1791                   0.4000
 37           0.3944                   0.2070                   0.3965                   0.2515
 36           0.1663                   0.2670                   0.1930                   0.2602
 35           0.1499                   0.2488                   0.1850                   0.2017
 34           0.3047                   1.1713                   0.3453                   0.3325
 33           0.2124                   0.1801                   0.2454                   0.2110
[... cut ...]
 19           0.3619                   0.4317                   0.1745                   0.2757
 18           2.7886                   0.3534                   0.2713                   0.4104
-------------------------------------------------------------------------------------------------------------
 All          0.5263                   0.4477                   0.8434                   0.6175
-------------------------------------------------------------------------------------------------------------
```

miguelriemoliveira, 2020-08-25:
Hi @eupedrosa ,

looks good. I would suggest push in a new script. Two questions:

1. We talked about this before and you promised you would write some latex similar to the RAS paper section 4.1 talking about your metric, which I could never understand entirely.
2. Do you consider the undistortion?

eupedrosa, 2020-08-26:
>  I would suggest push in a new script.

Why?  The changes are small. I added the "--po, --pattern_object" option to tell the script to use "my" method. Is this ok with you?

> We talked about this before and you promised you would write some latex similar to the RAS paper section 4.1 talking about your metric, which I could never understand entirely. Are you going to?

I do not promised this kind o things.. But yes, I said I would do it. However, I think if I push the code and you analyse the source code you will understand immediately

> Do you consider the undistortion?

Yes. But the correct question would be "Do you consider the distortion?"

aaguiar96, 2020-08-26:
Hi @eupedrosa 

Great news, thanks! Did you detect any bug on my code? I will look at it again this afternoon.

eupedrosa, 2020-08-26:
Yes @aaguiar96..

https://github.com/lardemua/atom/blob/596a71ca73d75805456cec13b2cd15bf7674bc24/atom_evaluation/scripts/camera_to_camera_evalutation.py#L87-L91

This actually does `distortion` and not `undistortion`

aaguiar96, 2020-08-26:
> This actually does `distortion` and not `undistortion`

Ok, I did not validate that since it seems to work in the deprecated camera-to-camera evaluation script, that actually generated the RAS results. So, are you sure?...
But, in any case, we can substitute that function with [this one](https://docs.opencv.org/2.4/modules/imgproc/doc/geometric_transformations.html#undistortpoints) right?

eupedrosa, 2020-08-26:
> Ok, I did not validate that since it seems to work in the deprecated camera-to-camera evaluation script, that actually generated the RAS results. So, are you sure?...

The lines 89, 90 and 91 are the same piece of code that you can find in the `projectToCamera` function. @miguelriemoliveira any comment about this?
If this is true, then RAS results may be slighly incorrect.

> But, in any case, we can substitute that function with this one right?

Yes, do you wanna try it?

aaguiar96, 2020-08-26:
> Yes, do you wanna try it?

Yes, I'll try!

eupedrosa, 2020-08-26:
Sorry @aaguiar96, I already try it...

aaguiar96, 2020-08-26:
> Sorry @aaguiar96, I already try it... You took to long to answer :p

Eheh :-) And did it solve the issue?

eupedrosa, 2020-08-26:
I guess so, take a look at the results
```
-------------------------------------------------------------------------------------------------------------
  #           X Error                  Y Error           X Standard Deviation     Y Standard Deviation
-------------------------------------------------------------------------------------------------------------
 54           0.3273                   0.3935                   0.4140                   0.3952
 42           0.4516                   0.5508                   0.5534                   0.3792
 43           0.2612                   0.3228                   0.2610                   0.3892
 49           0.2391                   0.4100                   0.2969                   0.3149
 52           0.3434                   0.2558                   0.3599                   0.3161
 53           0.2954                   0.3455                   0.3861                   0.3914
 24           6.2378                   0.9279                   1.5434                   1.0152
 25           0.5934                   0.6921                   0.4840                   0.7782
 26           0.6475                   1.3307                   0.7598                   0.5316
 27           0.9382                   0.5544                   0.5969                   0.6393
 20           0.3040                   0.4704                   0.3245                   0.6347
 21           0.2004                   0.3152                   0.2283                   0.2108
 48           0.6152                   0.3713                   0.2366                   0.2616
 23           0.2720                   0.3567                   0.3228                   0.2545
 46           0.8017                   0.2958                   0.5807                   0.3701
 47           0.9005                   0.3256                   0.4404                   0.3936
 44           0.4173                   0.6182                   0.4422                   0.5706
 45           0.9265                   0.9456                   0.7686                   1.0560
 28           0.8272                   0.4068                   0.5306                   0.4269
 29           2.3853                   2.0643                   2.8052                   2.2701
 40           1.9813                   0.6554                   0.2480                   0.3013
 41           0.6007                   0.4186                   0.3908                   0.4162
  1           0.2085                   0.3346                   0.2692                   0.3446
  0           0.2288                   0.3434                   0.2661                   0.3355
  3           4.0545                   3.9058                   0.3912                   0.7449
  2           0.1994                   0.2828                   0.2636                   0.2544
  5           0.4639                   0.6997                   0.5599                   0.4159
  4           5.2704                   2.8592                   1.4099                   1.1019
  7           0.6352                   1.4969                   0.7708                   0.9549
  6           0.2773                   0.2526                   0.3468                   0.3156
  9           0.5832                   1.0545                   0.3731                   0.4084
  8           0.3830                   0.2705                   0.4598                   0.3158
 51           0.2836                   0.4368                   0.3274                   0.3122
 39           0.6157                   0.2963                   0.3850                   0.3546
 38           1.9814                   1.4130                   1.5185                   1.3130
 11           2.5919                   3.4461                   0.8551                   0.3681
 10           0.3646                   0.2688                   0.3185                   0.3086
 13           0.3730                   0.4562                   0.4274                   0.5088
 12           1.4290                   1.3410                   1.0756                   0.8816
 15           0.3062                   0.4961                   0.3870                   0.3694
 14           0.5344                   0.2944                   0.4073                   0.3091
 17           0.4672                   0.3799                   0.3700                   0.4031
 16           0.3702                   0.5269                   0.4294                   0.3804
 19           0.7087                   0.9958                   0.3634                   0.3297
 18           0.5761                   0.6418                   0.6789                   0.7323
 31           0.4737                   0.5147                   0.3344                   0.2883
 30           0.3978                   0.3530                   0.3267                   0.3537
 37           0.3008                   0.2611                   0.2788                   0.3150
 36           0.2303                   0.2882                   0.2847                   0.3006
 35           0.2145                   0.3130                   0.2534                   0.2959
 34           0.3321                   1.2157                   0.3697                   0.3709
 33           0.2821                   0.2092                   0.3447                   0.2579
 55           0.3009                   0.3447                   0.3816                   0.4045
[... cut ...]
 50           1.4605                   1.1214                   1.6068                   1.3309
-------------------------------------------------------------------------------------------------------------
 All          0.8339                   0.7139                   1.5016                   1.1131
-------------------------------------------------------------------------------------------------------------
```

The results are very similar to "my" approach. The error is slighly higher, but that is expected.

aaguiar96, 2020-08-26:
Ok, great!

> The results are very similar to "my" approach. The error is slighly higher, but that is expected.

Why? Because of the homography?

Can you push the code, please?

eupedrosa, 2020-08-26:
> Why? Because of the homography?

The homography uses the detected corners from both images while "my" approach uses the points from the pattern object that have no detection error because they are known.

> Can you push the code, please?

Yes, I am gonna push the code. I also corrected the drawing of the projected corners. Using Homography, these corners do not have distortion, but the image has.

miguelriemoliveira, 2020-08-26:
Hi,

A lot of work you guys are doing ... great!

we use distortion, yes.
The idea is to undistort, then project, the distort for the other camera.

The writting of a latex is important because we need to explain how we
compute the results. And, in my opinion, the current homography looks quite
nice on a paper .

You could write it in Andre's overleaf ... then we can compare ...

aaguiar96, 2020-08-26:
So, I tested a camera-camera calibration with @eupedrosa's dataset with only 10 collections (I cannot do it with all of them, the calibration takes too long, do you know why @miguelriemoliveira?). These are the results:
```
Final errors:
Sensor left_camera 2.2872869808447853
Sensor right_camera 1.9953841216533612
Saving the json output file to /home/andre-criis/Documents/saved_datasets/eurico_dataset/atom_calibration.json, please wait, it could take a while ...
Completed.
[INFO] [1598450615.322321]: Reading description file /home/andre-criis/Source/catkin_ws/src/agrob/agrob_description/urdf/agrob.urdf.xacro...
Parsing description file /home/andre-criis/Source/catkin_ws/src/agrob/agrob_description/urdf/agrob.urdf.xacro
```

Then, I run the evaluation with the same train and test json (the one obtained) and I got:

With homography:
```
-------------------------------------------------------------------------------------------------------------
  #           X Error                  Y Error           X Standard Deviation     Y Standard Deviation
-------------------------------------------------------------------------------------------------------------
  0           0.7664                   0.5475                   0.7264                   0.6303
  1           0.7440                   0.7578                   0.7827                   0.7181
  2           0.5640                   0.6720                   0.6264                   0.6594
  3           2.7848                   2.4121                   1.9428                   0.9293
  4           5.4983                   1.7507                   0.8045                   1.9261
  5           1.6737                   1.8730                   1.7484                   1.3980
  6           1.3969                   1.0985                   1.5537                   1.0886
  7           0.7678                   1.1630                   0.5893                   1.0885
  8           1.8116                   0.8544                   1.9607                   0.8361
  9           1.3227                   0.5923                   1.4459                   0.7257
-------------------------------------------------------------------------------------------------------------
 All          1.5903                   1.1299                   2.1994                   1.4707
-------------------------------------------------------------------------------------------------------------
```

@eupedrosa's approach:
```
-------------------------------------------------------------------------------------------------------------
  #           X Error                  Y Error           X Standard Deviation     Y Standard Deviation
-------------------------------------------------------------------------------------------------------------
  0           1.1613                   1.1485                   1.2275                   1.3396
  1           1.3264                   1.2749                   1.4798                   1.3745
  2           1.0705                   1.3813                   1.2472                   1.4753
  3           2.9823                   2.4401                   2.3504                   1.3104
  4           5.3215                   1.7866                   1.8328                   2.0481
  5           2.2011                   1.7217                   2.4403                   1.5070
  6           1.9319                   1.1132                   2.1678                   1.2601
  7           1.9462                   2.1626                   2.2917                   2.6899
  8           3.2532                   2.0928                   3.7635                   2.6447
  9           2.7888                   1.4836                   3.4372                   1.8130
-------------------------------------------------------------------------------------------------------------
 All          2.2960                   1.6425                   2.9434                   2.1053
-------------------------------------------------------------------------------------------------------------
```

Here, homography have lower error... 
Also, the errors printed by the calibration are `sqrt(x ** 2 + y ** 2)` to all collections? Just to be able to compare what's the approach that is more similar to the calibration output.

eupedrosa, 2020-08-26:
>  I cannot do it with all of them, the calibration takes too long, do you know why @miguelriemoliveira?

* Make sure you have the latest version of [OptimizationUtils](https://github.com/miguelriemoliveira/OptimizationUtils#installation)
* Use Ubuntu package for numpy `sudo apt install python-numpy`
* Make sure you remove pip's numpy: `pip uninstall numpy` or `sudo -H pip uninstall numpy` (You may have to execute this more that once)

> Also, the errors printed by the calibration are sqrt(x ** 2 + y ** 2) to all collections?

It is the error __mean__  for all collections per sensor, i.e. `mean(sqrt(x**2 + y**2))`. I think we could also use the Root Mean Squared Error, which is `sqrt(mean(x**2 + y**2))`

aaguiar96, 2020-08-26:
Ok thanks @eupedrosa 

Anyway, the evaluation seems to be working. Your version seems closer to the calibration result than homography.

So, 
* Which cam-2-cam evaluation method to use?
* Are we able to start generating results for the paper?

eupedrosa, 2020-08-26:
Both methods are working, but the homography has a nice math behind it.
We could present both, but they are a little redundant.

But more important than this is which metrics to use:
1. The mean of pixel difference, per component
2. The mean of the pixel error, i.e. euclidean distance.
3. The RMSE
4. The translation and rotation error - The calculated pattern pose in `source_sensor`  should be equal to the calculated pattern pose in `target_sensor`, in the base frame.

@miguelriemoliveira, do you have more ideias, more suggestions?

miguelriemoliveira, 2020-08-26:
Hi guys,

you make me feel bad for being on my last vacation days .. thanks a lot.

I did not get what was going wrong ... I guess now both are giving similar results so that's very good.

About going slowly, in my machine it is quite fast. Calibrated the 55 collections in 3/4 minutes. Try @eupedrosa 's suggestions. Another suggestion is to run without visualization. Take the -v -rv -si -vo flags away.

I agree with @eupedrosa  the homography is nice to show off in a paper (would have to be updated with the undistort stuff). But we can go for the other if we have a similar mathematical formulation ...

About metrics, I agree with all and don't have any other ideas for now.

I should work a bit on this tonight ... what should I work on?

BTW, did you see the updates I did in the README? We have a new white background image like @eupedrosa wanted.

aaguiar96, 2020-08-26:
> I should work a bit on this tonight ... what should I work on?

Hi @miguelriemoliveira 

If you want, you can take a look at [this](https://github.com/lardemua/atom/issues/207#issuecomment-680932308).
If you think it's working, maybe tomorrow I can start generating the results for the paper.

> BTW, did you see the updates I did in the README? We have a new white background image like @eupedrosa wanted.

I saw it. Really nice!

aaguiar96, 2020-09-01:
Hi @miguelriemoliveira  and @eupedrosa 

I finished all the paper sections except the abstract, results, and conclusions.
Soon, I'll start to work on the results. 
Here (camera to camera), what should I used? Homography, @eupedrosa's approach, or both?

miguelriemoliveira, 2020-09-01:
Hi @aaguiar96 ,

great. @eupedrosa , you want to split the sections for review of the paper? Or should we revise first one and then the other?

About the results: For sure we should use just one, since they are very similar. We can use Eurico's approach, no problem by me, but for that he must first write down the methodology. The homography approach needs only be updated with the undistort/distort part. For me, whatever @eupedrosa says is ok.

We can talk this afternoon if you want to ... I have a meeting at 14h30 which should end around 15h15 ...

aaguiar96, 2020-09-01:
> We can talk this afternoon if you want to ... I have a meeting at 14h30 which should end around 15h15 ...

Fine by me, 15h30 then?

miguelriemoliveira, 2020-09-01:
OK.

*  Miguel Riem de Oliveira                                             *
email:    mriem@ua.pt    |    m.riem.oliveira@gmail.com
office: 22.2.18                 |    ext: 23887
Dep. de Engenharia Mecânica, Universidade de Aveiro
*Campus Universitário de Santiago, 3810-193  Portugal*

eupedrosa, 2020-09-01:
> @eupedrosa , you want to split the sections for review of the paper? Or should we revise first one and then the other?

I agree that we should meet to talk about this. 15h30 then?

aaguiar96, 2020-09-03:
Hi @eupedrosa 

Will you had the e_R and e_t metrics to the script or do you want me to?
I think it is the only thing that is missing.

[50 more replies]

closed 2020-10-05 by aaguiar96