lardemua/atom #222: Reorganization of ATOM

miguelriemoliveira, 2020-08-02:
Hi @eupedrosa and @aaguiar96 l,

I am thinking of improving the structure of atom. The utilities whic are in atom core are now huge. 

My proposal is to create an atom_lib package which will contain the utilities (eventually divided into different modules, for better organization).

In the meantime I will also pickup functions from the optimization_utils utilities module which should be in atom.

I  should work on this sometime today or tomorrow, if you have any suggestions please let me know.

eupedrosa, 2020-08-02:
> My proposal is to create an atom_lib package which will contain the utilities (eventually divided into different modules, for better organization).

I agree, except with the `atom_lib` package. I think that `atom_core` is the right place. `lib` and `core` can mean the same thing in this case.  We can further discuss this if you like.

> I should work on this sometime today or tomorrow, if you have any suggestions please let me know.

May we should talk first before diving into this...

miguelriemoliveira, 2020-08-02:
Hi Eurico,

OK, I wait for our talk. Are you available today?

Actually, I was thinking that the names we have don't make a lot of sense
and perhaps we could change more stuff.
My comprehensive solution would be:

atom_core
    just the utilities, ios, cache, etc,

atom_framework or atom_accessories (old atom_calibration)
     scripts for set_initial_estimate, configure calibration package, etc

atom_calibration or atom_optimization
    scripts for running the actual calibration or optimization, e.g
calibrate, getters_and_setters, objective_function, etc.

What do you think?

*  Miguel Riem de Oliveira                                             *
email:    mriem@ua.pt    |    m.riem.oliveira@gmail.com
office: 22.2.18                 |    ext: 23887
Dep. de Engenharia Mecânica, Universidade de Aveiro
*Campus Universitário de Santiago, 3810-193  Portugal*

miguelriemoliveira, 2020-08-03:
@eupedrosa 's suggestion makes more sense.

atom_core keeps the utilities
atom_calibration takes also the calibrate and related functions.

I am on it.

miguelriemoliveira, 2020-08-04:
@eupedrosa don't forget to pull before doing anything else. This was a huge commit.

miguelriemoliveira, 2020-08-04:
Hi @eupedrosa ,

as you can see the reorganization is in full steam. One proposal: How about creating a new ros package called atom_calibration_patterns which would contain the code for creating and detecting the patterns, as well as the meshes for the patterns. 

Right now these meshes are in atom_calibration and it is not very pretty. Any other ideas?

Another point: atom_core now does not have scripts, nor should it (like a true library), right?

eupedrosa, 2020-08-04:
> as you can see the reorganization is in full steam. One proposal: How about creating a new ros package called atom_calibration_patterns which would contain the code for creating and detecting the patterns, as well as the meshes for the patterns.

I think it is not necessary to create a new ros package. Why should the patterns be special?

> Right now these meshes are in atom_calibration and it is not very pretty. Any other ideas?

I'm more concerned with the unecessary meshes that we have in the package. And why it is not pretty?

> Another point: atom_core now does not have scripts, nor should it (like a true library), right?

Right!

miguelriemoliveira, 2020-08-04:
Ok. I will remove unnecessary meshes and keep it in atom calibration

miguelriemoliveira, 2020-08-05:
I have done all I could remember. Will close. If something else comes up I will reopen.

closed 2020-08-05 by miguelriemoliveira