lardemua/atom #225: Unknown visual in visualization.

eupedrosa, 2020-08-07:
This happens when I launch the visualization for agrob:

```python
Traceback (most recent call last):
  File "/home/melodic-dev/src/atom/atom/atom_calibration/scripts/calibrate", line 609, in <module>
    main()
  File "/home/melodic-dev/src/atom/atom/atom_calibration/scripts/calibrate", line 268, in main
    graphics = visualization.setupVisualization(dataset, args, selected_collection_key)
  File "/home/melodic-dev/src/atom/atom/atom_calibration/src/atom_calibration/calibration/visualization.py", line 313, in setupVisualization
    rgba=rgba)
  File "/home/melodic-dev/src/atom/atom/atom_core/src/atom_core/rospy_urdf_to_rviz_converter.py", line 127, in urdfToMarkerArray
    link.name)
ValueError: Unknown visual.geom type<class 'urdf_parser_py.urdf.Cylinder'> for link vlp16_frame
```
Can you fix this @miguelriemoliveira ?

miguelriemoliveira, 2020-08-07:
Hum, strange. It works fine for me. Are you sure you are using the correct rospy_urdf_to_rviz_converter.py?

From the print it seems the correct one, since its inside the atom_core.

How about the urdf_parser_py, do you have it installed?

miguelriemoliveira, 2021-03-07:
No news from @eupedrosa ... closing.

closed 2021-03-07 by miguelriemoliveira