lardemua/atom #227: Missing wheels on agrob description visualization

aaguiar96, 2020-08-26:
Hi @miguelriemoliveira  and @eupedrosa 

Did you noticed that when running the calibration we get these warnings:
```
Parsing description file /home/andre-criis/Source/catkin_ws/src/agrob/agrob_description/urdf/agrob.urdf.xacro

Unknown attribute "type" in /robot[@name='agrob']/transmission[@name='front_left_wheel_trans']
Unknown attribute "type" in /robot[@name='agrob']/transmission[@name='front_right_wheel_trans']
Unknown attribute "type" in /robot[@name='agrob']/transmission[@name='rear_left_wheel_trans']
Unknown attribute "type" in /robot[@name='agrob']/transmission[@name='rear_right_wheel_trans']
Robot has at least joint front_left_wheel non fixed. Will render all collections
```
and so, agrob wheels are not displayed?

It is strange, because if you run agrob bringup package
```
roslaunch agrob_bringup bringup.launch
```
everything works and the wheels are shown.

Do you know how to fix this? It's somehow important because I want to generate figures for the paper, but with this I can't.

miguelriemoliveira, 2020-08-27:
Hi @aaguiar96 ,

I will try to look at this now. Will give you some feedback in a moment.

miguelriemoliveira, 2020-08-27:
Hi @aaguiar96 ,

I was searching around and I think the good news is that our code is working ok. 

I think the reason why you do not have the wheels is because you have used a bag file which did not contain the joint_state messages. You remember I told you we needed these also (https://github.com/lardemua/atom/issues/157)?

Because there are no joint state messages, no transformations exist for the wheels when we playback the bagfile, then these do not appear in the list of transforms of our dataset collections, and that's why the wheels are not shown.

Can you confirm the joint_state messages are missing from the bagfile you used to record the dataset?

aaguiar96, 2020-08-27:
I @miguelriemoliveira

> I think the reason why you do not have the wheels is because you have used a bag file which did not contain the joint_state messages. You remember I told you we needed these also (#157)?

I do remember. I never confirmed if agrob contains anything publishing joint_state...
It's strange because it worked in previous bagfiles, and in this one, it doesn't.

> Can you confirm the joint_state messages are missing from the bagfile you used to record the dataset?

How do I do that? Aren't the joint_state messages generated by the robot description? That does not come on the bagfile right?

miguelriemoliveira, 2020-08-27:
Hi aaguiar96,

you can use the command 

> rosbag info <bag file>

I think agob has joint states, take a look at #128

Why it was working before I am not really sure, but I think at some time I abolished the usage of robot_state_publisher in the calibrate, perhaps it was before this.

aaguiar96, 2020-08-27:
Hm... 

From [here](https://github.com/lardemua/atom/issues/128#issuecomment-595693527) I can see that agrob has a topic publishing the `joint_states` but I never recorded it in any rosbag.

miguelriemoliveira, 2020-08-27:
yes, that's the problem. If you don't record this then you cannot see the
wheels.

*  Miguel Riem de Oliveira                                             *
email:    mriem@ua.pt    |    m.riem.oliveira@gmail.com
office: 22.2.18                 |    ext: 23887
Dep. de Engenharia Mecânica, Universidade de Aveiro
*Campus Universitário de Santiago, 3810-193  Portugal*

aaguiar96, 2020-08-27:
> yes, that's the problem. If you don't record this then you cannot see the wheels.

Oh, ok... So, do you suggest that I record a new rosbag or there is any workaround to draw the wheel with the ones we have?

miguelriemoliveira, 2020-08-27:
Hi André,

I can do a hack ... kust for taking nice images. Give me 1 hour

aaguiar96, 2020-08-27:
> I can do a hack ... kust for taking nice images. Give me 1 hour

Ok thanks
Meanwhile, while I'm waiting for yours and eurico's validation of the LiDAR-to-camera evaluation script, I'm writing the paper.
I am using RAS and hand-eye papers as a baseline. I think it's getting nice!

miguelriemoliveira, 2020-08-27:
Hi @aaguiar96 ,

I did the hack. Its here 

https://github.com/lardemua/atom/blob/34f2f38593849603c1287f8c23c38d4b99a515c9/atom_calibration/src/atom_calibration/calibration/visualization.py#L420-L433

You can then comment / delete the lines once you have the images.

I also added a new flag --all_joints_fixed  or -ajf which you should use.

When doing calibration the visualization does not really know whether or not to draw a robot mesh for all collections. In principle this should be the way (for example in the hand eye the robot actually moves). In the case of the atlascar the robot is static, meaning the joints are all fixed, and in these cases it does not make sense to redraw the robot in the same exact position.

To tackle this I did an automatic detection mechanism which asserts if the robot has all joints fixed or not. The problem is that the agrob  has the wheel joints non fixed, and because of that the mesh is being drawn for all collections. 

Use the -ajf flag to override. You will notice rviz will show the model without transparency when you use the flag.

Please close it it is working.

aaguiar96, 2020-08-27:
Ok, thanks @miguelriemoliveira 

It's working. As soon as I end up writing the paper, I'll delete this code.

Closing the issue.

closed 2020-08-27 by aaguiar96