lardemua/atom #228: Error visualizing LaserBeams

aaguiar96, 2020-08-28:
Hi @miguelriemoliveira  and @eupedrosa 

Can you visualize the laser beams on rviz?
I'm getting this error:
```
Points should not be empty for specified marker type.
```

miguelriemoliveira, 2020-08-28:
Hi @aaguiar96 ,

I think we removed the calculation of the laser beams and replaced it with the orthogonal distance to speed up the objective function. 

Now that we have a much faster objective function we could consider retrieving the beam distance ...

miguelriemoliveira, 2020-08-28:
Do you want me to work on this?

aaguiar96, 2020-08-28:
Not exacly what I wanted. 
I just wanted to visualize something like this:
[image]

to show the impact of orthogonal distance before and after calibration on the paper.
But, if it implies too much work, it's not important, I can use other images!

aaguiar96, 2020-08-28:
By "like this" I only meant adding the blue laser beams to our visualization system.

miguelriemoliveira, 2020-08-28:
Hi @aaguiar96 ,

To be plotted we need to compute the blue lines first, and these were being computed in the beam distance code.

If your idea is to use the orthogonal distance, then I would use a different plot uniting each lidar data point to the closest point on the plane, i.e. the brown line in this image 

https://mathinsight.org/distance_point_plane

Makes sense?

aaguiar96, 2020-08-28:
Ok, makes sense.
I'll try to make that code to work on rviz. 

Thank's @miguelriemoliveira, I think we can close the issue right?

miguelriemoliveira, 2020-08-28:
Yes.

closed 2020-08-28 by miguelriemoliveira