lardemua/atom #229: When to go to Ubuntu 20.04 and ROS Noetic?

miguelriemoliveira, 2020-09-06:
Hi @eupedrosa and @aaguiar96 ,

I am preparing classes in which I will be teaching linux / ROS. I was wondering when we should jump to Ubntu 20.04 and the new ROS. Any opinions on this one?

aaguiar96, 2020-09-06:
Hi @miguelriemoliveira 

For an instant, I thought you will say ROS2 :-)

I think If we do so, we should create a new branch "noetic" where we update the code for this version.
The official code (master) should be the one we have now that runs on melodic since it is the version that most people use by now right?

eupedrosa, 2020-09-06:
In terms of development is like @aaguiar96 says. Create a "noetic-devel" branch and work on it.

But out focus, for now, should be melodic. Mainly because noetic requires a rewrite to python3.

@miguelriemoliveira, if you are teaching the basics os ROS you should go with noetic and ubuntu 20.04.

miguelriemoliveira, 2020-09-06:
Thanks for the inputs. I now I should go into 20.04, but I don't want to be using both 18.04 for research and 20.04 for teaching. I think I will just use 18.04. Next year, when we already migrated, I will use 20.04.

closed 2020-09-06 by miguelriemoliveira