lardemua/atom #231: Error while calibrating without partial detections

aaguiar96, 2020-09-09:
Hi @miguelriemoliveira 

Can you solve this one?

```
Removing collection 8 -> pattern was not found in sensor left_camera (must be found in all sensors).
Partial detection removed: label from collection 11, sensor left_camera
Partial detection removed: label from collection 11, sensor right_camera
Partial detection removed: label from collection 10, sensor left_camera
Partial detection removed: label from collection 10, sensor right_camera
Partial detection removed: label from collection 13, sensor right_camera
Partial detection removed: label from collection 1, sensor left_camera
Partial detection removed: label from collection 0, sensor left_camera
Partial detection removed: label from collection 3, sensor left_camera
Partial detection removed: label from collection 5, sensor left_camera
Partial detection removed: label from collection 5, sensor right_camera
Partial detection removed: label from collection 4, sensor left_camera
Partial detection removed: label from collection 7, sensor left_camera
Partial detection removed: label from collection 7, sensor right_camera
Partial detection removed: label from collection 6, sensor left_camera
Partial detection removed: label from collection 6, sensor right_camera
Partial detection removed: label from collection 9, sensor left_camera
Partial detection removed: label from collection 9, sensor right_camera
Collection 11: Chessboard not detected by sensor left_camera
Collection 11: Chessboard not detected by sensor right_camera
Traceback (most recent call last):
  File "/home/andreaguiar/Source/catkin_ws/src/atom/atom_calibration/scripts/calibrate", line 606, in <module>
    main()
  File "/home/andreaguiar/Source/catkin_ws/src/atom/atom_calibration/scripts/calibrate", line 137, in main
    dataset['patterns'] = patterns.createPatternLabels(args, dataset)  # TODO: Solve this strange dependency.
  File "/home/andreaguiar/Source/catkin_ws/src/atom/atom_calibration/src/atom_calibration/calibration/patterns_config.py", line 223, in createPatternLabels
    raise ValueError('Collection ' + collection_key + ' could not find pattern.')
ValueError: Collection 11 could not find pattern.
```

miguelriemoliveira, 2020-09-09:
On me for tonight ...

aaguiar96, 2020-09-09:
Thanks @miguelriemoliveira 

It is @eupedrosa's fault...

miguelriemoliveira, 2020-09-10:
Hi @aaguiar96 ,

I think it is solved. The problem was this: 

```
 # It may occur that some collections do not have any detection in a camera sensor (because all detections were
 # partial and have been removed, or just because no detection existed). Since we need at lease one camera sensor
 # detection of the pattern in a collection in order initialize the parameters (check calibrate line 133),
 # we will remove collections which do not have at least one detection by a camera.
```

Please test and close if appropriate.

aaguiar96, 2020-09-10:
I @miguelriemoliveira 

It works. Thanks!

closed 2020-09-10 by aaguiar96

miguelriemoliveira, 2020-09-10:
Great!

commit 173a3c66 (2020-09-10): fix #231