lardemua/atom #232: Fix bag file with empty stamp in camera_info

miguelriemoliveira, 2020-09-15:
@aaguiar96 , I could not find your issue so I created a new one ...

miguelriemoliveira, 2020-09-15:
Bag info is:

```bash
path:        miguel_bag.bag
version:     2.0
duration:    29.6s
start:       Sep 15 2020 10:18:44.50 (1600161524.50)
end:         Sep 15 2020 10:19:14.05 (1600161554.05)
size:        1.2 GB
messages:    15664
compression: none [359/359 chunks]
types:       sensor_msgs/CameraInfo      [c9a58c1b0b154e0e6da7578cb991d214]
             sensor_msgs/CompressedImage [8f7a12909da2c9d3332d540a0977563f]
             sensor_msgs/PointCloud2     [1158d486dd51d683ce2f1be655c3c181]
             tf2_msgs/TFMessage          [94810edda583a504dfda3829e70d7eec]
topics:      /camerat265/fisheye1/camera_info               887 msgs    : sensor_msgs/CameraInfo     
             /camerat265/fisheye1/image_raw/compressed      887 msgs    : sensor_msgs/CompressedImage
             /camerat265/fisheye2/camera_info               887 msgs    : sensor_msgs/CameraInfo     
             /camerat265/fisheye2/image_raw/compressed      886 msgs    : sensor_msgs/CompressedImage
             /tf                                          10313 msgs    : tf2_msgs/TFMessage         
             /velodyne_points                               293 msgs    : sensor_msgs/PointCloud2    
             /zed/zed_node/left_raw/camera_info             390 msgs    : sensor_msgs/CameraInfo     
             /zed/zed_node/left_raw/image/compressed        330 msgs    : sensor_msgs/CompressedImage
             /zed/zed_node/point_cloud/cloud_registered      62 msgs    : sensor_msgs/PointCloud2    
             /zed/zed_node/right_raw/camera_info            390 msgs    : sensor_msgs/CameraInfo     
             /zed/zed_node/right_raw/image/compressed       339 msgs    : sensor_msgs/CompressedImage

```

The problem is with topics  _/zed/zed_node/left_raw/camera_info _ and  _/zed/zed_node/right_raw/camera_info_ which have the stamp always equal to 0.

miguelriemoliveira, 2020-09-16:
I think it is fixed. The logic was to play the bag remapping the problematic topics to new names, e.g.

_/zed/zed_node/left_raw/camera_info _
->
_/zed/zed_node/left_raw/camera_info/incorrect _

Then, we run a topic tools transform (http://wiki.ros.org/topic_tools/transform) to fix the message for each camera.

This script should play the bag with the corrected stamps. You could make a new bag file of the fixed playback (in that case remember to exclude bagging the incorrect topics).

Here's the script:

[run_and_fix_bag.sh.zip](https://github.com/lardemua/atom/files/5228810/run_and_fix_bag.sh.zip)

Please close if it works.

aaguiar96, 2020-09-21:
Hi @miguelriemoliveira 

This is working. Just had to add some minor changes.
Thanks

closed 2020-09-21 by aaguiar96

miguelriemoliveira, 2020-09-21:
Great!